coolsnowwolf/lede #11910: OpenWrt R22.7.7，编译的闭源驱动固件无法桥接无线ra0、rai0

hpx502766238, 2024-02-22:
### 详细叙述

如题，想给RedmiAX6S编译一个闭源驱动固件，mt7622平台，lede自带的闭源驱动暂不支持7622
但是https://github.com/bricco1981/MT7622-mtkwifi这里有人做了闭源驱动移植，但经我多次克隆openwrt官方源码进行测试，这个驱动仅支持5.10内核
现在想用LEDE做一份相同的，所以选择了这个commit(https://github.com/coolsnowwolf/lede/tree/20220716)进行了编译：
1.将package/lean/mt移除，替换为上述闭源驱动
2.中途有报错，但找到了解决方法https://github.com/coolsnowwolf/lede/issues/11908
3.一切编译正常结束后，刷入路由器，有线连接成功进入web界面

但出现了问题：
进入无线设置勾选**add rai0 to br-lan**保存，重载驱动
但发现无线设备ra0、rai0始终无法成功桥接，**在接口-物理设置，始终不显示ra和rai0设备，但在终端使用ifconfig命令却可以看见有ra0和rai0启用**
反复重载驱动、重启路由都无效，终端连接wifi也始终无法获取IP地址，不知道如何解决这个问题

[image]
[image]
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

XiaomiRedmiAX6S

### 详细日志

成功编译

hpx502766238, 2024-02-24:
找到暂时解决办法，接口-lan-物理设置，勾选自定义接口，输入ra0、rai0，保存应用。
打开命令行，uci show network.lan，发现已成功修改，无线成功桥接
网络-接口-接口总览可以看到多了两个无线的图标，鼠标指上去显示ra0 rai0。
但是，再添加新接口时，可勾选的物理接口依然只有以太网口（eth0 lan1-lan3），没有出现ra0、rai0、wds等
所以我怀疑，是不是**luci里对显示的接口做了过滤**，官方openwrt是全部接口显示的

hpx502766238, 2024-02-25:
折腾了两天，桥接的问题已经解决了，通过改闭源驱动mtkwifi.lua脚本，实现了适配。
原来lean的代码uci相关命令和OP官方是有些区别的
官方：		
uci add_list network.cfg030f15.ports="ra0"
uci add_list network.cfg030f15.ports="rai0"
uci commit network
/etc/init.d/network reload
lean：
uci set network.lan.ifname="$(uci get network.lan.ifname) ra0 rai0"
uci commit network
/etc/init.d/network reload

但是，luci界面管理的设备依然不显示除LAN、WAN以外的所有接口，只有在概览中显示，这依然是个问题

hpx502766238, 2024-02-25:
[image]

hpx502766238, 2024-02-25:
[image]

hpx502766238, 2024-02-25:
[image]

WYC-2020, 2024-02-26:
你那个 仓库别人都说了适配的 
[image]
22.03，官方op 22.03 使用的luci和l的luci完全就是不同的 l的更老 ，所以自然命令不同了

hpx502766238, 2024-02-26:
> 你那个 仓库别人都说了适配的 [...]

原来如此。希望lean能更新一下luci就好

WYC-2020, 2024-02-26:
> 原来如此。希望lean能更新一下luci就好

别想 那个js版的luci 拉胯的很 lua 才是经典

hpx502766238, 2024-02-26:
> 别想 那个js版的luci 拉胯的很 lua 才是经典

但这个接口显示不全还是得修复一下吧，强迫症看着难受

blzadx, 2024-03-12:
> ![屏幕截图 2024-02-25 191656](https://private-user-images.githubusercontent.com/50779358/307586385-e950287d-2c9d-4e97-b366-d4cd2e3ad517.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDEyNDAsIm5iZiI6MTcxMDI0MDk0MCwicGF0aCI6Ii81MDc3OTM1OC8zMDc1ODYzODUtZTk1MDI4N2QtMmM5ZC00ZTk3LWIzNjYtZDRjZDJlM2FkNTE3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEyVDEwNTU0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU4MmY4ZDVmZDdkYzA3ZjYyZWVhZTZlYTIzYTQ0OTdiYzg1NTcxYjU5YTdiZDAxNmJiYzgyZjg2NDMyZmUxYTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.a-Uuu1raz_uhuWRobQsjSH4rfwMGQso4Ijv4O6Ffpyg)

- 你这个问题我也碰到过，这跟luci里的一个接口扫描函数有关。
- 这个界面的接口是通过luci/modules/luci-base/luasrc/model/network.lua中的get_interfaces函数扫描列举的。标准的openwrt会将接口分为以太网接口，vlan接口和wireless接口，分别记录在network.interface，network.switch_vlan，以及wireless.wifi-iface中，你打开这个页面的时候，get_interfaces就会通过uci从这些属性中读取。
- openwrt自带的开源驱动是符合前面所说的规范的，但闭源驱动似乎都不符合，也就是使用闭源驱动的时候wireless.wifi-iface其实是空的。
- 但是我用padavanonly的immortalwrt源码编译固件时，这个界面又看得到各个无线接口。然后我对比了一下他和官方源码的区别，发现network.lua最开头有一些接口判别列表，再仔细看一下get_interfaces的逻辑，它在读取network.interface的时候会排除被忽略的接口，虚拟接口，以及无线接口。也就是说network.interface里实际上是有无线接口的，但函数只想从这里获取以太网接口，所以就把它们排除了。
- 实际上padavanonly的解决办法就是把开头的接口判别列表（IFACE_PATTERNS_WIRELESS）给删了不少东西，于是无线接口就可以被识别为以太网接口，也就能在这个界面显示出来了。我自己是稍微改了改get_interfaces的逻辑，让它读取到network.interface的无线接口时不要排除掉，直接列入无线接口列表。
- 但以上两种做法仅仅只适合为特殊机型更改，因为这个问题说到底是闭源驱动没有完全符合openwrt规范，直接把lede源码改了是不合适的，毕竟还有不少正常的机型。但是我不知道怎么改，大雕估计也不想费事。不过这个问题要命的地方在于，大雕只是简单粗暴地用uci set network.lan.ifname="$(uci get network.lan.ifname) ra0 rai0"将无线接口加入lan，而luci界面里实际上读不到ra0 rai0，因此一旦用页面修改network配置，再保存应用，lan口会立即丢失ra0 rai0，必须手动重新加入。

hpx502766238, 2024-03-22:
> > ![屏幕截图 2024-02-25 191656](https://private-user-images.githubusercontent.com/50779358/307586385-e950287d-2c9d-4e97-b366-d4cd2e3ad517.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDEyNDAsIm5iZiI6MTcxMDI0MDk0MCwicGF0aCI6Ii81MDc3OTM1OC8zMDc1ODYzODUtZTk1MDI4N2QtMmM5ZC00ZTk3LWIzNjYtZDRjZDJlM2FkNTE3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEyVDEwNTU0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU4MmY4ZDVmZDdkYzA3ZjYyZWVhZTZlYTIzYTQ0OTdiYzg1NTcxYjU5YTdiZDAxNmJiYzgyZjg2NDMyZmUxYTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.a-Uuu1raz_uhuWRobQsjSH4rfwMGQso4Ijv4O6Ffpyg)
> 
> * 你这个问题我也碰到过，这跟luci里的一个接口扫描函数有关。
> * 这个界面的接口是通过luci/modules/luci-base/luasrc/model/network.lua中的get_interfaces函数扫描列举的。标准的openwrt会将接口分为以太网接口，vlan接口和wireless接口，分别记录在network.interface，network.switch_vlan，以及wireless.wifi-iface中，你打开这个页面的时候，get_interfaces就会通过uci从这些属性中读取。
> * openwrt自带的开源驱动是符合前面所说的规范的，但闭源驱动似乎都不符合，也就是使用闭源驱动的时候wireless.wifi-iface其实是空的。
> * 但是我用padavanonly的immortalwrt源码编译固件时，这个界面又看得到各个无线接口。然后我对比了一下他和官方源码，发现padavanonly将network.lua开头的接口判别列表（IFACE_PATTERNS_WIRELESS）给删了不少东西。再仔细看一下get_interfaces的逻辑，原来它在读取network.interface的时候会排除“被忽略的接口“，”虚拟接口“，以及”无线接口”。也就是说network.interface里实际上是有无线接口的，但函数只想从这里获取以太网接口，所以就把它们排除了。
> * 我自己是稍微改了改get_interfaces的逻辑，让它读取到network.interface的无线接口时不要排除掉，直接列入无线接口列表。
> * 但以上两种做法仅仅只适合为特殊机型更改，因为这个问题说到底是闭源驱动没有完全符合openwrt规范，直接把lede源码改了是不合适的，毕竟还有不少正常的机型。但是我不知道怎么改，大雕估计也不想费事。不过这个问题要命的地方在于，大雕只是简单粗暴地用uci set network.lan.ifname="$(uci get network.lan.ifname) ra0 rai0"将无线接口加入lan，而luci界面里实际上读不到ra0 rai0，因此一旦用页面修改network配置，再保存应用，lan口会立即丢失ra0 rai0，必须手动重新加入。

那有没有办法让闭源驱动符合openwrt规范呢？从而让openwrt识别到？

blzadx, 2024-03-24:
> 那有没有办法让闭源驱动符合openwrt规范呢？从而让openwrt识别到？

很遗憾我不知道具体要改哪些东西，只知道是跟netifd有关的内容。

hpx502766238, 2024-04-05:
> 很遗憾我不知道具体要改哪些东西，只知道是跟netifd有关的内容。

我尝试了用padavanonly的network.lua替代，发现可以识别了，但识别为以太网接口。看来padavan的逻辑是直接把闭源驱动当作普通接口处理，虽然这样可以解决问题，但自己还是有些强迫症觉得看着不舒服，还是希望op将它识别为无线接口。我看了开源驱动固件下的无线接口识别情况，luci界面会获取无线ssid、工作模式，而这些对于闭源驱动来说获取方法是完全不一样的，所以我觉得是不是这里导致了界面不显示，所以我是想改一下op无线信息的获取逻辑，就是不知道要改哪些代码。

hpx502766238, 2024-04-05:
> > > > > ![屏幕截图 2024-02-25 191656](https://private-user-images.githubusercontent.com/50779358/307586385-e950287d-2c9d-4e97-b366-d4cd2e3ad517.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDEyNDAsIm5iZiI6MTcxMDI0MDk0MCwicGF0aCI6Ii81MDc3OTM1OC8zMDc1ODYzODUtZTk1MDI4N2QtMmM5ZC00ZTk3LWIzNjYtZDRjZDJlM2FkNTE3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEyVDEwNTU0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU4MmY4ZDVmZDdkYzA3ZjYyZWVhZTZlYTIzYTQ0OTdiYzg1NTcxYjU5YTdiZDAxNmJiYzgyZjg2NDMyZmUxYTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.a-Uuu1raz_uhuWRobQsjSH4rfwMGQso4Ijv4O6Ffpyg)
> > > > 
> > > > 
> > > > 
> > > > * 你这个问题我也碰到过，这跟luci里的一个接口扫描函数有关。
> > > > * 这个界面的接口是通过luci/modules/luci-base/luasrc/model/network.lua中的get_interfaces函数扫描列举的。标准的openwrt会将接口分为以太网接口，vlan接口和wireless接口，分别记录在network.interface，network.switch_vlan，以及wireless.wifi-iface中，你打开这个页面的时候，get_interfaces就会通过uci从这些属性中读取。
> > > > * openwrt自带的开源驱动是符合前面所说的规范的，但闭源驱动似乎都不符合，也就是使用闭源驱动的时候wireless.wifi-iface其实是空的。
> > > > * 但是我用padavanonly的immortalwrt源码编译固件时，这个界面又看得到各个无线接口。然后我对比了一下他和官方源码，发现padavanonly将network.lua开头的接口判别列表（IFACE_PATTERNS_WIRELESS）给删了不少东西。再仔细看一下get_interfaces的逻辑，原来它在读取network.interface的时候会排除“被忽略的接口“，”虚拟接口“，以及”无线接口”。也就是说network.interface里实际上是有无线接口的，但函数只想从这里获取以太网接口，所以就把它们排除了。
> > > > * 我自己是稍微改了改get_interfaces的逻辑，让它读取到network.interface的无线接口时不要排除掉，直接列入无线接口列表。
> > > > * 但以上两种做法仅仅只适合为特殊机型更改，因为这个问题说到底是闭源驱动没有完全符合openwrt规范，直接把lede源码改了是不合适的，毕竟还有不少正常的机型。但是我不知道怎么改，大雕估计也不想费事。不过这个问题要命的地方在于，大雕只是简单粗暴地用uci set network.lan.ifname="$(uci get network.lan.ifname) ra0 rai0"将无线接口加入lan，而luci界面里实际上读不到ra0 rai0，因此一旦用页面修改network配置，再保存应用，lan口会立即丢失ra0 rai0，必须手动重新加入。
> > > 
> > > 
> > > 那有没有办法让闭源驱动符合openwrt规范呢？从而让openwrt识别到？
> > 
> > 
> > 很遗憾我不知道具体要改哪些东西，只知道是跟netifd有关的内容。
> 
> 我尝试了用padavanonly的network.lua替代，发现可以识别了，但识别为以太网接口，而原先虽然接口列表不显示，但在概览中是可以看到ra0被识别为无线接口的。看来padavan的逻辑是直接把闭源驱动当作普通以太网接口处理，虽然这样可以解决问题，但自己还是有些强迫症，觉得图标和文字与实际不匹配看着不舒服，还是希望op将它识别为无线接口。我看了开源驱动固件下的无线接口识别情况，luci界面会获取无线ssid、工作模式，而这些对于闭源驱动来说获取方法是完全不一样的，闭源驱动使用wireless-tools中iwconfig获取，所以我觉得是不是op获取不到无线信息导致了界面不显示，所以我是想改一下op无线信息的获取逻辑，就是不知道要改哪些代码。

折腾了一下午，尝试了手动新建/etc/config/wireless配置文件，发现原生无线配置界面出来了，但只能读到部分无线信息（ssid，bssid等），而其他很多设置残缺，接口中也显示了无线接口，但是为灰色，尝试在原生无线配置界面重启无线，接口依然为灰色。看来确实存在闭源驱动与openwrt管理方式不完全兼容的问题（可能驱动缺少部分API吧）。所以综上，还是照padavanonly将其识别为以太网接口为最佳方案。
[image]

lucian521, 2024-08-05:
有详细的步骤吗 ？ 我也遇到这个问题了呢

coolsnowwolf, 2026-09-05:
问题已被后续排查定位：这是第三方闭源 MT7622 驱动不符合 netifd/LuCI 的无线接口模型（缺少标准 `wireless.wifi-iface`/iwinfo 能力），不是 LuCI 对标准无线设备的过滤回归。评论中已确认手动加入接口可桥接，或按 padavanonly 方式当作普通以太网接口显示，但无线配置/状态能力仍不完整。为该闭源驱动改动通用 LuCI 会影响正常设备，当前没有安全的上游修复，关闭为 not planned。

closed 2026-09-05 by coolsnowwolf