coolsnowwolf/lede #11918: ERROR: package/feeds/small/gn [host] failed to build.

Samre, 2024-02-25:
### 详细叙述

make[3]: *** [Makefile:69: /workdir/openwrt/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.built] Error 1
[3717](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3718)
make[3]: Leaving directory '/workdir/openwrt/feeds/small/gn'
[3718](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3719)
time: package/feeds/small/gn/host-compile#0.17#0.10#0.27
[3719](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3720)
    ERROR: package/feeds/small/gn [host] failed to build.
[3720](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3721)
make[2]: *** [package/Makefile:114: package/feeds/small/gn/host/compile] Error 1
[3721](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3722)
make[2]: Leaving directory '/workdir/openwrt'
[3722](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3723)
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
[3723](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3724)
make[1]: Leaving directory '/workdir/openwrt'
[3724](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3725)
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
[3725](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3726)
Error: Process completed with exit code 2.

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

g++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++20 -Wno-redundant-move -c ../src/base/command_line.cc -o src/base/command_line.o
[3714](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3715)
g++: error: unrecognized command line option '-std=c++20'; did you mean '-std=c++2a'?
[3715](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3716)
ninja: build stopped: subcommand failed.
[3716](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3717)
make[3]: *** [Makefile:69: /workdir/openwrt/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.built] Error 1
[3717](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3718)
make[3]: Leaving directory '/workdir/openwrt/feeds/small/gn'
[3718](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3719)
time: package/feeds/small/gn/host-compile#0.17#0.10#0.27
[3719](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3720)
    ERROR: package/feeds/small/gn [host] failed to build.
[3720](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3721)
make[2]: *** [package/Makefile:114: package/feeds/small/gn/host/compile] Error 1
[3721](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3722)
make[2]: Leaving directory '/workdir/openwrt'
[3722](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3723)
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
[3723](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3724)
make[1]: Leaving directory '/workdir/openwrt'
[3724](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3725)
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
[3725](https://github.com/Samre/OpenWrt-x86_64/actions/runs/7986640951/job/21807458019#step:11:3726)
Error: Process completed with exit code 2.

lxg2001, 2024-02-25:
gcc版本太低

WYC-2020, 2024-02-26:
你这个工具链太低了呗 根本就不支持c++20这个选项

unkonwuser01, 2024-02-27:
这个问题怎么解救呢。  用action 编译的时候，脚本里修改工具链么

closed 2024-02-28 by coolsnowwolf

fairok, 2024-03-01:
楼主这个问题咋解决的啊？

NeedforGit, 2024-03-03:
> 你这个工具链太低了呗 根本就不支持c++20这个选项

请问云编译的话，如何更新这个工具链啊？

smallprogram, 2024-07-17:
```
# 添加PPA源
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt update

# 安装最新的GCC（假设为GCC 11）
sudo apt install gcc-11 g++-11

# 设置默认版本
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 100
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-11 100

# 配置默认版本
sudo update-alternatives --config gcc
sudo update-alternatives --config g++

# 验证安装
gcc --version
g++ --version

```