coolsnowwolf/lede #11919: 新版Xray需要go 1.22以上环境才能编译，目前go为1.21.5，烦请大佬升级一下go版本

bi4tms, 2024-02-26:
### 详细叙述

如题
@big

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

同上

closed 2024-02-26 by bi4tms

ChrisLiu0343, 2024-03-02:
同样。。。。。

15298867332, 2024-03-07:
go: ../../go.mod requires go >= 1.22 (running go 1.21.5; GOTOOLCHAIN=local)
同样的问题

HongJie-Master, 2024-03-24:
方法一、
升级golang 为 1.22.x 版本
rm -rf feeds/packages/lang/golang
git clone https://github.com/sbwml/packages_lang_golang -b 22.x feeds/packages/lang/golang

方法二、
关于go的版本定义在：feeds/packages/lang/golang/golang/Makefile
修改此Makefile文件即可
例如:
GO_VERSION_MAJOR_MINOR:=1.22
GO_VERSION_PATCH:=1
同时，对应的hash值也应修改
PKG_HASH:=79c9b91d7f109515a25fc3ecdaad125d67e6bdb54f6d4d98580f46799caea321

aceslup, 2024-04-29:
> 方法一、 升级golang 为 1.22.x 版本 rm -rf feeds/packages/lang/golang git clone https://github.com/sbwml/packages_lang_golang -b 22.x feeds/packages/lang/golang
>
> 方法二、 [...]

方法二我这边替换了，依旧会编译不过，不知道原因。然后大雕仓库里就更新好版本了。