coolsnowwolf/lede #11921: xray需要go 1.22 目前版本太低

GreatMichaelLee, 2024-02-27:
### 详细叙述

touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.v2ray-plugin_installed
echo "v2ray-plugin" >> /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/v2ray-plugin.default.install
make[3]: Leaving directory '/home/builder/lede_x86/feeds/helloworld/v2ray-plugin'
time: package/feeds/helloworld/v2ray-plugin/compile#0.09#0.06#0.17
make[3]: Entering directory '/home/builder/lede_x86/feeds/helloworld/xray-plugin'
rm -f /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.built
touch /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/builder/lede_x86/tmp/go-build" GO_MOD_CACHE_DIR="/home/builder/lede_x86/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/teddysun/xray-plugin" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/teddysun/xray-plugin/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8" GO_BUILD_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.go_work/build" GO_BUILD_BIN_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8=xray-plugin-1.8.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8=xray-plugin-1.8.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.go_work/build" GOCACHE="/home/builder/lede_x86/tmp/go-build" GOMODCACHE="/home/builder/lede_x86/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash /home/builder/lede_x86/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1708954693' -linkmode external -extldflags '-L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"
Finding targets
go: github.com/xtls/xray-core@v1.8.8 requires go >= 1.22 (running go 1.21.5; GOTOOLCHAIN=local)

Building targets
no Go files in /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.go_work/build

make[3]: *** [Makefile:42: /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.built] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/helloworld/xray-plugin'
time: package/feeds/helloworld/xray-plugin/compile#0.22#1.25#7.97
    ERROR: package/feeds/helloworld/xray-plugin failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/xray-plugin/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$
builder@Build-Server:/home/builder/lede_x86$ go version

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.v2ray-plugin_installed
echo "v2ray-plugin" >> /home/builder/lede_x86/staging_dir/target-x86_64_musl/pkginfo/v2ray-plugin.default.install
make[3]: Leaving directory '/home/builder/lede_x86/feeds/helloworld/v2ray-plugin'
time: package/feeds/helloworld/v2ray-plugin/compile#0.09#0.06#0.17
make[3]: Entering directory '/home/builder/lede_x86/feeds/helloworld/xray-plugin'
rm -f /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.built
touch /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/builder/lede_x86/tmp/go-build" GO_MOD_CACHE_DIR="/home/builder/lede_x86/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/teddysun/xray-plugin" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/teddysun/xray-plugin/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8" GO_BUILD_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.go_work/build" GO_BUILD_BIN_DIR="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8=xray-plugin-1.8.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8=xray-plugin-1.8.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.go_work/build" GOCACHE="/home/builder/lede_x86/tmp/go-build" GOMODCACHE="/home/builder/lede_x86/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash /home/builder/lede_x86/feeds/packages/lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1708954693' -linkmode external -extldflags '-L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"
Finding targets
go: github.com/xtls/xray-core@v1.8.8 requires go >= 1.22 (running go 1.21.5; GOTOOLCHAIN=local)

Building targets
no Go files in /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.go_work/build

make[3]: *** [Makefile:42: /home/builder/lede_x86/build_dir/target-x86_64_musl/xray-plugin-1.8.8/.built] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/helloworld/xray-plugin'
time: package/feeds/helloworld/xray-plugin/compile#0.22#1.25#7.97
    ERROR: package/feeds/helloworld/xray-plugin failed to build.
make[2]: *** [package/Makefile:116: package/feeds/helloworld/xray-plugin/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$
builder@Build-Server:/home/builder/lede_x86$ go version

sbwml, 2024-02-27:
解决方法：https://github.com/xiaorouji/openwrt-passwall/discussions/2969

GreatMichaelLee, 2024-02-27:
> 解决方法：[xiaorouji/openwrt-passwall#2969](https://github.com/xiaorouji/openwrt-passwall/discussions/2969)

多谢，coolsnowwolf大佬能不能把这里的package仓库的go版本也升级一下呢，这样我就没有必要让自己本地维护一个跟仓库分支不同步的go版本了。

GreatMichaelLee, 2024-02-27:
> > 解决方法：[xiaorouji/openwrt-passwall#2969](https://github.com/xiaorouji/openwrt-passwall/discussions/2969)
> 
> 多谢，coolsnowwolf大佬能不能把这里的package仓库的go版本也升级一下呢，这样我就没有必要让自己本地维护一个跟仓库分支不同步的go版本了。

看到昨天本来update go到1.22了，不知道什么原因revert掉了。。。。

sbwml, 2024-02-27:
> > > 解决方法：[xiaorouji/openwrt-passwall#2969](https://github.com/xiaorouji/openwrt-passwall/discussions/2969)
> > 
> > 
> > 多谢，coolsnowwolf大佬能不能把这里的package仓库的go版本也升级一下呢，这样我就没有必要让自己本地维护一个跟仓库分支不同步的go版本了。
> 
> 看到昨天本来update go到1.22了，不知道什么原因revert掉了。。。。

用屁股想呀，go1.21 需要最低 go1.17.3 工具链编译，而 go1.17.3 又要 go1.4 来编译，也就是一个 go1.21 从源码构建，需要用到两个工具链。

而 go1.22 需要最低 1.20.6，但是这提交https://github.com/coolsnowwolf/packages/commit/5d77e8e4266265b7d1f1be99a446a2beae47321d 直接 bootstrap 1.20.14，这怎么可能行呢？

从源码编译到最新的 go1.22，它最开始必须是由 gcc 工具链开始编译，也就是 go1.4 版本支持 gcc -> go，更网上的版本就只能是  go -> go。

编译最新的 go1.22 它的顺序应该是：go1.4-bootstrap -> go1.17.3 -> go1.20.6/go1.21.x > go1.22

GreatMichaelLee, 2024-02-27:
> > > > 解决方法：[xiaorouji/openwrt-passwall#2969](https://github.com/xiaorouji/openwrt-passwall/discussions/2969)
> > > 
> > > 
> > > 多谢，coolsnowwolf大佬能不能把这里的package仓库的go版本也升级一下呢，这样我就没有必要让自己本地维护一个跟仓库分支不同步的go版本了。
> > 
> > 
> > 看到昨天本来update go到1.22了，不知道什么原因revert掉了。。。。
> 
> 用屁股想呀，go1.21 需要最低 go1.17.3 工具链编译，而 go1.17.3 又要 go1.4 来编译，也就是一个 go1.21 从源码构建，需要用到两个工具链。
> 
> 而 go1.22 需要最低 1.20.6，但是这提交[coolsnowwolf/packages@5d77e8e](https://github.com/coolsnowwolf/packages/commit/5d77e8e4266265b7d1f1be99a446a2beae47321d) 直接 bootstrap 1.20.14，这怎么可能行呢？
> 
> 从源码编译到最新的 go1.22，它最开始必须是由 gcc 工具链开始编译，也就是 go1.4 版本支持 gcc -> go，更往上的版本就只能是 go -> go。
> 
> 编译最新的 go1.22 它的顺序应该是：go1.4-bootstrap -> go1.17.3 -> go1.20.6/go1.21.x > go1.22

不懂go的依赖和编译，你这等于对牛弹琴，哈哈

WYC-2020, 2024-02-27:
垃圾玩意 没事天天再升 这个xray 不知道一天哪来那么多更新 没事就更新go go一个编程语言非要没事去搞那个玩意，看下xray打个补丁给他降回去 看下可行不

WYC-2020, 2024-02-27:
就和c++ 一天追寻最新标准一样 最那么勤没必要，能用稳定才是王道

yanxin152133, 2024-02-27:
啥时候修啥时候修

GreatMichaelLee, 2024-02-27:
领导都说了，要以旧换新，哈哈哈

WYC-2020, 2024-02-27:
> 领导都说了，要以旧换新，哈哈哈
> […](#)

换新个鸡毛 看上面那个解决方案 直接打patch就能编译过，就证明根本没用上最新的go什么特性，有可能新版go 有什么特别的优化 但是感觉微乎其微 反而还一大推问题，经常搞这种一刀切 都不考虑编译兼容问题的

qqsir-dev, 2024-02-27:
试试这个
rm -rf feeds/packages/lang/golang && git clone -b 21.x https://github.com/sbwml/packages_lang_golang feeds/packages/lang/golang

yanxin152133, 2024-02-27:
> 试试这个 rm -rf feeds/packages/lang/golang && git clone -b 22.x https://github.com/sbwml/packages_lang_golang feeds/packages/lang/golang

等领导来修不香嘛：）

WYC-2020, 2024-02-27:
[image]
[image]
 来吧看吧 如此修复，go的自己编译自己太恶心了，1.4 编译1.17 1.17编译1.20 1.20才能编译1.22 真鸡儿恶心，相当于要多增加一个1.20的编译 相当恶心 居然1.17不能直接编译1.22 怀念以前1.4通吃好几个版本的情况，如果不想这样 你们本机安装go1.20编译好的版本 然后设置 GOROOT_BOOTSTRAP 也是可以的，看这个 https://github.com/golang/go/issues/44505 go的说明，反而这样导致后期编译更加老火，当然是在 本机没安装go二进制的情况下 肯定后续还会陆续增加 编译好几个工具链 才能获取到最后一个，伤不起，明天看下能否有方法跳过这狗屎玩意

WYC-2020, 2024-02-27:
改名Makefile 替换golang 同名文件 编译完事
[Makefile.txt](https://github.com/coolsnowwolf/lede/files/14418110/Makefile.txt)

yanxin152133, 2024-02-27:
> 你们本机安装go1.20编译好的版本 然后设置 GOROOT_BOOTSTRAP 也是可以的，看这个 golang/go#44505

加油加油

WYC-2020, 2024-02-27:
看你们说隔壁修复的直接可以1.4编译1.20咋我试不行，明天在试下，要想一个办法根治这多级问题

hzjnet, 2024-02-27:
> 垃圾玩意 没事天天再升 这个xray 不知道一天哪来那么多更新 没事就更新go go一个编程语言非要没事去搞那个玩意，看下xray打个补丁给他降回去 看下可行不

内核版本不也是一样吗？  修安全问题就行了，没事儿动不动该内核是不是也是吃饱撑了？（~~）

sbwml, 2024-02-27:
编译个鸡毛，自己玩的东西，遵循个屁的规范。自1.20 开始需要两个低版本go才能编译到 1.20，当时我意识到这垃圾玩意是个无底洞，就直接用 prebuilt 做 bootstrap 去编译目标版本 https://github.com/sbwml/packages_lang_golang 。好处是无论往后更新多少个版本，操作永远是下载官方 prebuilt 去编译最新版本，出来的东西一模一样。

如果要遵循 openwrt 规范，以后再更新 1.23、24。。。30 呢？五六七八九十个 bootstrap，直接不用编译 openwrt 了 ，耗时直接全给 golang 好不好

按这种尿性来看，golang 必定要支持使用宿主机版本，就像 openwrt 现有的 external toolchain 一样

coolsnowwolf, 2024-02-27:
> 编译个鸡毛，自己玩的东西，遵循个屁的规范。自1.20 开始需要两个低版本go才能编译到 1.20，当时我意识到这垃圾玩意是个无底洞，就直接用 prebuilt 做 bootstrap 去编译目标版本 https://github.com/sbwml/packages_lang_golang 。好处是无论往后更新多少个版本，操作永远是下载官方 prebuilt [...]

你这样有个问题 ，我用龙芯的 linux 编译 openwrt 就失败了。。。反而是 从 1.4 编译1.17 ，1.17编译1.20， 1.20才能编译1.22 就好了。。。

WYC-2020, 2024-02-27:
> > 垃圾玩意 没事天天再升 这个xray 不知道一天哪来那么多更新 没事就更新go go一个编程语言非要没事去搞那个玩意，看下xray打个补丁给他降回去 看下可行不
> 
> 内核版本不也是一样吗？ 修安全问题就行了，没事儿动不动该内核是不是也是吃饱撑了？（~~）

升内核有一点这个，不过升内核别人并不会说我升个内核你就必须的用gcc13才能编译，别人内核升级并不会升级附加的编译环境做了兼容，但是你看看这玩意，没事就升级依赖的，升级一个xray吧go也升级，但是这玩意并没用到最新go的特性，有点过度追了，也不兼容，今天再看看有啥永久解决的办法，内核我现在还在用5.4虽然现在内核比较多，我都稳定快一年半了，很多东西往往不是最新的就最好，路由器这玩意稳定才可以，工控系统还有4.14内核的多的是

WYC-2020, 2024-02-28:
看了不行 即使吧cmd/dist下面编译过了 还有好多库都是1.4没有的，所以目前只能这样做，要用go1.22 必须得用与之相近的版本进行编译，如果没有新增什么新的库 就可以夸好几本版本都用这个工具链 如果有新增猜测就不行了，暂时先这样 收工

WYC-2020, 2024-02-28:
> 解决方法：[xiaorouji/openwrt-passwall#2969](https://github.com/xiaorouji/openwrt-passwall/discussions/2969)

同时这个解决方案的第一种方案直接那样改 是不行的 1.4是没法直接编译1.20的，不信的可以试试

sbwml, 2024-02-28:
> > 解决方法：[xiaorouji/openwrt-passwall#2969](https://github.com/xiaorouji/openwrt-passwall/discussions/2969)
> 
> 同时这个解决方案的第一种方案直接那样改 是不行的 1.4是没法直接编译1.20的，不信的可以试试

我可不是要去编译1.20。我是直接编译 1.22

sbwml, 2024-02-28:
> 你这样有个问题 ，我用龙芯的 linux 编译 openwrt 就失败了。。。反而是 从 1.4 编译1.17 ，1.17编译1.20， 1.20才能编译1.22 就好了。。。

因为是个人自用的东西，用作 bootstrap 的 prebuilt 只添加了 linux-x64 & aarch64 两个宿主机平台

sbwml, 2024-02-28:
> > 解决方法：[xiaorouji/openwrt-passwall#2969](https://github.com/xiaorouji/openwrt-passwall/discussions/2969)
> 
> 1.4是没法直接编译1.20的

这是  go1.20.6 prebuilt，不是 src

WYC-2020, 2024-02-28:
> > > 解决方法：[xiaorouji/openwrt-passwall#2969](https://github.com/xiaorouji/openwrt-passwall/discussions/2969)
> > 
> > 
> > 1.4是没法直接编译1.20的
> 
> 这是 go1.20.6 prebuilt，不是 src

失误失误 看明白了 你下载的是直接别人编译好的 工具链 用这个来编译 1.22  嗯是的 你这个可以，但是平台的话 可能不一定全

sbwml, 2024-02-28:
> 你下载的是直接别人编译好的 工具链 用这个来编译 1.22 嗯是的 你这个可以，但是平台的话 可能不一定全

它来自官方 https://go.dev/dl/ ，通常能叫得出的平台，它都有。只是我个人需求只有 linux-x64，其它平台并没有添加。然而添加一个宿主机平台比想象中要简单得多，只需要写一个 ARCH信息、SHA256 就完成。

甚至 SHA256 这些信息都是直接复制粘贴，官方直接给出列表 https://go.dev/dl/ ：

```
go1.22.0.darwin-amd64.tar.gz	Archive	macOS	x86-64	67MB	ebca81df938d2d1047cc992be6c6c759543cf309d401b86af38a6aed3d4090f4
go1.22.0.darwin-amd64.pkg	Installer	macOS	x86-64	68MB	4ef52d6242bc673702a038becd1d431aeb930ebc596db169648a622a99239107
go1.22.0.darwin-arm64.tar.gz	Archive	macOS	ARM64	64MB	bf8e388b09134164717cd52d3285a4ab3b68691b80515212da0e9f56f518fb1e
go1.22.0.darwin-arm64.pkg	Installer	macOS	ARM64	65MB	7ca20c077148b69bb99887e04983955e60bd7c70c55e00bf85f9f7194874a3b2
go1.22.0.linux-386.tar.gz	Archive	Linux	x86	64MB	1e209c4abde069067ac9afb341c8003db6a210f8173c77777f02d3a524313da3
go1.22.0.linux-amd64.tar.gz	Archive	Linux	x86-64	66MB	f6c8a87aa03b92c4b0bf3d558e28ea03006eb29db78917daec5cfb6ec1046265
go1.22.0.linux-arm64.tar.gz	Archive	Linux	ARM64	63MB	6a63fef0e050146f275bf02a0896badfe77c11b6f05499bb647e7bd613a45a10
go1.22.0.linux-armv6l.tar.gz	Archive	Linux	ARMv6	64MB	0525f92f79df7ed5877147bce7b955f159f3962711b69faac66bc7121d36dcc4
go1.22.0.windows-386.zip	Archive	Windows	x86	71MB	553d44928509965cbda02a45b35ab01cf8b925534bc526a34e2d9dc7794b57e8
go1.22.0.windows-386.msi	Installer	Windows	x86	58MB	09ec10a7a42c47102f9f4889c59b3da132b7b1c7b099f4664488eedc6f278999
go1.22.0.windows-amd64.zip	Archive	Windows	x86-64	73MB	78b3158fe3aa358e0b6c9f26ecd338f9a11441e88bc434ae2e9f0ca2b0cc4dd3
go1.22.0.windows-amd64.msi	Installer	Windows	x86-64	60MB	11a47de052db9971359e8c2f3a1667f8d56fa4c6bbec0687cf4cf2403a07628a
Other Ports
go1.22.0.aix-ppc64.tar.gz	Archive	aix	ppc64	64MB	190e105fc4133a8b5bb1492f368fa89aa4b729270441120714be7ee82e871ebc
go1.22.0.dragonfly-amd64.tar.gz	Archive	dragonfly	x86-64	66MB	357ab446200effa26c73ccaf3e8551426428950bf2490859fb296a09e53228b1
go1.22.0.freebsd-386.tar.gz	Archive	FreeBSD	x86	64MB	b8065da37783e8b9e7086365a54d74537e832c92311b61101a66989ab2458d8e
go1.22.0.freebsd-amd64.tar.gz	Archive	FreeBSD	x86-64	66MB	50f421c7f217083ac94aab1e09400cb9c2fea7d337679ec11f1638a11460da30
go1.22.0.freebsd-arm64.tar.gz	Archive	FreeBSD	ARM64	63MB	e23385e5c640787fa02cd58f2301ea09e162c4d99f8ca9fa6d52766f428a933d
go1.22.0.freebsd-arm.tar.gz	Archive	FreeBSD	ARMv6	64MB	c9c8b305f90903536f4981bad9f029828c2483b3216ca1783777344fbe603f2d
go1.22.0.freebsd-riscv64.tar.gz	Archive	FreeBSD	riscv64	63MB	c8f94d1de6024546194d58e7b9370dc7ea06176aad94a675b0062c25c40cb645
go1.22.0.illumos-amd64.tar.gz	Archive	illumos	x86-64	66MB	d6792f11ad6ee5fc42d2fe51e1f1683471aa2ee4f20e3ad1be22a4afdbd38e7f
go1.22.0.linux-loong64.tar.gz	Archive	Linux	loong64	64MB	b4b1d425cc113608452a32768469b6e34e538fd072bde9f508a75c8dbbdb843f
go1.22.0.linux-mips.tar.gz	Archive	Linux	mips	62MB	ec0f9194df750c78492c02e4c70ffe6f3864f2511e47c894415320db752719f8
go1.22.0.linux-mips64.tar.gz	Archive	Linux	mips64	61MB	47e938d215e4968ab42afb7307792e5e64184a717e8f176d0de7c411af96d63d
go1.22.0.linux-mips64le.tar.gz	Archive	Linux	mips64le	61MB	c162a1a516b8bc8839fc0c0283ad90f6b511e5aca3da4939ed4800f124b9e72c
go1.22.0.linux-mipsle.tar.gz	Archive	Linux	mipsle	62MB	6ce3e9a06e3a8ca0312dc1f85046b6914c19852eba5393c5cbf26acf698f8977
go1.22.0.linux-ppc64.tar.gz	Archive	Linux	ppc64	63MB	5ae23bf460731eb078f5700b01a50a303308b9f7174a2994892e28bf061f7c85
go1.22.0.linux-ppc64le.tar.gz	Archive	Linux	ppc64le	63MB	0e57f421df9449066f00155ce98a5be93744b3d81b00ee4c2c9b511be2a31d93
go1.22.0.linux-riscv64.tar.gz	Archive	Linux	riscv64	63MB	afe9cedcdbd6fdff27c57efd30aa5ce0f666f471fed5fa96cd4fb38d6b577086
go1.22.0.linux-s390x.tar.gz	Archive	Linux	s390x	65MB	2e546a3583ba7bd3988f8f476245698f6a93dfa9fe206a8ca8f85c1ceecb2446
go1.22.0.netbsd-386.tar.gz	Archive	netbsd	x86	64MB	9b7e2dbd43a169bff18bf533a6c2f462eebe04126ab01c138d0d669c046e1658
go1.22.0.netbsd-amd64.tar.gz	Archive	netbsd	x86-64	66MB	b11995c271d2256dfe85cf54882ca3655e18c49c4d7db0502bff9977767894e7
go1.22.0.netbsd-arm64.tar.gz	Archive	netbsd	ARM64	63MB	499918ccfedde78264d194244d251bf41d95cf606cc0adad13b36b8103cb974f
go1.22.0.netbsd-arm.tar.gz	Archive	netbsd	ARMv6	64MB	b57a3aa9c862300ec0ee8609ce5e0f430f132044f351677fd34711a504081872
go1.22.0.openbsd-386.tar.gz	Archive	openbsd	x86	64MB	379e0829104c18a139d78b33378e6bd3ab2e0792f6c43b4c38e7f20d8d77b99d
go1.22.0.openbsd-amd64.tar.gz	Archive	openbsd	x86-64	66MB	ceb0c97ffc3bfaf74e1df843cb8571d7fc3173a08432f0f42112495df6a31520
go1.22.0.openbsd-arm64.tar.gz	Archive	openbsd	ARM64	63MB	358801cab7122ab50d7d92727644f26d818e9e973403f09e85c2e935a625db75
go1.22.0.openbsd-arm.tar.gz	Archive	openbsd	ARMv6	64MB	8af5aea3df539bc95ed412c0a176fe84baced70ea1dd29f4aa82d0e9ce27fd9f
go1.22.0.plan9-386.tar.gz	Archive	plan9	x86	62MB	fa42c545c9025c45ca9af176dc13a0f4af0cc26bacff6fcb35bb4a170ac538e8
go1.22.0.plan9-amd64.tar.gz	Archive	plan9	x86-64	64MB	d8cf64f37a1dfd8e190c5a303c43ab2d49324868f098d88a3106072d137a5a0b
go1.22.0.plan9-arm.tar.gz	Archive	plan9	ARMv6	62MB	86fd6165f0cbb47ad551094f74b3e5a6c5e09de858d8b99de72d978d41be6e2a
go1.22.0.solaris-amd64.tar.gz	Archive	solaris	x86-64	66MB	a6c12651768d3a74f16104502b4b7bef513ea6b646d99990a28d934c261d1689
go1.22.0.windows-arm64.zip	Archive	Windows	ARM64	69MB	31a61e41d06a3bb2189a303f5f3e777ca4b454eff439f0a67bc2b166330021f4
go1.22.0.windows-arm64.msi	Installer	Windows	ARM64	58MB	c7cdd5f033fabce3bf3215b0ec53688c64bc30955cc544bc65ac14345b88de8d
go1.22.0.windows-arm.zip	Archive	Windows	ARMv6	71MB	495c7dfaea4e2bf48643662bb622e4ce6378d6d9840015238ad4b8792b99ddbf
go1.22.0.windows-arm.msi	Installer	Windows	ARMv6	58MB	d5c9525a3d9302d8c2f614b09279c9df2da0f7c8f54d1a0294c62101e50f40e9
```

WYC-2020, 2024-02-28:
> 它来自官方 https://go.dev/dl/ ，通常能叫得出的平台，它都有。只是我个人需求只有 linux-x64，其它平台并没有添加。然而添加一个宿主机平台比想象中要简单得多，只需要写一个 ARCH信息、SHA256 就完成。 https://github.com/sbwml/packages_lang_golang/blob/22.x/golang/Makefile#L75
>
> 甚至 [...]

如果都是常用平台就可以这样，上面不是l说 他在龙芯上面就会失败，不过大多数都是linux 上面编译吧 所以适用大部分人，这go我是不知道他是如何设计搞的，就不能学学gcc 别人gcc13代码用gcc7 都能编译，都没他这个这么怪

sbwml, 2024-02-28:
> 如果都是常用平台就可以这样，上面不是l说 他在龙芯上面就会失败

go1.22.0.linux-loong64.tar.gz 就是龙芯，go1.19 就原生支持龙芯了，上面的失败单纯是我没加 linux-x64 以外的其它平台而已，毕竟我自己只用linux 环境弄op

WYC-2020, 2024-02-28:
> > 如果都是常用平台就可以这样，上面不是l说 他在龙芯上面就会失败
> 
> go1.22.0.linux-loong64.tar.gz 就是龙芯，go1.19 就原生支持龙芯了，上面的失败单纯是我没加 linux-x64 以外的其它平台而已，毕竟我自己只用linux 环境弄op

那感觉 你这个可能更好如果都比较全的话，可以节省一大笔编译时间 虽然编译也没多久

sKyissKy, 2024-03-04:
> 你这样有个问题 ，我用龙芯的 linux 编译 openwrt 就失败了。。。反而是 从 1.4 编译1.17 ，1.17编译1.20， 1.20才能编译1.22 就好了。。。

啥时候大雕可以用飞腾+银河麒麟编译OP？

mygithub1981, 2024-03-04:
修复好了吗，刚准备编译个新固件？

mygithub1981, 2024-03-05:
> ### 详细叙述
>
>
> touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.v2ray-plugin_installed
> echo "v2ray-plugin" >> [...]

修复了吗？

yanxin152133, 2024-03-07:
[image]

又G了

WYC-2020, 2024-03-07:
就知道问不知道看提交么，告诉你们方法了这种简单替换还要手把手教么

yanxin152133, 2024-03-07:
天天火气那么大干啥呢，修复过能编译的又出问题了不能发出来？还是说这问题以后解决办法都是这样搞？找个完美的方法修复了节省我们大家的时间也节省你们开发者的时间不是嘛。

unkonwuser01, 2024-03-08:
> 改名Makefile 替换golang 同名文件 编译完事 [Makefile.txt](https://github.com/coolsnowwolf/lede/files/14418110/Makefile.txt)

请问大佬，现在是用这个方式修复这个问题吗。  具体是替换的哪个路径下的问题，请指教下。辛苦了。

mygithub1981, 2024-03-08:
> > 改名Makefile 替换golang 同名文件 编译完事 [Makefile.txt](https://github.com/coolsnowwolf/lede/files/14418110/Makefile.txt)
> 
> 请问大佬，现在是用这个方式修复这个问题吗。 具体是替换的哪个路径下的问题，请指教下。辛苦了。

passwall那边修复了。

sKyissKy, 2024-03-14:
先用 sbwml https://github.com/sbwml/packages_lang_golang  替换feeds里面的golang编译 xray 然后再替换回来，不然1.22编译插件好多问题。

King-Of-Knights, 2024-05-09:
在里面加一句

King-Of-Knights, 2024-05-09:
在文件`feeds.conf.default`里面加一句：
`src-git golang https://github.com/kenzok8/golang feeds/packages/lang/golang.git`
更新一下，重新编译就好