coolsnowwolf/lede #11923: feeds install的时候有warning

GreatMichaelLee, 2024-02-28:
### 详细叙述

这几个是什么呀？最近这一两天才出来的，强迫症，怎么resolve?

builder@Build-Server:/home/builder/lede_x86$ ./scripts/feeds install -a
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-video-dma', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-video-mem2mem', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-video-dma', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed helloworld.
Installing all packages from feed passwall.
Installing all packages from feed luci_passwall.
Installing all packages from feed openclash.
Installing all packages from feed istore.
Installing all packages from feed opentopd.
builder@Build-Server:/home/builder/lede_x86$

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

builder@Build-Server:/home/builder/lede_x86$ ./scripts/feeds install -a
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-video-dma', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-video-mem2mem', which does not exist
WARNING: Makefile 'package/kernel/linux/Makefile' has a dependency on 'kmod-video-dma', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed helloworld.
Installing all packages from feed passwall.
Installing all packages from feed luci_passwall.
Installing all packages from feed openclash.
Installing all packages from feed istore.
Installing all packages from feed opentopd.
builder@Build-Server:/home/builder/lede_x86$

WYC-2020, 2024-02-28:
什么玩意 搜索l仓库package/kernel/linux下所有模块 就没那个模块 依赖 你说的这个什么kmod-video-dma，不知道自己有到哪里搞了什么模块 怕不是组合的吧

GreatMichaelLee, 2024-02-28:
[image]
[image]

menuconfig里找的这几个不知道是啥

kxdn3, 2024-02-28:
云编译，同有这个错误

WYC-2020, 2024-02-28:
没遇到你们这些杂七杂八的问题 自己排查

Lujiang0111, 2024-02-28:
强迫症+1
看了一下是 92238a38e65f12bf4a0afd3da0c14976b8c3919a 导致的，切换回 5945bf489fa831f6f990832c27739ad2598a2b34 即可。

WYC-2020, 2024-02-28:
> 强迫症+1 看了一下是 [92238a3](https://github.com/coolsnowwolf/lede/commit/92238a38e65f12bf4a0afd3da0c14976b8c3919a) 导致的，切换回 [5945bf4](https://github.com/coolsnowwolf/lede/commit/5945bf489fa831f6f990832c27739ad2598a2b34) 即可。

我感觉不像，我更新了这个提交 也没你们那个问题 哈哈
[image]
这个我memconfig 都搜不到这玩意，不知道你们如何玩的哈哈

WYC-2020, 2024-02-28:
我的就是这样 
[image]
不知道你们这玩意是怎么来的

WYC-2020, 2024-02-28:
> 强迫症+1 看了一下是 [92238a3](https://github.com/coolsnowwolf/lede/commit/92238a38e65f12bf4a0afd3da0c14976b8c3919a) 导致的，切换回 [5945bf4](https://github.com/coolsnowwolf/lede/commit/5945bf489fa831f6f990832c27739ad2598a2b34) 即可。

看了下确实是这个 
[image]
但是我这边 有合并这个 没报这个 哈哈 见鬼

GreatMichaelLee, 2024-02-28:
> 看了下确实是这个 [...]

你 rm -rf ./tmp && make clean试试

WYC-2020, 2024-02-28:
靠 删除tmp 重新install 就能复现了哈哈 这是缺少这两个模块，等会看下如何休下

fynhnt, 2024-02-28:
> rm -rf ./tmp && make clean

请问老大怎么切换，命令是什么，小白求支援。

fynhnt, 2024-02-28:
> 强迫症+1 看了一下是 [92238a3](https://github.com/coolsnowwolf/lede/commit/92238a38e65f12bf4a0afd3da0c14976b8c3919a) 导致的，切换回 [5945bf4](https://github.com/coolsnowwolf/lede/commit/5945bf489fa831f6f990832c27739ad2598a2b34) 即可。

请问老大怎么切换，命令是什么，小白求支援。

swxk521, 2024-02-28:
> > 强迫症+1 看了一下是 [92238a3](https://github.com/coolsnowwolf/lede/commit/92238a38e65f12bf4a0afd3da0c14976b8c3919a) 导致的，切换回 [5945bf4](https://github.com/coolsnowwolf/lede/commit/5945bf489fa831f6f990832c27739ad2598a2b34) 即可。
> 
> 请问老大怎么切换，命令是什么，小白求支援。

lede/package/kernel/linux/modules/video.mk
这个文件结尾加上下面代码
`define KernelPackage/video-mem2mem
  SUBMENU:=$(VIDEO_MENU)
  TITLE:=Memory 2 Memory device support
  HIDDEN:=1
  DEPENDS:=+kmod-video-videobuf2
  KCONFIG:= \
    CONFIG_V4L_MEM2MEM_DRIVERS=y \
    CONFIG_V4L2_MEM2MEM_DEV
  FILES:= $(LINUX_DIR)/drivers/media/$(V4L2_DIR)/v4l2-mem2mem.ko
  AUTOLOAD:=$(call AutoLoad,66,v4l2-mem2mem)
  $(call AddDepends/video)
endef

define KernelPackage/video-mem2mem/description
  Memory 2 memory device support
endef

$(eval $(call KernelPackage,video-mem2mem))

define KernelPackage/video-dma
  SUBMENU:=$(VIDEO_MENU)
  TITLE:=Video DMA support
  HIDDEN:=1
  DEPENDS:=+kmod-video-videobuf2
  KCONFIG:= \
	CONFIG_VIDEOBUF2_DMA_CONTIG \
	CONFIG_VIDEOBUF2_DMA_SG
  FILES:= $(LINUX_DIR)/drivers/media/common/videobuf2/videobuf2-dma-*.ko
  AUTOLOAD:=$(call AutoLoad,66,videobuf2-dma-contig videobuf2-dma-sg)
  $(call AddDepends/video)
endef

define KernelPackage/video-dma/description
  Video DMA support
endef

$(eval $(call KernelPackage,video-dma))`

Lujiang0111, 2024-02-28:
> > 强迫症+1 看了一下是 [92238a3](https://github.com/coolsnowwolf/lede/commit/92238a38e65f12bf4a0afd3da0c14976b8c3919a) 导致的，切换回 [5945bf4](https://github.com/coolsnowwolf/lede/commit/5945bf489fa831f6f990832c27739ad2598a2b34) 即可。
> 
> 请问老大怎么切换，命令是什么，小白求支援。

```shell
git clone https://github.com/coolsnowwolf/lede
cd lede
git checkout 5945bf489fa831f6f990832c27739ad2598a2b34
```

fynhnt, 2024-02-28:
> ```shell
> git clone https://github.com/coolsnowwolf/lede
> cd lede
> git checkout 5945bf489fa831f6f990832c27739ad2598a2b34
> ```

感谢大佬，马上去试试。

closed 2024-02-29 by coolsnowwolf