coolsnowwolf/lede #11927: CR660X编译出错

Kuron88, 2024-02-29:
### 详细叙述

重新下载了编译还是出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

CR660X

### 详细日志

make[3]: *** [Makefile:91: /home/kuron/lede/build_dir/target-mipsel_24kc_musl/Xray-core-1.8.8/.built] Error 1
make[3]: Leaving directory '/home/kuron/lede/package/passwall-packages/xray-core'
time: package/passwall-packages/xray-core/compile#1.68#1.27#2.99
    ERROR: package/passwall-packages/xray-core failed to build.
make[2]: *** [package/Makefile:116: package/passwall-packages/xray-core/compile] Error 1
make[2]: Leaving directory '/home/kuron/lede'
make[1]: *** [package/Makefile:110: /home/kuron/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kuron/lede'
make: *** [/home/kuron/lede/include/toplevel.mk:231：world] 错误 2

yanxin152133, 2024-03-01:
make dirclean

lmxslpc, 2024-03-01:
是go版本的缘故 要求至少1.22  实际源码中的go是1.21.5  可以自己手动升级go版本

yanxin152133, 2024-03-01:
> make dirclean

[image]

dongfangy2011, 2024-03-10:
我也遇到同样的问题，maike dirclean 还是一样的

yanxin152133, 2024-03-10:
就没修复好，怎么可能编译成功

coolsnowwolf, 2026-09-05:
日志实际失败在 passwall-packages 的 xray-core 1.8.8，并非 CR660X 设备树或 ramips 编译器。Xray/Go 上游已停止为 mipsel 提供该版本所需构建支持，当前 luci-app-passwall 也明确对 mips/mips64 禁止选择 Xray 依赖。应在 MIPS 设备使用 sing-box/其他支持架构的核心，或在 x86/ARM64 构建 Xray；该旧 issue 现关闭。

closed 2026-09-05 by coolsnowwolf