coolsnowwolf/lede #11929: 大佬们，这是什么问题    ERROR: package/feeds/packages/v2dat failed to build.

authon, 2024-03-01:
### 详细叙述

编译出现这个，请问怎么解决

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

 make[3] -C feeds/packages/libs/gperf host-compile
 make[3] -C feeds/packages/utils/v2dat compile
    ERROR: package/feeds/packages/v2dat failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/Ledemaster/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/Ledemaster/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/Ledemaster/openwrt/scripts/config'
make[1]: Entering directory '/Ledemaster/openwrt'
make[2]: Entering directory '/Ledemaster/openwrt'
make[3]: Entering directory '/Ledemaster/openwrt/target/linux'
make[4]: Entering directory '/Ledemaster/openwrt/target/linux/x86'

lmxslpc, 2024-03-01:
让人算命么

authon, 2024-03-01:
> 让人算命么

还需要i什么吗，就是v2dat编译失败

Jejz168, 2024-03-01:
删掉这个v2dat，直接用mosdns中的v2dat

authon, 2024-03-01:
> 删掉这个v2dat，直接用mosdns中的v2dat

没有选择编译mosdns

Jejz168, 2024-03-01:
> > 删掉这个v2dat，直接用mosdns中的v2dat
> 
> 
> 
> 没有选择编译mosdns

CGO_ENABLED=0改为CGO_ENABLED=1

authon, 2024-03-01:
> CGO_ENABLED=0

请问在哪设置

authon, 2024-03-01:
> > > 删掉这个v2dat，直接用mosdns中的v2dat
> > 
> > 
> > 没有选择编译mosdns
> 
> CGO_ENABLED=0改为CGO_ENABLED=1

请问在哪设置

xuexijiaoben, 2024-03-03:
sed -i 's/CGO_ENABLED=0/CGO_ENABLED=1/g' feeds/packages/net/mosdns/Makefile

你试试

authon, 2024-03-03:
> sed -i 's/CGO_ENABLED=0/CGO_ENABLED=1/g' feeds/packages/net/mosdns/Makefile
> 
> 你试试

出现这个

lxg2001, 2024-03-04:
> > sed -i 's/CGO_ENABLED=0/CGO_ENABLED=1/g' feeds/packages/net/mosdns/Makefile
> > 
> > 你试试
> 
> 出现这个
> sed: can't read feeds/packages/net/mosdns/Makefile: No such file or directory

sed -i 's/CGO_ENABLED=0/CGO_ENABLED=1/g' /Ledemaster/openwrt/feeds/packages/net/mosdns/Makefile

mygithub1981, 2024-03-09:
我也一样情况，报错都一模一样，我是选了mosdns的。

woni928, 2024-03-14:
已经解决
sed -i 's#GO_PKG_TARGET_VARS.*# #g' feeds/packages/utils/v2dat/Makefile

windybird, 2024-04-01:
> 已经解决 sed -i 's#GO_PKG_TARGET_VARS.*# #g' feeds/packages/utils/v2dat/Makefile

有这个？我都没找到

closed 2024-07-24 by aiamadeus