coolsnowwolf/lede #11930: -linkmode requires external (cgo) linking, but cgo is not enabled

binge8, 2024-03-01:
### 详细叙述

actions编译报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86，armv8

### 详细日志

```bash
2024-02-29T19:06:54.2527754Z time: package/feeds/luci/luci-app-mjpg-streamer/compile#0.51#0.37#0.83
2024-02-29T19:06:54.2650482Z make[3]: Entering directory '/mnt/openwrt/feeds/packages/net/mosdns'
2024-02-29T19:06:54.4002618Z rm -f /home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1/.built
2024-02-29T19:06:54.4029940Z touch /home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1/.built_check
2024-02-29T19:06:54.4075542Z CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/runner/work/op/op/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/home/runner/work/op/op/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/IrineSistiana/mosdns" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/IrineSistiana/mosdns/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1" GO_BUILD_DIR="/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1/.go_work/build" GO_BUILD_BIN_DIR="/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1/.go_work/build/bin/linux_arm64" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/runner/work/op/op/openwrt/staging_dir/target-aarch64_generic_musl/usr/share/gocode/src" GOOS="linux" GOARCH="arm64" GO386="" GOAMD64="" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1=mosdns-5.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1=mosdns-5.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -znow -zrelro" CGO_ENABLED=0 GOPATH="/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1/.go_work/build" GOCACHE="/home/runner/work/op/op/openwrt/tmp/go-build" GOMODCACHE="/home/runner/work/op/op/openwrt/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash ../../lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1709215206' -linkmode external -extldflags '-L/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags " -X main.version=v5.3.1 -buildid '1709215206' -linkmode external -extldflags '-L/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
2024-02-29T19:06:54.4108888Z Finding targets
2024-02-29T19:06:54.6000319Z 
2024-02-29T19:06:54.6001053Z Building targets
2024-02-29T19:06:54.6521894Z -linkmode requires external (cgo) linking, but cgo is not enabled
2024-02-29T19:06:54.6535714Z 
2024-02-29T19:06:54.6885394Z make[3]: *** [Makefile:48: /home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/mosdns-5.3.1/.built] Error 1
2024-02-29T19:06:54.6886443Z make[3]: Leaving directory '/mnt/openwrt/feeds/packages/net/mosdns'
2024-02-29T19:06:54.6889602Z time: package/feeds/packages/mosdns/compile#0.41#0.27#0.43
2024-02-29T19:06:54.6897249Z     ERROR: package/feeds/packages/mosdns failed to build.
2024-02-29T19:06:54.6900769Z make[2]: *** [package/Makefile:116: package/feeds/packages/mosdns/compile] Error 1
2024-02-29T19:06:54.6905159Z make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
2024-02-29T19:06:54.6911671Z make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
2024-02-29T19:06:54.6915268Z make[1]: Leaving directory '/home/runner/work/op/op/openwrt'
2024-02-29T19:06:54.6923372Z make: *** [/home/runner/work/op/op/openwrt/include/toplevel.mk:231: world] Error 2
```

lxg2001, 2024-03-01:
packages/net/mosdns/Makefile
`GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=0`
改成`GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=1`

binge8, 2024-03-01:
> packages/net/mosdns/Makefile `GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=0` 改成`GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=1`

感谢

closed 2024-03-01 by binge8

mxf524723, 2024-04-24:
改成1了还是同样的错误提示

lpflhh, 2024-04-29:
> packages/net/mosdns/Makefile `GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=0` 改成`GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=1`

packages/net/mosdns/Makefile 请问这个路径是lede目录下吗？找了半天没找到这个

lxg2001, 2024-04-29:
> > packages/net/mosdns/Makefile `GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=0` 改成`GO_PKG_TARGET_VARS:=$(filter-out CGO_ENABLED=%,$(GO_PKG_TARGET_VARS)) CGO_ENABLED=1`
> 
> packages/net/mosdns/Makefile 请问这个路径是lede目录下吗？找了半天没找到这个

lede/feeds/packages/net/mosdns/Makefile
最新版已经修好了，直接把这一行删掉就行