coolsnowwolf/lede #11931: 编译ipq806X系列固件，Turbo ACC插件显示异常。

sxf5487, 2024-03-01:
### 详细叙述

[image]
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

IPQ806X系列

### 详细日志

[image]
[image]

lirg-111, 2024-03-01:
试试：按向下箭头，再按y，看看turbo acc是不是出来了？

sxf5487, 2024-03-01:
> 试试：按向下箭头，再按y，看看turbo acc是不是出来了？

选中Include Flow Offload后，sfe就没有了。

Hirro029, 2024-05-08:
可以选择Include Shortcut-FE CM，这样的话luci-app-turboacc就出来了，反正我的EA8500现在就是这么编译的，但是为什么不能选择Include Shortcut-FE ECM就不知道了，不知道是不是更改了kernel5.10造成的。

FenghenHome, 2024-05-11:
我也一样遇到，暂时不知道怎么解决

SQ528, 2024-07-19:
话说现在ipq806的LEDE可以跑满千兆吗？以前用turboacc过只有40MB/S下载速度就没用了

SQ528, 2024-07-26:
话说你们的ath10k-ct编译正常吗？我这老是错误。
make[2]: Circular package/kernel/ath10k-ct/compile <- package/kernel/mac80211/compile dependency dropped.
make[3]: Entering directory '/workdir/openwrt/package/kernel/ath10k-ct'
make[3]: *** No rule to make target '/workdir/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/backport/autoconf.h', needed by '/workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.configured_68b329da9893e34099c7d8ad5cb9c940'.  Stop.
make[3]: Leaving directory '/workdir/openwrt/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.16#0.06#0.20
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

coolsnowwolf, 2026-09-05:
已核对当前 luci-app-turboacc：配置菜单使用互斥 choice，只能选择一个 FastPath 引擎；选择 `Include Flow Offloading` 后不再同时打包/显示 SFE（`shortcut-fe-cm`），这是当前 Makefile 的明确设计，不是 LuCI 显示异常。当前包还会根据实际模块文件通过 rpcd 动态显示可用引擎，并为 ipq806x/Qualcomm 默认 NSS ECM 路径。该 issue 对应的旧版行为已被现行实现覆盖，关闭为 completed。

closed 2026-09-05 by coolsnowwolf