coolsnowwolf/lede #11939: 新内核中nf_flow_table.h的flow_offload_route_init()定义修改导致内核编译netfilter报错

MilesPoupart, 2024-03-02:
### 详细叙述

如题。
涉及的flow_offload_route_init()函数在内核代码中的改动(返回值由int变为void)：https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/diff/include/net/netfilter/nf_flow_table.h?id=v6.1.80&id2=v6.1.79
导致```target/linux/generic/hack-6.1/650-netfilter-add-xt_FLOWOFFLOAD-target.patch```这一patch报错（可能也涉及其他内核版本，只尝试了6.1编译）
```
net/netfilter/xt_FLOWOFFLOAD.c:524:13: error: void value not ignored as it ought to be
524 |         if (flow_offload_route_init(flow, &route) < 0)
    |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86，可能涉及其他型号

### 详细日志

```
2024-03-02T09:43:27.3220763Z make[4]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.80'
2024-03-02T09:43:27.7898438Z   CALL    scripts/checksyscalls.sh
2024-03-02T09:43:27.8083483Z   DESCEND objtool
2024-03-02T09:43:27.9726749Z   DESCEND bpf/resolve_btfids
2024-03-02T09:43:28.0501615Z   INSTALL libsubcmd_headers
2024-03-02T09:43:28.5663873Z   AR      net/netfilter/built-in.a
2024-03-02T09:43:28.6006878Z   LD [M]  net/netfilter/nf_conntrack.o
2024-03-02T09:43:28.6466501Z   LD [M]  net/netfilter/nf_conntrack_h323.o
2024-03-02T09:43:28.6654829Z   LD [M]  net/netfilter/nf_nat.o
2024-03-02T09:43:28.7279589Z   LD [M]  net/netfilter/nf_tables.o
2024-03-02T09:43:28.7978215Z   LD [M]  net/netfilter/nf_flow_table.o
2024-03-02T09:43:28.8099821Z   CC [M]  net/netfilter/xt_FLOWOFFLOAD.o
2024-03-02T09:43:29.5117801Z net/netfilter/xt_FLOWOFFLOAD.c: In function 'flowoffload_tg':
2024-03-02T09:43:29.5119395Z net/netfilter/xt_FLOWOFFLOAD.c:524:13: error: void value not ignored as it ought to be
2024-03-02T09:43:29.5120638Z   524 |         if (flow_offload_route_init(flow, &route) < 0)
2024-03-02T09:43:29.5126347Z       |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2024-03-02T09:43:29.5401846Z make[7]: *** [scripts/Makefile.build:250: net/netfilter/xt_FLOWOFFLOAD.o] Error 1
2024-03-02T09:43:29.5413242Z make[6]: *** [scripts/Makefile.build:500: net/netfilter] Error 2
2024-03-02T09:43:29.5414628Z make[5]: *** [scripts/Makefile.build:500: net] Error 2
2024-03-02T09:43:29.5415555Z make[4]: *** [Makefile:2010: .] Error 2
2024-03-02T09:43:29.5416962Z make[4]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.80'
2024-03-02T09:43:29.5422149Z make[3]: *** [Makefile:28: /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.80/.modules] Error 2
2024-03-02T09:43:29.5423842Z make[3]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux/x86'
2024-03-02T09:43:29.5424727Z make[2]: *** [Makefile:11: compile] Error 2
2024-03-02T09:43:29.5425679Z make[2]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux'
2024-03-02T09:43:29.5426522Z time: target/linux/compile#3.24#0.48#3.66
2024-03-02T09:43:29.5432118Z     ERROR: target/linux failed to build.
```

Beginner-Go, 2024-03-02:
等一下我看看

Michaol, 2024-03-02:
同错。
make[5]: Entering directory '/home/michaol/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.80'
  CALL    scripts/checksyscalls.sh
  DESCEND objtool
  AR      net/netfilter/built-in.a
  LD [M]  net/netfilter/nf_conntrack.o
  LD [M]  net/netfilter/nf_conntrack_h323.o
  LD [M]  net/netfilter/nf_nat.o
  LD [M]  net/netfilter/nf_tables.o
  LD [M]  net/netfilter/nf_flow_table.o
  CC [M]  net/netfilter/xt_FLOWOFFLOAD.o
net/netfilter/xt_FLOWOFFLOAD.c: In function 'flowoffload_tg':
net/netfilter/xt_FLOWOFFLOAD.c:524:13: error: void value not ignored as it ought to be
  524 |         if (flow_offload_route_init(flow, &route) < 0)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[8]: *** [scripts/Makefile.build:250: net/netfilter/xt_FLOWOFFLOAD.o] Error 1
make[7]: *** [scripts/Makefile.build:500: net/netfilter] Error 2
make[6]: *** [scripts/Makefile.build:500: net] Error 2
make[5]: *** [Makefile:2010: .] Error 2
make[5]: Leaving directory '/home/michaol/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.80'
make[4]: *** [Makefile:28: /home/michaol/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.80/.modules] Error 2
make[4]: Leaving directory '/home/michaol/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/michaol/lede/target/linux'
time: target/linux/compile#2.74#0.82#3.73
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/michaol/lede'
make[1]: *** [target/Makefile:23: /home/michaol/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/michaol/lede'
make: *** [/home/michaol/lede/include/toplevel.mk:231: world] Error 2

MilesPoupart, 2024-03-02:
> 等一下我看看

佬更新内核辛苦了（鞠躬

woni928, 2024-03-03:
CC [M]  net/netfilter/xt_FLOWOFFLOAD.o
net/netfilter/xt_FLOWOFFLOAD.c: In function 'flowoffload_tg':
net/netfilter/xt_FLOWOFFLOAD.c:524:13: error: void value not ignored as it ought to be
  524 |         if (flow_offload_route_init(flow, &route) < 0)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[7]: *** [scripts/Makefile.build:289: net/netfilter/xt_FLOWOFFLOAD.o] Error 1
make[6]: *** [scripts/Makefile.build:552: net/netfilter] Error 2
make[5]: *** [Makefile:1907: net] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.150'
make[4]: *** [Makefile:20: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-5.15.150/.modules] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/armvirt'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#180.89#24.82#205.64
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

目前为止，还是错误。

MilesPoupart, 2024-03-03:
> CC [M] net/netfilter/xt_FLOWOFFLOAD.o net/netfilter/xt_FLOWOFFLOAD.c: In function 'flowoffload_tg': net/netfilter/xt_FLOWOFFLOAD.c:524:13: error: void value not ignored as it ought to be 524 | if [...]

还在改PR 稍等下

oyksoft, 2024-03-03:
打了这个补丁虽然不会报错，但编译出来的固件不稳定，网络断续。还是多检查下吧。

closed 2024-03-03 by aiamadeus

fxvbv, 2024-03-03:
虽然6.1.80内核固件编译成功，但是开机cpu占用约30%，设置界面很卡，过一会就自动重启了，导致我连日志都没看到。

JuiChao, 2024-03-03:
同上，5.10.150内核，不停自动重启

MilesPoupart, 2024-03-03:
[image]
x86 非常正常

commit f903cc89 (2024-03-03): target: generic: fix xt_FLOWOFFLOAD patches

Fixes: #11939, #11943