coolsnowwolf/lede #11950: R5S编译错误

gssiwmd, 2024-03-05:
### 详细叙述

make[5]: Leaving directory '/home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150'
make -C /home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150 KCFLAGS="-fmacro-prefix-map=/home/gswangmd/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/gswangmd/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Mar  5 10:31:57 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/gswangmd/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.150 CC="aarch64-openwrt-linux-musl-gcc" Image dtbs modules
make[5]: Entering directory '/home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
  MODPOST vmlinux.symvers
  MODINFO modules.builtin.modinfo
  GEN     modules.builtin
  LD      .tmp_vmlinux.kallsyms1
  KSYMS   .tmp_vmlinux.kallsyms1.S
  AS      .tmp_vmlinux.kallsyms1.S
  LD      .tmp_vmlinux.kallsyms2
  KSYMS   .tmp_vmlinux.kallsyms2.S
  AS      .tmp_vmlinux.kallsyms2.S
  LD      vmlinux
  SORTTAB vmlinux
  SYSMAP  System.map
  OBJCOPY arch/arm64/boot/Image
  DTC     arch/arm64/boot/dts/rockchip/rk3568-t68m.dtb
Error: arch/arm64/boot/dts/rockchip/rk3568-t68m.dts:570.1-2 syntax error
FATAL ERROR: Unable to parse input tree
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3568-t68m.dtb] Error 1
make[6]: *** [scripts/Makefile.build:552: arch/arm64/boot/dts/rockchip] Error 2
make[5]: *** [Makefile:1450: dtbs] Error 2
make[5]: Leaving directory '/home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150'
make[4]: *** [Makefile:26: /home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150/.modules] Error 2
make[4]: Leaving directory '/home/gswangmd/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/gswangmd/lede/target/linux'
time: target/linux/compile#5.82#1.42#9.81
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [target/Makefile:23: /home/gswangmd/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:231: world] Error 2
gswangmd@wmdhonor:~$ 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

r5s

### 详细日志

make[5]: Leaving directory '/home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150'
make -C /home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150 KCFLAGS="-fmacro-prefix-map=/home/gswangmd/lede/build_dir/target-aarch64_generic_musl=target-aarch64_generic_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/gswangmd/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Mar  5 10:31:57 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/gswangmd/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.15.150 CC="aarch64-openwrt-linux-musl-gcc" Image dtbs modules
make[5]: Entering directory '/home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
  MODPOST vmlinux.symvers
  MODINFO modules.builtin.modinfo
  GEN     modules.builtin
  LD      .tmp_vmlinux.kallsyms1
  KSYMS   .tmp_vmlinux.kallsyms1.S
  AS      .tmp_vmlinux.kallsyms1.S
  LD      .tmp_vmlinux.kallsyms2
  KSYMS   .tmp_vmlinux.kallsyms2.S
  AS      .tmp_vmlinux.kallsyms2.S
  LD      vmlinux
  SORTTAB vmlinux
  SYSMAP  System.map
  OBJCOPY arch/arm64/boot/Image
  DTC     arch/arm64/boot/dts/rockchip/rk3568-t68m.dtb
Error: arch/arm64/boot/dts/rockchip/rk3568-t68m.dts:570.1-2 syntax error
FATAL ERROR: Unable to parse input tree
make[7]: *** [scripts/Makefile.lib:347: arch/arm64/boot/dts/rockchip/rk3568-t68m.dtb] Error 1
make[6]: *** [scripts/Makefile.build:552: arch/arm64/boot/dts/rockchip] Error 2
make[5]: *** [Makefile:1450: dtbs] Error 2
make[5]: Leaving directory '/home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150'
make[4]: *** [Makefile:26: /home/gswangmd/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.150/.modules] Error 2
make[4]: Leaving directory '/home/gswangmd/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/gswangmd/lede/target/linux'
time: target/linux/compile#5.82#1.42#9.81
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/gswangmd/lede'
make[1]: *** [target/Makefile:23: /home/gswangmd/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/gswangmd/lede'
make: *** [/home/gswangmd/lede/include/toplevel.mk:231: world] Error 2
gswangmd@wmdhonor:~$

swxk521, 2024-03-05:
x86编译错误

Ryananana, 2024-03-05:
DTC     arch/arm64/boot/dts/rockchip/rk3566-pinenote-v1.1.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3566-pinenote-v1.2.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3566-quartz64-a.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3566-quartz64-b.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3566-roc-pc.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3566-soquartz-cm4.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-bpi-r2-pro.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-evb1-v10.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-rock-3a.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-photonicat.dtb
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts:492.3-13: Warning (reg_format): /mmc@fe2c0000/sdio_wifi@1:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dtb: Warning (pci_device_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dtb: Warning (pci_device_bus_num): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts:491.14-496.4: Warning (avoid_default_addr_size): /mmc@fe2c0000/sdio_wifi@1: Relying on default #address-cells value
arch/arm64/boot/dts/rockchip/rk3568-photonicat.dts:491.14-496.4: Warning (avoid_default_addr_size): /mmc@fe2c0000/sdio_wifi@1: Relying on default #size-cells value
  DTC     arch/arm64/boot/dts/rockchip/rk3568-radxa-e25.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-mrkaio-m68s.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-mrkaio-m68s-plus.dtb
arch/arm64/boot/dts/rockchip/rk3568-mrkaio-m68s-plus.dts:61.22-63.5: Warning (unit_address_format): /pcie@fe260000/pcie@0,0/pcie@01,0: unit name should not have leading 0s
  DTC     arch/arm64/boot/dts/rockchip/rk3568-nanopi-r5c.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-nanopi-r5s.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-opc-h66k.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-opc-h68k.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-opc-h69k.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-r66s.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-r68s.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-roc-pc.dtb
  DTC     arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dtb
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dts:517.3-13: Warning (reg_format): /usb@fd000000/device@2:reg: property has invalid length (4 bytes) (#address-cells == 2, #size-cells == 1)
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dtb: Warning (pci_device_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dtb: Warning (pci_device_bus_num): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dtb: Warning (i2c_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dtb: Warning (spi_bus_reg): Failed prerequisite 'reg_format'
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dts:515.20-519.4: Warning (avoid_default_addr_size): /usb@fd000000/device@2: Relying on default #address-cells value
arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dts:515.20-519.4: Warning (avoid_default_addr_size): /usb@fd000000/device@2: Relying on default #size-cells value
  DTC     arch/arm64/boot/dts/rockchip/rk3568-t68m.dtb
**Error: arch/arm64/boot/dts/rockchip/rk3568-t68m.dts:570.1-2 syntax error**
FATAL ERROR: Unable to parse input tree
make[7]: *** [scripts/Makefile.lib:406: arch/arm64/boot/dts/rockchip/rk3568-t68m.dtb] Error 1
make[6]: *** [scripts/Makefile.build:500: arch/arm64/boot/dts/rockchip] Error 2
make[5]: *** [Makefile:1474: dtbs] Error 2
make[5]: Leaving directory '/home/morri0on/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.80'
make[4]: *** [Makefile:26: /home/morri0on/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.80/.modules] Error 2
make[4]: Leaving directory '/home/morri0on/lede/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/morri0on/lede/target/linux'
time: target/linux/compile#556.86#47.28#633.38
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/morri0on/lede'
make[1]: *** [target/Makefile:23: /home/morri0on/lede/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/morri0on/lede'
make: *** [/home/morri0on/lede/include/toplevel.mk:231: world] Error 2

同样的问题发生在编译rk3568 on 6.1.80 kernel

Ryananana, 2024-03-05:
commit 2023968 已经解决问题了

WYC-2020, 2024-03-06:
rk3568-t68m.dts 这dts有问题 解析失败了

iHub-2020, 2024-07-08:
已经果断放弃R5S，貌似L大的固件不支持.M2的固态硬盘

closed 2025-01-09 by aiamadeus