coolsnowwolf/lede #11951: hostapd编译报错

binge8, 2024-03-06:
### 详细叙述

armv8编译hostapd报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

armv8

### 详细日志

```bash
2024-03-06T02:44:58.1265640Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /mnt/openwrt/build_dir/target-aarch64_generic_musl/hostapd-supplicant-basic/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_request_scan':
2024-03-06T02:44:58.1267961Z acs.c:(.text+0x488): undefined reference to `hostapd_driver_scan'
2024-03-06T02:44:58.1269440Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x494): undefined reference to `hostapd_driver_scan'
2024-03-06T02:44:58.1272177Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /mnt/openwrt/build_dir/target-aarch64_generic_musl/hostapd-supplicant-basic/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_scan_complete':
2024-03-06T02:44:58.1274020Z acs.c:(.text+0x564): undefined reference to `hostapd_disable_iface'
2024-03-06T02:44:58.1275596Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x570): undefined reference to `hostapd_disable_iface'
2024-03-06T02:44:58.1278744Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /mnt/openwrt/build_dir/target-aarch64_generic_musl/hostapd-supplicant-basic/hostapd-2022-07-29-b704dc72/build/wpa_supplicant/src/ap/acs.o: in function `acs_init':
2024-03-06T02:44:58.1280485Z acs.c:(.text+0x1198): undefined reference to `hostapd_drv_do_acs'
2024-03-06T02:44:58.1281941Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x119c): undefined reference to `hostapd_drv_do_acs'
2024-03-06T02:44:58.1284135Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x11f0): undefined reference to `hostapd_set_state'
2024-03-06T02:44:58.1286307Z /home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: acs.c:(.text+0x11fc): undefined reference to `hostapd_set_state'
2024-03-06T02:44:58.1310828Z collect2: error: ld returned 1 exit status
2024-03-06T02:44:58.1314618Z make[4]: *** [Makefile:2037: wpa_supplicant] Error 1
2024-03-06T02:44:58.1315898Z make[4]: Leaving directory '/mnt/openwrt/build_dir/target-aarch64_generic_musl/hostapd-supplicant-basic/hostapd-2022-07-29-b704dc72/wpa_supplicant'
2024-03-06T02:44:58.1318445Z make[3]: *** [Makefile:790: /home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/hostapd-supplicant-basic/hostapd-2022-07-29-b704dc72/.built] Error 2
2024-03-06T02:44:58.1320480Z make[3]: Leaving directory '/home/runner/work/op/op/openwrt/package/network/services/hostapd'
2024-03-06T02:44:58.1326776Z time: package/network/services/hostapd/supplicant-basic/compile#14.10#1.48#15.13
2024-03-06T02:44:58.1332436Z     ERROR: package/network/services/hostapd failed to build (build variant: supplicant-basic).
2024-03-06T02:44:58.1336835Z make[2]: *** [package/Makefile:119: package/network/services/hostapd/compile] Error 1
2024-03-06T02:44:58.1341469Z make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
2024-03-06T02:44:58.1346487Z make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
```

Firsgith, 2024-03-10:
那只选wpad试试

WYC-2020, 2024-03-11:
[hostapd.tar.gz](https://github.com/coolsnowwolf/lede/files/14553041/hostapd.tar.gz)
用我这个替换试试 路径lede/package/network/services

BenjaminX, 2025-11-13:
X86 同样问题
未选任何无线相关的包

coolsnowwolf, 2026-09-05:
当前 hostapd 已更新至 2024-09-15（commit `5ace39b0a4cdbe18ddbc4e18f80ee3876233c20b`），并保留 `CONFIG_ACS=y` 与 nl80211 驱动配置。已用当前 x86_64 工具链执行 `make package/network/services/hostapd/compile BUILD_VARIANT=supplicant-basic -j30 V=s`，编译和打包成功，生成 `hostapd-common_2024-09-15-5ace39b0-1_x86_64.ipk`，未复现原来的 `hostapd_driver_scan` 等链接错误。该问题已解决。

closed 2026-09-05 by coolsnowwolf