coolsnowwolf/lede #11956: 各位大佬，编译6.1时总会这里报错，哪出问题了

xdfx1180, 2024-03-07:
### 详细叙述

编译最新版passwall时报错，如何更改

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

make[3]: Entering directory '/home/xdfx/lede/feeds/luci/applications/openwrt-passwall-packages/gn'
MAKEFLAGS="" /home/xdfx/lede/staging_dir/host/bin/ninja  -j1 -C /home/xdfx/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/out
ninja: Entering directory `/home/xdfx/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/out'
[1/287] CXX src/base/command_line.o
FAILED: src/base/command_line.o 
g++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++20 -Wno-redundant-move -c ../src/base/command_line.cc -o src/base/command_line.o
g++: error: unrecognized command line option '-std=c++20'; did you mean '-std=c++2a'?
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:69: /home/xdfx/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.built] Error 1
make[3]: Leaving directory '/home/xdfx/lede/feeds/luci/applications/openwrt-passwall-packages/gn'
time: package/feeds/luci/gn/host-compile#0.23#0.09#0.37
    ERROR: package/feeds/luci/gn [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/luci/gn/host/compile] Error 1
make[2]: Leaving directory '/home/xdfx/lede'
make[1]: *** [package/Makefile:110: /home/xdfx/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xdfx/lede'
make: *** [/home/xdfx/lede/include/toplevel.mk:231: world] Error 2

lxg2001, 2024-03-07:
之前已经有人提过了，工具链版本太低

free-diy, 2024-03-08:
> 这个问题之前已经有人提过了#11918

提是提过，但没有说解决方法啊。

hzjnet, 2024-03-09:
> > 这个问题之前已经有人提过了#11918
> 
> 提是提过，但没有说解决方法啊。

贴一下你系统的gcc版本喃？

CYaNu, 2024-03-09:
gcc版本问题

free-diy, 2024-03-09:
> > > 这个问题之前已经有人提过了#11918
> > 
> > 
> > 提是提过，但没有说解决方法啊。
> 
> 贴一下你系统的gcc版本喃？

要gcc哪个版本？

hzjnet, 2024-03-09:
> > > > 这个问题之前已经有人提过了#11918
> > > 
> > > 
> > > 提是提过，但没有说解决方法啊。
> > 
> > 
> > 贴一下你系统的gcc版本喃？
> 
> 要gcc哪个版本？

gcc 11

xdfx1180, 2024-03-09:
> ![image](https://private-user-images.githubusercontent.com/3481363/311426370-4afe089e-5432-43dc-9b33-c223b325a6be.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk5ODk1ODYsIm5iZiI6MTcwOTk4OTI4NiwicGF0aCI6Ii8zNDgxMzYzLzMxMTQyNjM3MC00YWZlMDg5ZS01NDMyLTQzZGMtOWIzMy1jMjIzYjMyNWE2YmUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMDlUMTMwMTI2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MDgzMTY3MjQyZDliYTZhMDA5MTk0ZWJmZWEwMjY2NjZlMGY0MTEwMGM2OTQxN2JjODhmZDAwNmYyYjI4NThkNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.duDc4WC1Zk2hv-tUYZbEw_ZB_QxdYStuzpZLRMWoSfI)

感谢，已成功编译

xdfx1180, 2024-03-09:
> > > > 这个问题之前已经有人提过了#11918
> > > 
> > > 
> > > 提是提过，但没有说解决方法啊。
> > 
> > 
> > 贴一下你系统的gcc版本喃？
> 
> 要gcc哪个版本？

https://www.cnblogs.com/zengjianrong/p/16624273.html

closed 2024-03-09 by xdfx1180