coolsnowwolf/lede #11960: docker网络页面错误

caixin1218, 2024-03-07:
### 详细叙述

/usr/lib/lua/luci/dispatcher.lua:482: Failed to execute cbi dispatcher target for entry '/admin/docker/networks'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/dockerman/networks.lua:44: attempt to index field 'Config' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:482: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:129: in function </usr/lib/lua/luci/dispatcher.lua:128>

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

无

tignioj, 2024-03-08:
同样的错误。

wuyuqq, 2024-03-08:
同样的错误。+1

firkerword, 2024-03-08:
我还以为就我有这个毛病

dph5199278, 2024-03-08:
找上游，上游已解决

caixin1218, 2024-03-09:
> 找上游，上游已解决

刚刚测试过，还没有解决。

dph5199278, 2024-03-09:
> > 找上游，上游已解决
> 
> 刚刚测试过，还没有解决。

上游的这个[提交](https://github.com/lisaac/luci-app-dockerman/commit/7292955a1b415bb60fa2e403bb3a437b4b7f7846)，我试过是可以的

caixin1218, 2024-03-09:
外部分支，只能手工替换？

caixin1218, 2024-03-09:
外部分支，只能手工替换？

HHJLKK, 2024-03-09:
> 外部分支，只能手工替换？ 两个文件都替换了，还不行啊。

自己也是手工替换的，然后重新编译就可以了。
[image]

oyksoft, 2024-03-09:
补丁在我这试了，没用，还是报错。虽然已经合并到了 https://github.com/coolsnowwolf/luci/ 。不知道其它人情况怎么样

shu0802, 2024-03-10:
docker容器启动不了，如何解决。提示如下：
[image]

caixin1218, 2024-03-10:
> 自己也是手工替换的，然后重新编译就可以了。 [...]

docker未启动时网络页面可以打开，但docker启动后网络界面就打不开了。

HHJLKK, 2024-03-10:
> docker未启动时网络页面可以打开，但docker启动后网络界面就打不开了。

修这个bug的新代码已合并，代码重新pull 重新编译。
就没这个问题了

dph5199278, 2024-03-10:
> docker未启动时网络页面可以打开，但docker启动后网络界面就打不开了。

[image]
[image]
[image]

我这边正常，报错信息还是一样的？
不行就全清了，重新再编译一次
```
# 目录全清了
make dirclean
```

wuyuqq, 2024-03-10:
今天编译 docker 网络页面已经可以正常打开显示了。

closed 2024-03-11 by caixin1218