coolsnowwolf/lede #11966: 编译固件选择diskman就会报错

c455025066, 2024-03-09:
### 详细叙述

[image]
选择这个插件就会报错，不选择就可以正常编译，百度和谷歌搜了一圈没找到教程，求问大佬如何解决！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kmod-md-raid456 for luci-app-diskman
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-diskman found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-diskman.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-diskman-zh-cn:
 * 	kmod-md-raid456
 * opkg_install_cmd: Cannot install package luci-i18n-diskman-zh-cn.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/chenxiaoxin/lede'
make[1]: *** [package/Makefile:111: /home/chenxiaoxin/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/chenxiaoxin/lede'
make: *** [/home/chenxiaoxin/lede/include/toplevel.mk:231：world] 错误 2

caixin1218, 2024-03-09:
看不出来什么问题，但我刚刚编译的没有问题
[image]

lxg2001, 2024-03-10:
缺少依赖：kmod-md-raid456

closed 2024-03-10 by c455025066