coolsnowwolf/lede #11968: docker容器启动不了，如何解决。

shu0802, 2024-03-10:
### 详细叙述

docker容器启动不了，如何解决。提示如下：
Containers: start e3df9c6a74fd...code:400 failed to create task for container: failed to create shim task: OCI runtime create failed: unable to retrieve OCI runtime error (open /var/run/docker/containerd/daemon/io.containerd.runtime.v2.task/moby/e3df9c6a74fd355bb19e8080bd8be556c1d9dc44a82fb54c27bea509336ccb55/log.json: no such file or directory): runc did not terminate successfully: exit status 139: unknown

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

3865U软路由

### 详细日志

编译成功后openwrt里的docker启动容器不能正常启动

shu0802, 2024-03-10:
> 是的，docker未启动时网络配置界面可以打开，但docker只要启动了之后网络配置就出不来了，和之前一样的。

我的容器现在是没法启动

WYC-2020, 2024-03-11:
[luci-app-dockerman.tar.gz](https://github.com/coolsnowwolf/lede/files/14552795/luci-app-dockerman.tar.gz)
用我这个试试 我这边 一切正常

WYC-2020, 2024-03-11:
替换重新编译

caixin1218, 2024-03-11:
make distclean  ，
重新编译已经正常
关闭反馈吧。

closed 2024-03-15 by shu0802