coolsnowwolf/lede #11973: 最新编译出错

qq1316237299, 2024-03-11:
### 详细叙述

最新编译出错[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../extras/outbounds/acl/v2geo/v2geo.pb.go:10:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../extras/outbounds/acl/v2geo/v2geo.pb.go:11:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../extras/outbounds/acl/compile.go:11:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../extras/outbounds/acl/matchers.go:7:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../app/internal/utils/qr.go:6:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../extras/obfs/salamander.go:9:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../extras/outbounds/dns_https.go:9:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../extras/outbounds/dns_standard.go:8:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../app/cmd/server.go:17:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../app/cmd/server.go:18:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../app/cmd/client.go:14:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../app/cmd/client.go:15:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../app/cmd/client.go:16:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
../../app/cmd/root.go:12:2: github.com/apernet/quic-go@v0.41.1-0.20240301003057-e18162de481d: Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout

Makefile:59: recipe for target '/home/qq1316237299/lede/build_dir/target-x86_64_musl/hysteria-app-v2.3.0/.built' failed
make[3]: *** [/home/qq1316237299/lede/build_dir/target-x86_64_musl/hysteria-app-v2.3.0/.built] Error 1
make[3]: Leaving directory '/home/qq1316237299/lede/feeds/passwall_packages/hysteria'
time: package/feeds/passwall_packages/hysteria/compile#0.29#0.21#60.35
    ERROR: package/feeds/passwall_packages/hysteria failed to build.
package/Makefile:114: recipe for target 'package/feeds/passwall_packages/hysteria/compile' failed
make[2]: *** [package/feeds/passwall_packages/hysteria/compile] Error 1
make[2]: Leaving directory '/home/qq1316237299/lede'
package/Makefile:110: recipe for target '/home/qq1316237299/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/qq1316237299/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qq1316237299/lede'
/home/qq1316237299/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2
[image]

techblack, 2024-03-11:
网络环境不够科学

qq1316237299, 2024-03-11:
> 网络环境不够科学

[image]

yuhongwei380, 2024-03-11:
机器挂代理去跑；

qq1316237299, 2024-03-11:
> 机器挂代理去跑；

挂了还是没用啊兄弟

qq1316237299, 2024-03-11:
> 1:443: i/o timeout
> ../../extras/outbounds/acl/v2geo/v2geo.pb.go:11:2: github.com/apernet/[quic-go@v0.41.1-0.20240301003057-e18162de481d](mailto:quic-go@v0.41.1-0.20240301003057-e18162de481d): Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
> ../../extras/outbounds/acl/compile.go:11:2: github.com/apernet/[quic-go@v0.41.1-0.20240301003057-e18162de481d](mailto:quic-go@v0.41.1-0.20240301003057-e18162de481d): Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
> ../../extras/outbounds/acl/matchers.go:7:2: github.com/apernet/[quic-go@v0.41.1-0.20240301003057-e18162de481d](mailto:quic-go@v0.41.1-0.20240301003057-e18162de481d): Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
> ../../app/internal/utils/qr.go:6:2: github.com/apernet/[quic-go@v0.41.1-0.20240301003057-e18162de481d](mailto:quic-go@v0.41.1-0.20240301003057-e18162de481d): Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
> ../../extras/obfs/salamander.go:9:2: github.com/apernet/[quic-go@v0.41.1-0.20240301003057-e18162de481d](mailto:quic-go@v0.41.1-0.20240301003057-e18162de481d): Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
> ../../extras/outbounds/dns_https.go:9:2: github.com/apernet/[quic-go@v0.41.1-0.20240301003057-e18162de481d](mailto:quic-go@v0.41.1-0.20240301003057-e18162de481d): Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
> ../../extras/outbounds/dns_standard.go:8:2: github.com/apernet/[quic-go@v0.41.1-0.20240301003057-e18162de481d](mailto:quic-go@v0.41.1-0.20240301003057-e18162de481d): Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
> ../../app/cmd/server.go:17:2: github.com/apernet/[quic-go@v0.41.1-0.20240301003057-e18162de481d](mailto:quic-go@v0.41.1-0.20240301003057-e18162de481d): Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
> ../../app/cmd/server.go:18:2: github.com/apernet/[quic-go@v0.41.1-0.20240301003057-e18162de481d](mailto:quic-go@v0.41.1-0.20240301003057-e18162de481d): Get "https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod": dial tcp 142.250.188.241:443: i/o timeout
> ../../app/cmd/client.go:14:2: github.com/apernet/[quic-go@v0.41.1-0.20240301003057-e18162de481d](mailto:quic-go@v0.41.1-0.20240301003057-e18162de481d): Get "[https://proxy.go](https://proxy.golang.org/github.com/apernet/quic-go/@v/v0.41.1-0.20240301003057-e18162de481d.mod)

[image]
现在是这个错误

yuhongwei380, 2024-03-11:
重新git clone整个项目，重新编译试试看吧，我看你这图里都报缺文件了
[image]

qq1316237299, 2024-03-11:
> 重新git clone整个项目，重新编译试试看吧，我看你这图里都报缺文件了 [...]

哦哦 谢谢

lxg2001, 2024-03-11:
echo "export GO111MODULE=on" >> ~/.profile
echo "export GOPROXY=https://goproxy.cn" >> ~/.profile
source ~/.profile

lxg2001, 2024-03-11:
`echo "export GO111MODULE=on" >> ~/.profile`
`echo "export GOPROXY=https://goproxy.cn" >> ~/.profile`
`source ~/.profile`

qq1316237299, 2024-03-11:
> [lxg2001](/lxg2001)

哇，大佬啊 ，可以编译了

closed 2024-03-13 by aiamadeus