coolsnowwolf/lede #11974: zerotier把配置文件从持久化分区调到var分区后，自定义行星节点不可用力

GreatMichaelLee, 2024-03-11:
### 详细叙述

如题，这个提交导致我自己生产的planet文件没法在var分区上持久化。在/etc/config/zero下可持久化，每次重启服务或者重启路由，都可一直在，放var后都是默认zerotier自己生成的planet给覆盖了。求revert
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

见图
持久化自定义的planet 415字节
[image]
var分区每次重启后变成官方的了
[image]

WYC-2020, 2024-03-11:
放var还想持久化 逗我呢

GreatMichaelLee, 2024-03-11:
所以我说原提交有问题啊

发自我的iPhone

syb999, 2024-03-11:
因为zeoriters会每分钟更新一下networks.d里的网络配置文件。如果配置文件在/etc目录内，对于spi芯片的路由器是灾难级的。长期运行zerotier后，会引发spi芯片报错日志刷屏，同时会出现无法修改配置的问题。

GreatMichaelLee, 2024-03-11:
> 因为zeoriters会每分钟更新一下networks.d里的网络配置文件。如果配置文件在/etc目录内，对于spi芯片的路由器是灾难级的。长期运行zerotier后，会引发spi芯片报错日志刷屏，同时会出现无法修改配置的问题。

那显然这个修改没有很好的把需要持久化的例如自定义的planet或者moon文件，和把诸如networs.d等运行态文件区分开来和merge到一起。持久化配置应该全量体现在/etc/config/zerotier里，init.d脚本里应该把持久化配置通过loadcfg读取例如路径，参数，密钥等连同生成的运行态文件的相关路径一起merge生成完整的目录配置到/var/lib/zerotier下面。

WYC-2020, 2024-03-18:
> > 因为zeoriters会每分钟更新一下networks.d里的网络配置文件。如果配置文件在/etc目录内，对于spi芯片的路由器是灾难级的。长期运行zerotier后，会引发spi芯片报错日志刷屏，同时会出现无法修改配置的问题。
> 
> 那显然这个修改没有很好的把需要持久化的例如自定义的planet或者moon文件，和把诸如networs.d等运行态文件区分开来和merge到一起。持久化配置应该全量体现在/etc/config/zerotier里（这里planet文件属于没爹没娘的），init.d脚本里应该把持久化配置通过loadcfg读取例如路径，参数，密钥等连同生成的运行态文件的相关路径一起merge生成完整的目录配置到/var/lib/zerotier下面。

这个改下呗改成 自动读取 config 产生临时的config到var

GreatMichaelLee, 2024-03-18:
还是得靠大佬，我从来不碰代码:P

WYC-2020, 2024-03-18:
> 还是得靠大佬，我从来不碰代码:P
> […](#)

看了这玩意就是一个软连接嘛 
[image]
，原来那套逻辑没问题，判断是否有这个文件夹 没有进行创建 然后进行软连接  后面那个提交 不管3721直接删除原来配置 每次都是全新生成 自然不行了，反而最后这个提交 读写还更多，我看l这边有人已经撤回了 就用那个撤回的就行

CrazyBoyFeng, 2024-09-20:
这样改是否可行：
将 `init.d` 里 `/var/lib/zerotier-one/` 到 `/etc/config/zero/` 的链接，改为，从 `/etc/config/zero/` 复制到 `/var/lib/zerotier-one/`。  
这样既能实现对存放在 `/etc/config/zero/` 里面的 moon 配置的保留，也能避免对 flash 的频繁读写。  
zerotier 大部分对运行时文件的修改，都是在更新密钥，这其实并没有持久化的必要。考虑到低端机的性能，应避免这样价值较低的对 flash 的频繁读写。

我并不是 moon 的使用者，我不清楚这样修改是否会对 moon 的使用造成什么问题。所以这个方案可能需要 moon 的使用者进行测试。

coolsnowwolf, 2026-09-05:
已修复并推送 packages commit `e296dcf0`。ZeroTier 现在默认将 `config_path` 持久化目录复制到 `/var/lib/zerotier-one`，而不是直接软链接：自定义 `planet`/`moon` 文件会随复制保留，ZeroTier 运行时对 `networks.d`、identity 等频繁写入只落在 RAM，避免覆盖自定义行星节点并减少 flash 写入。两个 init/config 文件均通过 shell/diff 校验。

closed 2026-09-05 by coolsnowwolf