coolsnowwolf/lede #11980: 编译失败

jwkinko, 2024-03-12:
### 详细叙述

编译失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

aarch64

### 详细日志

[image]
[image]

jwkinko, 2024-03-12:
[image]
[image]

jwkinko, 2024-03-12:
还报错，没修好

jwkinko, 2024-03-12:
> 修好了，重新拉源码吧

[image]

hzjnet, 2024-03-13:
这个rust发现错误也搞不了，郁闷

WYC-2020, 2024-03-13:
额 这玩意难道还人平台，我这边刚x86试了没问题，等会有时间试试你们这个平台

hzjnet, 2024-03-13:
刚测试了今天的源码编译ipq806x，还是rust编译错误,具体错误如下：
          /openwrt/openwrt-ub18-cools-M-git-d20240313.feeds/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: error: /openwrt/openwrt-ub18-cools-M-git-d20240313.feeds/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.1-src/build/x86_64-unknown-linux-gnu/stage2-std/armv7-unknown-linux-musleabihf/release/deps/libcompiler_builtins-22577728d5e7d6ac.rlib(compiler_builtins-22577728d5e7d6ac.compiler_builtins.f917b5701cb07aec-cgu.005.rcgu.o) uses VFP register arguments, /openwrt/openwrt-ub18-cools-M-git-d20240313.feeds/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.1-src/build/x86_64-unknown-linux-gnu/stage2-std/armv7-unknown-linux-musleabihf/release/deps/libstd-b324c5b30289a46c.so does not
          /openwrt/openwrt-ub18-cools-M-git-d20240313.feeds/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: failed to merge target specific data of file /openwrt/openwrt-ub18-cools-M-git-d20240313.feeds/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.1-src/build/x86_64-unknown-linux-gnu/stage2-std/armv7-unknown-linux-musleabihf/release/deps/libcompiler_builtins-22577728d5e7d6ac.rlib(compiler_builtins-22577728d5e7d6ac.compiler_builtins.f917b5701cb07aec-cgu.005.rcgu.o)
          collect2: error: ld returned 1 exit status
          

error: could not compile `std` (lib) due to previous error
Building bootstrap
Build completed unsuccessfully in 0:25:08
Makefile:110: recipe for target '/openwrt/openwrt-ub18-cools-M-git-d20240313.feeds/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.1-src/.built' failed
make[3]: *** [/openwrt/openwrt-ub18-cools-M-git-d20240313.feeds/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/host/rustc-1.74.1-src/.built] Error 1
make[3]: Leaving directory '/openwrt/openwrt-ub18-cools-M-git-d20240313.feeds/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#2812.31#95.82#1519.84
    ERROR: package/feeds/packages/rust [host] failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/rust/host/compile' failed
make[2]: Leaving directory '/openwrt/openwrt-ub18-cools-M-git-d20240313.feeds'
package/Makefile:110: recipe for target '/openwrt/openwrt-ub18-cools-M-git-d20240313.feeds/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile' failed
make[1]: Leaving directory '/openwrt/openwrt-ub18-cools-M-git-d20240313.feeds'
/openwrt/openwrt-ub18-cools-M-git-d20240313.feeds/include/toplevel.mk:229: recipe for target 'world' failed

用上面的rust.tar.gz也是以上错误，
对比了https://github.com/coolsnowwolf/lede/files/14583385/rust.tar.gz发现仅有一行差异：
hzjnet@ub18opip18-1:/openwrt/openwrt-ub18-cools-M-git-d20240313.feeds$ diff -r feeds/packages/lang/rust bak/feeds.packages.lang.rust/
diff -r feeds/packages/lang/rust/Makefile bak/feeds.packages.lang.rust/Makefile
21a22
> PKG_BUILD_FLAGS:=no-mips16

jwkinko, 2024-03-14:
还是跟之前一样的错误提示

jwkinko, 2024-03-14:
> [image]
> 我这边刚试了下 好像没问题 哈哈。我的是这个 好像和l的有细微差别 但是感觉不是 这个问题导致的 你试试替换后再编译看看，记得吧原来的全部删除 保活缓存以及编译的 缓存 
> [rust.tar.gz](https://github.com/coolsnowwolf/lede/files/14583385/rust.tar.gz)
> 

大佬看看跟这个有没有关系
https://forum.suricata.io/t/about-suricata-7-compilation-error-on-openwrt-snapshot-master/3978

closed 2024-03-21 by jwkinko