coolsnowwolf/lede #11981: Apple Silicon target编译失败

zhangzhenxi, 2024-03-12:
### 详细叙述

# screen fetch
```shell
                          ./+o+-       fury@opb
                  yyyyy- -yyyyyy+      OS: Ubuntu 20.04 focal
               ://+//////-yyyyyyo      Kernel: x86_64 Linux 6.5.13-orbstack-00122-g57b8027e2387
           .++ .:/++++++/-.+sss/`      Uptime: 1h 44m
         .:++o:  /++++++++/:--:/-      Packages: 796
        o:+o+:++.`..```.-/oo+++++/     Shell: bash 5.0.17
       .:+o:+o/.          `+sssoo+/    Disk: 1.8T / 5.6T (32%)
  .++/+:+oo+o:`             /sssooo.   CPU: 13th Gen Intel Core i7-13700K @ 24x 2GHz
 /+++//+:`oo+o               /::--:.   RAM: 4879MiB / 32092MiB
 \+/+o+++`o++o               ++////.
  .++.o+++oo+:`             /dddhhh.
       .+.o+oo:.          `oddhhhh+
        \+.++o+o``-````.:ohdhhhhh+
         `:o+++ `ohhhhhhhhyo++os:
           .o:`.syhhhhhhh/.oo++o`
               /osyyyyyyo++ooo+++/
                   ````` +oo+++o\:
                          `oo++.
```

# make download 
```shell
make[2]: Entering directory '/home/fury/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/fury/lede/scripts/config'
Makefile:14: .mk: No such file or directory
make[6]: *** No rule to make target '.mk'.  Stop.
make[5]: *** [Makefile:29: image-prereq] Error 2
make[4]: *** [Makefile:11: prereq] Error 2
time: target/linux/prereq#0.04#0.01#0.06
    ERROR: target/linux failed to build.
make[3]: *** [target/Makefile:30: target/linux/prereq] Error 1
make[2]: *** [target/Makefile:22: /home/fury/lede/staging_dir/target-arm_xscale_musl_eabi/stamp/.target_prereq] Error 2
make[1]: *** [/home/fury/lede/include/toplevel.mk:209: prereq] Error 2
make: *** [/home/fury/lede/include/toplevel.mk:229: tools/flock/compile] Error 2
```

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

i7-13700K hackintosh 13.4.1 (22F82)

### 详细日志

日志如上描述所述，我在一台X86的台式机黑苹果上，跑了一个20.04的ubuntu，想要编译出Apple Silicon target下的固件，在 `make download` 这步失败了

zhangzhenxi, 2024-03-21:
@WYC-2020 我已经放弃了 🌚

coolsnowwolf, 2026-09-05:
`.mk: No such file or directory` 来自 `target/linux/silicon/image/Makefile` 的 `include $(SUBTARGET).mk`：日志中的 `staging_dir/target-arm_xscale_musl_eabi` 表明目标配置实际落到了 IXP4xx/xscale，且没有选择 silicon 的 `armv8` 子目标，导致 `SUBTARGET` 为空。Apple Silicon 必须先在 menuconfig 选择 Target System=Apple Silicon、Subtarget=armv8（或使用对应 defconfig），再执行 `make download`。当前 silicon target 已存在且 KERNEL_PATCHVER=6.1；无需包修复。该旧配置问题现关闭。

closed 2026-09-05 by coolsnowwolf