coolsnowwolf/lede #11983: 编译失败

GreatMichaelLee, 2024-03-12:
### 详细叙述

见日志

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.21/crypto/aead.ko' is built-in.
Package kmod-crypto-aead is missing dependencies for the following libraries:
rng.ko
make[3]: *** [modules/crypto.mk:51: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-aead_6.6.21-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux'
time: package/kernel/linux/compile#1.66#0.59#2.31
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1

kxdn3, 2024-03-12:
+1
@Beginner-Go

Beginner-Go, 2024-03-12:
`kmod-crypto-aead` 这个模块编译错误

mygithub1981, 2024-03-13:
> `kmod-crypto-aead` 这个模块编译错误
同样编译错误，求修复

River888888, 2024-03-13:
同样编译错误
Applying /home/ubuntu/lede/target/linux/generic/pending-6.6/141-jffs2-add-RENAME_EXCHANGE-support.patch using plaintext: 
patching file fs/jffs2/dir.c
Hunk #1 FAILED at 791.
Hunk #2 succeeded at 820 (offset -17 lines).
Hunk #3 FAILED at 863.
Hunk #4 succeeded at 873 (offset -22 lines).
2 out of 4 hunks FAILED -- saving rejects to file fs/jffs2/dir.c.rej
Patch failed!  Please fix /home/ubuntu/lede/target/linux/generic/pending-6.6/141-jffs2-add-RENAME_EXCHANGE-support.patch!
make[4]: *** [Makefile:34: /home/ubuntu/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.21/.prepared_5376cff70f474e27f84a320bd577db9f] Error 1
make[4]: Leaving directory '/home/ubuntu/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/ubuntu/lede/target/linux'
time: target/linux/compile#1.09#15.18#19.99
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/ubuntu/lede'
make[1]: *** [target/Makefile:23: /home/ubuntu/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/ubuntu/lede'
make: *** [/home/ubuntu/lede/include/toplevel.mk:231：world] 错误 2

GreatMichaelLee, 2024-03-13:
> 同样编译错误 Applying /home/ubuntu/lede/target/linux/generic/pending-6.6/141-jffs2-add-RENAME_EXCHANGE-support.patch using plaintext: patching file fs/jffs2/dir.c Hunk #1 FAILED at 791. Hunk #2 succeeded [...]

你这个显然跟我这个不是一个问题.....

GreatMichaelLee, 2024-03-13:
> `kmod-crypto-aead` 这个模块编译错误

求大佬赶紧出手为谢~

LIUBANGJIAN, 2024-03-13:
2024-03-12T22:48:56.0792588Z touch /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-firmware-20231211/.pkgdir/i915-firmware.installed
2024-03-12T22:48:56.0817406Z mkdir -p /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp
2024-03-12T22:48:56.0848071Z SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-firmware-20231211/.pkgdir/amdgpu-firmware/. /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/'
2024-03-12T22:48:56.2013971Z touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.amdgpu-firmware_installed
2024-03-12T22:48:56.2107561Z echo "amdgpu-firmware" >> /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
2024-03-12T22:48:56.2125988Z mkdir -p /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp
2024-03-12T22:48:56.2153923Z SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-firmware-20231211/.pkgdir/bnx2-firmware/. /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/'
2024-03-12T22:48:56.2200121Z touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.bnx2-firmware_installed
2024-03-12T22:48:56.2222713Z echo "bnx2-firmware" >> /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
2024-03-12T22:48:56.2241099Z mkdir -p /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp
2024-03-12T22:48:56.2269081Z SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-firmware-20231211/.pkgdir/bnx2x-firmware/. /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/'
2024-03-12T22:48:56.2345309Z touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.bnx2x-firmware_installed
2024-03-12T22:48:56.2368835Z echo "bnx2x-firmware" >> /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
2024-03-12T22:48:56.2386794Z mkdir -p /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp
2024-03-12T22:48:56.2414684Z SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-firmware-20231211/.pkgdir/i915-firmware/. /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/'
2024-03-12T22:48:56.2711423Z touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.i915-firmware_installed
2024-03-12T22:48:56.2734851Z echo "i915-firmware" >> /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/pkginfo/linux-firmware.default.install
2024-03-12T22:48:56.2753491Z make[3]: Leaving directory '/home/runner/work/lede/lede/package/firmware/linux-firmware'
2024-03-12T22:48:56.2759295Z time: package/firmware/linux-firmware/compile#19.50#5.06#22.40
2024-03-12T22:48:56.2874882Z make[3]: Entering directory '/home/runner/work/lede/lede/package/kernel/linux'
2024-03-12T22:48:59.5074388Z mkdir -p /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp
2024-03-12T22:48:59.5075218Z
2024-03-12T22:48:59.5077163Z SHELL= flock /home/runner/work/lede/lede/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/'
2024-03-12T22:48:59.5079569Z touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
2024-03-12T22:48:59.5161490Z NOTICE: module '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.21/crypto/aead.ko' is built-in.
2024-03-12T22:48:59.5170032Z ERROR: module '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.21/crypto/geniv.ko' is missing.
2024-03-12T22:48:59.5171616Z make[3]: *** [modules/crypto.mk:51: /home/runner/work/lede/lede/bin/targets/x86/64/packages/kmod-crypto-aead_6.6.21-1_x86_64.ipk] Error 1
2024-03-12T22:48:59.5172619Z make[3]: Leaving directory '/home/runner/work/lede/lede/package/kernel/linux'
2024-03-12T22:48:59.5184315Z time: package/kernel/linux/compile#2.90#0.34#3.23
2024-03-12T22:48:59.5191319Z ERROR: package/kernel/linux failed to build.
2024-03-12T22:48:59.5194900Z make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
2024-03-12T22:48:59.5199836Z make[2]: Leaving directory '/home/runner/work/lede/lede'
2024-03-12T22:48:59.5225867Z make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-03-12T22:48:59.5232268Z make[1]: Leaving directory '/home/runner/work/lede/lede'
2024-03-12T22:48:59.5263095Z make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 2
2024-03-12T22:48:59.5278749Z ##[error]Process completed with exit code 2.

一样的错误

WYC-2020, 2024-03-13:
一天天的

LIUBANGJIAN, 2024-03-13:
> 一天天的

快两天了。

WYC-2020, 2024-03-13:
> > 一天天的
> 
> 快两天了。

刚不是修复了么，我的没问题因为我很早就修了

GreatMichaelLee, 2024-03-13:
> > > 一天天的
> > 
> > 
> > 快两天了。
> 
> 刚不是修复了么，我的没问题因为我很早就修了

拉了这个58147ef0f6c045ede4e1793f3305110a6f84a1bb还是不行，make clean后还是报一样的错误

GreatMichaelLee, 2024-03-13:
> > > > 一天天的
> > > 
> > > 
> > > 快两天了。
> > 
> > 
> > 刚不是修复了么，我的没问题因为我很早就修了
> 
> 拉了这个[58147ef](https://github.com/coolsnowwolf/lede/commit/58147ef0f6c045ede4e1793f3305110a6f84a1bb) 还是不行，make clean后还是报一样的错误

哦，才发现错误变了，不再是aead了，变成了user

SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-crypto-user is missing dependencies for the following libraries:
rng.ko
make[3]: *** [modules/crypto.mk:1118: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-user_6.6.21-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux'
time: package/kernel/linux/compile#2.14#0.44#2.57
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2024-03-13:
> > > > > 一天天的
> > > > 
> > > > 
> > > > 快两天了。
> > > 
> > > 
> > > 刚不是修复了么，我的没问题因为我很早就修了
> > 
> > 
> > 拉了这个[58147ef](https://github.com/coolsnowwolf/lede/commit/58147ef0f6c045ede4e1793f3305110a6f84a1bb) 还是不行，make clean后还是报一样的错误
> 
> 哦，才发现错误变了，不再是aead了，变成了user,这些个鬼ko的dependency真的是很恼火。
> 
> SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/' touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed Package kmod-crypto-user is missing dependencies for the following libraries: rng.ko make[3]: *** [modules/crypto.mk:1118: /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-user_6.6.21-1_x86_64.ipk] Error 1 make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/linux' time: package/kernel/linux/compile#2.14#0.44#2.57 ERROR: package/kernel/linux failed to build. make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1 make[2]: Leaving directory '/home/builder/lede_x86' make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2 make[1]: Leaving directory '/home/builder/lede_x86' make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2

更新后重新编译过内核没 ，先make menuconfig 然后直接保存，然后 make target/linux/compile V=s -j4  在make package/kernel/linux/compile V=s 试试，因为我这个和l 仓库这边修复的不一样 没法给你 你先这样试试，不行我在把我的给你 你直接替换 编译看看

GreatMichaelLee, 2024-03-13:
> 更新后重新编译过内核没 ，先make menuconfig 然后直接保存，然后 make target/linux/compile V=s -j4 在make package/kernel/linux/compile V=s 试试，因为我这个和l 仓库这边修复的不一样 没法给你 你先这样试试，不行我在把我的给你 你直接替换 编译看看

改了一下这个，我上面说的那个user的没了，但特么又出新的错误了，简直要疯了
[image]

新的鬼
[image]

WYC-2020, 2024-03-13:
> 改了一下这个，我上面说的那个user的没了，但特么又出新的错误了，简直要疯了 [...]

最后一个提交出的问题 
[image]
改回去试试 把这个模块,我最近的四个commit 都还没合 所以我是基于我这边来看的 目前没任何问题，你们说的这几个 我都遇不到

kxdn3, 2024-03-13:
同样错误

GreatMichaelLee, 2024-03-13:
> 最后一个提交出的问题 [...]

按你说的，改回原来的，这处好了，但又出新问题了。。。。
[image]

新错误。。。。疯了疯了

[image]

WYC-2020, 2024-03-13:
什么玩意 我编译node 没这个问题，看这个错误是 
[image]
这个 
[image]
这个没有定义 让你包含头文件 cstdint,这个要打patch 我这边没问题 所以这个你自己修了

GreatMichaelLee, 2024-03-13:
> 什么玩意 我编译node 没这个问题，看这个错误是 [...]

小白伸手党，求大佬们搞定^^

swxk521, 2024-03-13:
Package kmod-drm-i915 is missing dependencies for the following libraries:
drm_buddy.ko
make[3]: *** [modules/video.mk:1343: /workdir/openwrt/bin/targets/x86/64/packages/kmod-drm-i915_6.1.81-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/package/kernel/linux'
time: package/kernel/linux/compile#2.91#0.36#3.25
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

LIUBANGJIAN, 2024-03-13:
touch /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Package kmod-crypto-user is missing dependencies for the following libraries:
rng.ko
make[3]: *** [modules/crypto.mk:1118: /home/runner/work/lede/lede/bin/targets/x86/64/packages/kmod-crypto-user_6.6.21-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/runner/work/lede/lede/package/kernel/linux'
time: package/kernel/linux/compile#3.00#0.37#3.34
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/runner/work/lede/lede'
make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/lede/lede'
make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2024-03-13:
算了 我吧我这个给你们 你们替换再试试看 
[crypto.mk.txt](https://github.com/coolsnowwolf/lede/files/14583416/crypto.mk.txt)

mygithub1981, 2024-03-13:
> 哦，才发现错误变了，不再是aead了，变成了user,这些个鬼ko的dependency真的是很恼火。
>
> SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR [...]

一样的问题，求解决！

WYC-2020, 2024-03-13:
算了 我合并到最新 在来看看 是否有这些错误  因为我是 基于最新回退的四个提交 所有没有遇到你们这些 我合并了 在编译看看 是否有这些问题

WYC-2020, 2024-03-13:
> > 算了 我合并到最新 在来看看 是否有这些错误 因为我是 基于最新回退的四个提交 所有没有遇到你们这些 我合并了 在编译看看 是否有这些问题
> 
> 是12号的提交更新后出现的问题

那肯定了 
[image]
 这个提交算是大更新了，很多patch是删除了 重新加了很多patch  所以我这边都是 合并到这个提交之前的版本，这个提交之后的 我一直没合，等下我合下 看下会不会遇到你们这些问题

GreatMichaelLee, 2024-03-13:
> 那肯定了 [...]

其实我最想要的更新是那个6.8内核里的网络tcp性能优化的移植。据说tcp性能优化了40%，估计有些吹。

kxdn3, 2024-03-13:
> 其实我最想要的更新是那个6.8内核里的网络tcp性能优化的移植。据说tcp性能优化了40%，估计有些吹。

肯定是吹牛

boloc, 2024-03-13:
> 最新代码 github action出错 软件包 kmod-drm-i915 缺少以下库的依赖项： drm_buddy.ko make[3]： *** [modules/video.mk：1343： /workdir/openwrt/bin/targets/x86/64/packages/kmod-drm-i915_6.1.81-1_x86_64.ipk] 错误 1 make[3]： 离开目录 '/workdir/openwrt/package/kernel/linux' 时间： package/kernel/linux/compile#2.91#0.36#3.25 错误： package/kernel/linux 无法构建。 make[2]： *** [package/Makefile：116： package/kernel/linux/compile] 错误 1 make[2]： 离开目录 '/workdir/openwrt' make[1]： *** [package/Makefile：110： /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] 错误 2 make[1]： 离开目录 '/workdir/openwrt' make： *** [/workdir/openwrt/include/toplevel.mk：231： world] 错误 2 错误： 进程完成，退出代码 2.

同出现了这个问题，有大老知道什么情况吗

GreatMichaelLee, 2024-03-13:
前面两个问题刚才的提交修复了，这个头文件的问题请l大也一起看看 @coolsnowwolf 

```
tools/v8_gypfiles/v8_base_without_compiler.host.mk:26: warning: ignoring old recipe for target '/home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/obj/gen/inspector-generated-output-root/include/inspector/Schema.h'
  g++  -o /home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/obj.host/v8_libbase/deps/v8/src/base/cpu.o ../deps/v8/src/base/cpu.cc '-DNODE_OPENSSL_CONF_NAME=nodejs_conf' '-DICU_NO_USER_DATA_OVERRIDE' '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DV8_TARGET_ARCH_X64' '-DV8_EMBEDDER_STRING="-node.88"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DOBJECT_PRINT' '-DV8_CONCURRENT_MARKING' '-DV8_ARRAY_BUFFER_EXTENSION' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_SNAPSHOT_COMPRESSION' '-DBUILDING_V8_BASE_SHARED' -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include -I../deps/v8 -I../deps/v8/include  -pthread -Wno-unused-parameter -m64 -Wno-return-type -fno-strict-aliasing -m64 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/.deps//home/builder/lede_x86/build_dir/target-x86_64_musl/node-v14.21.3/out/Release/obj.host/v8_libbase/deps/v8/src/base/cpu.o.d.raw   -c
In file included from ../deps/v8/src/base/macros.h:12,
                 from ../deps/v8/src/base/cpu.h:17,
                 from ../deps/v8/src/base/cpu.cc:5:
../deps/v8/src/base/logging.h: In function 'typename std::enable_if<(std::is_enum<_Tp>::value && (! v8::base::has_output_operator<T>::value)), std::__cxx11::basic_string<char> >::type v8::base::PrintCheckOperand(T)':
../deps/v8/src/base/logging.h:170:34: error: 'uint8_t' was not declared in this scope
  170 |       std::is_same<underlying_t, uint8_t>::value, uint16_t,
      |                                  ^~~~~~~
../deps/v8/src/base/logging.h:16:1: note: 'uint8_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
   15 | #include "src/base/template-utils.h"
  +++ |+#include <cstdint>
   16 |
../deps/v8/src/base/logging.h:170:41: error: template argument 2 is invalid
  170 |       std::is_same<underlying_t, uint8_t>::value, uint16_t,
      |                                         ^
../deps/v8/src/base/logging.h:170:51: error: 'uint16_t' was not declared in this scope
  170 |       std::is_same<underlying_t, uint8_t>::value, uint16_t,
      |                                                   ^~~~~~~~
../deps/v8/src/base/logging.h:170:51: note: 'uint16_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?
../deps/v8/src/base/logging.h:172:61: error: template argument 1 is invalid
  172 |                                 int16_t, underlying_t>::type>::type;
      |                                                             ^
../deps/v8/src/base/logging.h:172:61: error: template argument 2 is invalid
../deps/v8/src/base/logging.h:172:62: error: '<declaration error>' is not a template [-fpermissive]
  172 |                                 int16_t, underlying_t>::type>::type;
      |                                                              ^~
../deps/v8/src/base/logging.h:173:40: error: 'int_t' does not name a type; did you mean 'wint_t'?
  173 |   return PrintCheckOperand(static_cast<int_t>(static_cast<underlying_t>(val)));
      |                                        ^~~~~
      |                                        wint_t
../deps/v8/src/base/macros.h: At global scope:
../deps/v8/src/base/macros.h:338:25: error: 'intptr_t' has not been declared
  338 | inline T RoundDown(T x, intptr_t m) {
      |                         ^~~~~~~~
../deps/v8/src/base/macros.h:344:11: error: 'intptr_t' has not been declared
  344 | template <intptr_t m, typename T>
      |           ^~~~~~~~
../deps/v8/src/base/macros.h: In function 'constexpr T RoundDown(T)':
../deps/v8/src/base/macros.h:348:17: error: 'm' was not declared in this scope
  348 |   STATIC_ASSERT(m != 0 && ((m & (m - 1)) == 0));
      |                 ^
../deps/v8/src/base/macros.h:200:43: note: in definition of macro 'STATIC_ASSERT'
  200 | #define STATIC_ASSERT(test) static_assert(test, #test)
      |                                           ^~~~
../deps/v8/src/base/macros.h: At global scope:
../deps/v8/src/base/macros.h:354:23: error: 'intptr_t' has not been declared
  354 | inline T RoundUp(T x, intptr_t m) {
      |                       ^~~~~~~~
../deps/v8/src/base/macros.h:358:11: error: 'intptr_t' has not been declared
  358 | template <intptr_t m, typename T>
      |           ^~~~~~~~
../deps/v8/src/base/macros.h: In function 'constexpr T RoundUp(T)':
../deps/v8/src/base/macros.h:361:20: error: 'm' was not declared in this scope
  361 |   return RoundDown<m, T>(static_cast<T>(x + (m - 1)));
      |                    ^
../deps/v8/src/base/macros.h: In function 'void* AlignedAddress(void*, size_t)':
../deps/v8/src/base/macros.h:372:51: error: 'uintptr_t' does not name a type
  372 |   return reinterpret_cast<void*>(reinterpret_cast<uintptr_t>(address) &
      |                                                   ^~~~~~~~~
[... cut ...]
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#0.28#0.48#0.95
    ERROR: package/feeds/packages/node failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/node/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
```

GreatMichaelLee, 2024-03-13:
非常奇怪，上面这个问题，我把vmware的虚拟机重新还原到早前的一个快照，然后再重新pull了一把，居然就编译过了，完全不知道发生了什么，但如果不还原，哪怕直接make clean都是报上面这个错误怎么着也过不了的，git status也确定没有touch过哪里。之前我也遇到出现过这个现象一次，也是只能靠还原快照再拉代码编译一次。

LIUBANGJIAN, 2024-03-14:
2024-03-13T23:08:42.6602619Z make[3]: Leaving directory '/home/runner/work/lede/lede/package/qca/shortcut-fe/shortcut-fe'
2024-03-13T23:08:42.6603799Z time: package/qca/shortcut-fe/shortcut-fe/compile#2.56#0.43#2.98
2024-03-13T23:08:42.6611859Z     ERROR: package/qca/shortcut-fe/shortcut-fe failed to build.
2024-03-13T23:08:42.6615167Z make[2]: *** [package/Makefile:116: package/qca/shortcut-fe/shortcut-fe/compile] Error 1
2024-03-13T23:08:42.6619662Z make[2]: Leaving directory '/home/runner/work/lede/lede'
2024-03-13T23:08:42.6628550Z make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-03-13T23:08:42.6630101Z make[1]: Leaving directory '/home/runner/work/lede/lede'
2024-03-13T23:08:42.6636419Z make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 2
2024-03-13T23:08:42.6657393Z ##[error]Process completed with exit code 2.
2024-03-13T23:08:42.6780837Z Post job cleanup.
2024-03-13T23:08:42.7571796Z [command]/usr/bin/git version
2024-03-13T23:08:42.7610395Z git version 2.43.2
2024-03-13T23:08:42.7661005Z Temporarily overriding HOME='/home/runner/work/_temp/34819fdf-f9f6-4ffb-96d9-9ed105e19ac8' before making global git config changes
2024-03-13T23:08:42.7662770Z Adding repository directory to the temporary git global config as a safe directory
2024-03-13T23:08:42.7666506Z [command]/usr/bin/git config --global --add safe.directory /home/runner/work/lede/lede
2024-03-13T23:08:42.7699025Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2024-03-13T23:08:42.7729997Z [command]/usr/bin/git submodule foreach --recursive sh -c "git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :"
2024-03-13T23:08:42.7985040Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2024-03-13T23:08:42.8005159Z http.https://github.com/.extraheader
2024-03-13T23:08:42.8015838Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2024-03-13T23:08:42.8043887Z [command]/usr/bin/git submodule foreach --recursive sh -c "git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :"
2024-03-13T23:08:42.8424363Z Post job cleanup.
2024-03-13T23:08:42.9193949Z [command]/usr/bin/git version
2024-03-13T23:08:42.9233093Z git version 2.43.2
2024-03-13T23:08:42.9274596Z Temporarily overriding HOME='/home/runner/work/_temp/56679731-1067-4853-90b6-534d8e35bc04' before making global git config changes
2024-03-13T23:08:42.9276184Z Adding repository directory to the temporary git global config as a safe directory
2024-03-13T23:08:42.9279194Z [command]/usr/bin/git config --global --add safe.directory /home/runner/work/lede/lede
2024-03-13T23:08:42.9311990Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2024-03-13T23:08:42.9342973Z [command]/usr/bin/git submodule foreach --recursive sh -c "git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :"
2024-03-13T23:08:42.9593900Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2024-03-13T23:08:42.9626329Z [command]/usr/bin/git submodule foreac

LIUBANGJIAN, 2024-03-14:
2024-03-14T03:14:57.7269089Z true
2024-03-14T03:14:57.7287674Z [ ! -d ./src/ ] || cp -fpR ./src/. /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe
2024-03-14T03:14:57.7352858Z touch /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.prepared_f9a38780b22ae5ad347cf83d8fceb9d8_6664517399ebbbc92a37c5bb081b5c53
2024-03-14T03:14:57.7376497Z rm -f /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.configured_*
2024-03-14T03:14:57.7403165Z rm -f /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.shortcut-fe_installed
2024-03-14T03:14:57.7462312Z (cd /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/./; if [ -x ./configure ]; then find /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/ -name config.guess | xargs -r chmod u+w; find /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/ -name config.guess | xargs -r -n1 cp --remove-destination /home/runner/work/lede/lede/scripts/config.guess; find /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/ -name config.sub | xargs -r chmod u+w; find /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/ -name config.sub | xargs -r -n1 cp --remove-destination /home/runner/work/lede/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe=shortcut-fe -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe=shortcut-fe -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe=shortcut-fe -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/runner/work/lede/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/runner/work/lede/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/runner/work/lede/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/runner/work/lede/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/runner/work/lede/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
2024-03-14T03:14:57.7476607Z touch /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.configured_68b329da9893e34099c7d8ad5cb9c940
2024-03-14T03:14:57.7494696Z rm -f /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built
2024-03-14T03:14:57.7521092Z touch /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built_check
2024-03-14T03:14:57.7550580Z make -j1 -C "/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.21" KCFLAGS="-fmacro-prefix-map=/home/runner/work/lede/lede/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/lede/lede/staging_dir/host/include -I/home/runner/work/lede/lede/staging_dir/hostpkg/include -I/home/runner/work/lede/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Mar 14 01:29:14 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/runner/work/lede/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/fullconenat.symvers" KERNELRELEASE=6.6.21  M="/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe" EXTRA_CFLAGS+="-DSFE_SUPPORT_IPV6" SFE_SUPPORT_IPV6=y modules
2024-03-14T03:14:57.7574871Z make[4]: Entering directory '/home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.21'
2024-03-14T03:14:58.1193699Z   CC [M]  /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.o
[... cut ...]
2024-03-14T03:15:00.6273042Z make[2]: Leaving directory '/home/runner/work/lede/lede'
2024-03-14T03:15:00.6279171Z make[1]: *** [package/Makefile:110: /home/runner/work/lede/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-03-14T03:15:00.6282973Z make[1]: Leaving directory '/home/runner/work/lede/lede'
2024-03-14T03:15:00.6292736Z make: *** [/home/runner/work/lede/lede/include/toplevel.mk:231: world] Error 2
2024-03-14T03:15:00.6311991Z ##[error]Process completed with exit code 2.

WYC-2020, 2024-03-14:
> 2024-03-14T03:14:57.7269089Z true 2024-03-14T03:14:57.7287674Z [ ! -d ./src/ ] || cp -fpR ./src/. /home/runner/work/lede/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe [...]

sfe而已l已经加回来了 更新编译

closed 2024-05-28 by aiamadeus