coolsnowwolf/lede #11985: 编译错误，gn-2024-01-24-a2e2717e.tar.xz无法下载

grassrecluse, 2024-03-13:
### 详细叙述

源里面没有相应的文件可供下载

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

WRT1900AC v2

### 详细日志

+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/gn-2024-01-24-a2e2717e.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/gn-2024-01-24-a2e2717e.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/gn-2024-01-24-a2e2717e.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.

WYC-2020, 2024-03-13:
用passwall 下面那个gn 啊，你这个是哪里的哦 这个url下面早就没有gn了吧

grassrecluse, 2024-03-13:
> [gn.tar.gz](https://github.com/coolsnowwolf/lede/files/14581927/gn.tar.gz) 用这个把原来的删除掉或者直接替换 用这个

Error 404 Not Found
Not Found

Error 54113
Details: cache-bur-kbur8200159-BUR 1710304335 1060980653

Varnish cache server

WYC-2020, 2024-03-13:
[gn.tar.gz](https://github.com/coolsnowwolf/lede/files/14582433/gn.tar.gz)

WYC-2020, 2024-03-13:
这什么玩意 github 抽风了

WYC-2020, 2024-03-13:
https://github.com/xiaorouji/openwrt-passwall-packages/tree/main/gn  自己去这里

closed 2024-05-02 by aiamadeus