coolsnowwolf/lede #11986: 反复编译失败 那位大神帮忙看看，感激不尽

shedkiss, 2024-03-13:
### 详细叙述

从错误提示看是Cargo.lock反复校验失败，不知为何。
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64

### 详细日志

 the listed checksum of `/root/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/vendor/icu_locid/Cargo.lock` has changed:
expected: 1aa9f21333d579ccbdf71478538b586518faa9d7f73f7ba8049649f5d4e33b43
actual:   568edd2ce05aa3cf0e0ee8487a5d328aca123897b945c431a9e58a87bf986f5e

directory sources are not intended to be edited, if modifications are required then it is recommended that `[patch]` is used with a forked copy of the source
Build completed unsuccessfully in 0:00:01
make[3]: *** [Makefile:111: /root/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/.built] Error 1
make[3]: Leaving directory '/root/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#0.92#0.35#1.61
    ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/root/lede'
make[1]: *** [package/Makefile:110: /root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'
make: *** [/root/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2024-03-13:
这玩意 每个人可能楼梯不一样 下载的地方不一样 这个校验就可能会失效 自己吧makefile 里面那个 hash改成skip 跳过检验

shedkiss, 2024-03-13:
> 这玩意 每个人可能楼梯不一样 下载的地方不一样 这个校验就可能会失效 自己吧makefile 里面那个 hash改成skip 跳过检验

大神 麻烦把路径告诉我 在哪个文件夹里的makefile  hash 是哪一行呀

WYC-2020, 2024-03-13:
> > 这玩意 每个人可能楼梯不一样 下载的地方不一样 这个校验就可能会失效 自己吧makefile 里面那个 hash改成skip 跳过检验
> 
> 大神 麻烦把路径告诉我 在哪个文件夹里的makefile hash 是哪一行呀

看情况 好像是rust内部的 这个没法改 要改只能打path 还是搞搞楼梯看下吧，因为我这边 x86没这个问题

shedkiss, 2024-03-13:
> > > 这玩意 每个人可能楼梯不一样 下载的地方不一样 这个校验就可能会失效 自己吧makefile 里面那个 hash改成skip 跳过检验
> > 
> > 
> > 大神 麻烦把路径告诉我 在哪个文件夹里的makefile hash 是哪一行呀
> 
> 看情况 好像是rust内部的 这个没法改 要改只能打path 还是搞搞楼梯看下吧，因为我这边 x86没这个问题

搞搞楼梯 是科学上网换代理吗
还是？

shedkiss, 2024-03-13:
> > > 这玩意 每个人可能楼梯不一样 下载的地方不一样 这个校验就可能会失效 自己吧makefile 里面那个 hash改成skip 跳过检验
> > 
> > 
> > 大神 麻烦把路径告诉我 在哪个文件夹里的makefile hash 是哪一行呀
> 
> 看情况 好像是rust内部的 这个没法改 要改只能打path 还是搞搞楼梯看下吧，因为我这边 x86没这个问题

大师能否编辑个最新的LEDE 结合helloworld ssr所有协议的镜像发给我啊 我搞了太多次了 都失败

WYC-2020, 2024-03-14:
> > > > 这玩意 每个人可能楼梯不一样 下载的地方不一样 这个校验就可能会失效 自己吧makefile 里面那个 hash改成skip 跳过检验
> > > 
> > > 
> > > 大神 麻烦把路径告诉我 在哪个文件夹里的makefile hash 是哪一行呀
> > 
> > 
> > 看情况 好像是rust内部的 这个没法改 要改只能打path 还是搞搞楼梯看下吧，因为我这边 x86没这个问题
> 
> 大师能否编辑个最新的LEDE 结合helloworld ssr所有协议的镜像发给我啊 我搞了太多次了 都失败
恩山到处都是 烂街了 自己去找

shedkiss, 2024-03-15:
换了太多梯子 故障依旧

closed 2024-05-28 by aiamadeus