coolsnowwolf/lede #11987: 拉取最新源码编译6.6内核x86固件报错

bi4tms, 2024-03-13:
### 详细叙述

拉取最新源码后编译6.6内核x86固件报错，6.1内核可以正常编译通过。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

make[3]: Entering directory '/home/x86.6.6/lede/package/qca/shortcut-fe/shortcut-fe'
touch /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.prepared_70fefbd5addc10e4ab1f1d84092204e8_6664517399ebbbc92a37c5bb081b5c53_check
true
[ ! -d ./src/ ] || cp -fpR ./src/. /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe
touch /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.prepared_70fefbd5addc10e4ab1f1d84092204e8_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.configured_*
rm -f /home/x86.6.6/lede/staging_dir/target-x86_64_musl/stamp/.shortcut-fe_installed
(cd /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/./; if [ -x ./configure ]; then find /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/ -name config.guess | xargs -r chmod u+w; find /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/ -name config.guess | xargs -r -n1 cp --remove-destination /home/x86.6.6/lede/scripts/config.guess; find /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/ -name config.sub | xargs -r chmod u+w; find /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/ -name config.sub | xargs -r -n1 cp --remove-destination /home/x86.6.6/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe=shortcut-fe -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe=shortcut-fe -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe=shortcut-fe -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/x86.6.6/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/x86.6.6/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/x86.6.6/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/x86.6.6/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/x86.6.6/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built
touch /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built_check
make -j1 -C "/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.21" KCFLAGS="-fmacro-prefix-map=/home/x86.6.6/lede/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/x86.6.6/lede/staging_dir/host/include -I/home/x86.6.6/lede/staging_dir/hostpkg/include -I/home/x86.6.6/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Mar 13 08:03:53 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/x86.6.6/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/fullconenat.symvers" KERNELRELEASE=6.6.21  M="/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe" EXTRA_CFLAGS+="-DSFE_SUPPORT_IPV6" SFE_SUPPORT_IPV6=y modules
make[4]: Entering directory '/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.21'
  CC [M]  /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv4.o
  LD [M]  /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/shortcut-fe.o
  CC [M]  /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_ipv6.o
  LD [M]  /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/shortcut-fe-ipv6.o
  CC [M]  /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_cm.o
/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_cm.c: In function 'sfe_cm_post_routing':
/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_cm.c:507:20: error: 'struct nf_tcp_net' has no member named 'tcp_no_window_check'
  507 |         if ((tn&&tn->tcp_no_window_check)
      |                    ^~
make[6]: *** [scripts/Makefile.build:243: /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/sfe_cm.o] Error 1
make[5]: *** [/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.21/Makefile:1913: /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe] Error 2
make[4]: *** [Makefile:234: __sub-make] Error 2
make[4]: Leaving directory '/home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.21'
make[3]: *** [Makefile:88: /home/x86.6.6/lede/build_dir/target-x86_64_musl/linux-x86_64/shortcut-fe/.built] Error 2
make[3]: Leaving directory '/home/x86.6.6/lede/package/qca/shortcut-fe/shortcut-fe'
time: package/qca/shortcut-fe/shortcut-fe/compile#5.71#0.60#6.35
    ERROR: package/qca/shortcut-fe/shortcut-fe failed to build.
make[2]: *** [package/Makefile:116: package/qca/shortcut-fe/shortcut-fe/compile] Error 1
make[2]: Leaving directory '/home/x86.6.6/lede'
make[1]: *** [package/Makefile:110: /home/x86.6.6/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x86.6.6/lede'
make: *** [/home/x86.6.6/lede/include/toplevel.mk:231：world] 错误 2

kxdn3, 2024-03-13:
去除sfe试试

WYC-2020, 2024-03-13:
因为 
[image]
这个更新吧 sfe的一个补丁给删除了，原因不清楚，你还原应该就可以编译过，也可能是计划6.6不在支持sfe了吧

liwenjie119, 2024-03-13:
同样是sfe报错，已经去除，选中OFFLOADING重新编译了

Ucopyrightable, 2024-03-13:
本来还不明白到底offloading和SFE该选哪个，现在单选了

qingtian110, 2024-03-13:
6.6 编译成功，感谢组织，感谢社会，感谢国家。

[image]
[image]
[image]

liwenjie119, 2024-03-14:
最新commit修了，但是我还一直有个问题是无论开flow还是sfe，acc界面的运行状态显示都不正常，一直显示正在加载

Ucopyrightable, 2024-03-14:
> 最新commit修了，但是我还一直有个问题是无论开flow还是sfe，acc界面的运行状态显示都不正常，一直显示正在加载

sfe能编了？

WYC-2020, 2024-03-14:
> > 最新commit修了，但是我还一直有个问题是无论开flow还是sfe，acc界面的运行状态显示都不正常，一直显示正在加载
> 
> sfe能编了？

是的l加回来了

closed 2024-03-25 by coolsnowwolf