coolsnowwolf/lede #11991: GL-iNet AX1800无线问题

niliovo, 2024-03-14:
### 详细叙述

GL-iNet AX1800刚启动无线能搜索到，过了一会就没信号

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

GL-iNet AX1800

### 详细日志

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x51af8014]
[    0.000000] Linux version 6.1.81 (nili@build-openwrt) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6484-525b77c08) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Mar 11 02:08:38 2024
[    0.000000] Machine model: GL.iNet GL-AX1800
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000004a0fffff]
[    0.000000]   node   0: [mem 0x000000004a100000-0x000000004a4fffff]
[    0.000000]   node   0: [mem 0x000000004a500000-0x000000004a5fffff]
[    0.000000]   node   0: [mem 0x000000004a600000-0x000000004e3fffff]
[    0.000000]   node   0: [mem 0x000000004e400000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 18 pages/cpu s34536 r8192 d31000 u73728
[    0.000000] pcpu-alloc: s34536 r8192 d31000 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 433608K/524288K available (7872K kernel code, 872K rwdata, 2232K rodata, 1280K init, 278K bss, 90680K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 24.00MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000076] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000090] pid_max: default: 32768 minimum: 301
[    0.000555] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.000568] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.002410] cblist_init_generic: Setting adjustable number of callback queues.
[    0.002417] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.002619] rcu: Hierarchical SRCU implementation.
[    0.002623] rcu: 	Max phase no-delay instances is 1000.
[    0.003279] smp: Bringing up secondary CPUs ...
[    0.003941] Detected VIPT I-cache on CPU1
[    0.004065] CPU1: Booted secondary processor 0x0000000001 [0x51af8014]
[    0.004733] Detected VIPT I-cache on CPU2
[    0.004826] CPU2: Booted secondary processor 0x0000000002 [0x51af8014]
[... 801 lines cut ...]
[  371.095427] ath11k c000000.wifi: could not request fw stats (-108)
[  371.100810] ath11k c000000.wifi: failed to request fw pdev stats: -108
[  376.806150] ath11k_warn: 138 callbacks suppressed
[  376.806170] ath11k c000000.wifi: failed to send WMI_REQUEST_STATS cmd
[  376.809855] ath11k c000000.wifi: could not request fw stats (-108)
[  376.816339] ath11k c000000.wifi: failed to request fw pdev stats: -108
[  376.823260] ath11k c000000.wifi: failed to send WMI_REQUEST_STATS cmd
[  376.828890] ath11k c000000.wifi: could not request fw stats (-108)
[  376.835390] ath11k c000000.wifi: failed to request fw pdev stats: -108
[  376.844410] ath11k c000000.wifi: failed to send WMI_REQUEST_STATS cmd
[  376.847983] ath11k c000000.wifi: could not request fw stats (-108)
[  376.854470] ath11k c000000.wifi: failed to request fw pdev stats: -108
[  376.861552] ath11k c000000.wifi: failed to send WMI_REQUEST_STATS cmd
[  378.483031] device wlan0 left promiscuous mode
[  378.483224] br-lan: port 5(wlan0) entered disabled state
[  378.544825] wlan0: HW problem - can not stop rx aggregation for be:9e:b1:99:32:e7 tid 0
[  378.544880] wlan0: HW problem - can not stop rx aggregation for be:9e:b1:99:32:e7 tid 1
[  378.551657] wlan0: HW problem - can not stop rx aggregation for be:9e:b1:99:32:e7 tid 6
[  378.559723] wlan0: HW problem - can not stop rx aggregation for c8:5e:a9:8f:fd:7c tid 0
[  383.674880] ath11k_warn: 122 callbacks suppressed
[  383.674902] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 15
[  383.679142] ath11k c000000.wifi: failed to send WMI_PEER_DELETE cmd
[  383.686438] ath11k c000000.wifi: failed to delete peer vdev_id 0 addr c8:5e:a9:8f:fd:7c ret -108
[  383.692485] ath11k c000000.wifi: Failed to delete peer: c8:5e:a9:8f:fd:7c for VDEV: 0
[  383.701533] ath11k c000000.wifi: Found peer entry 94:83:c4:44:cd:95 n vdev 0 after it was supposedly removed

niliovo, 2024-03-14:
[  383.709298] ------------[ cut here ]------------
[  383.719125] WARNING: CPU: 3 PID: 2604 at sta_set_sinfo+0xc74/0xcd0 [mac80211]
[  383.723731] Modules linked in: pppoe ppp_async ath11k_ahb ath11k pppox ppp_mppe ppp_generic mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_state xt_socket xt_recent xt_quota xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_iprange xt_helper xt_conntrack xt_connmark xt_connlimit xt_connlabel xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_FLOWOFFLOAD xt_CT ts_fsm ts_bm tcp_bbr slhc qrtr_smd qrtr qmi_helpers nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_ct nft_chain_nat nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc
[  383.724042]  nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan libcrc32c iptable_raw iptable_nat iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat asn1_decoder sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit netlink_diag tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun
[  383.795447]  crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic seqiv jitterentropy_rng drbg michael_mic hmac cmac arc4 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom qca_nss_dp qca_ssdk gpio_button_hotplug ext4 mbcache jbd2 crc32c_generic
[  383.906789] CPU: 3 PID: 2604 Comm: hostapd Not tainted 6.1.81 #0
[  383.929023] Hardware name: GL.iNet GL-AX1800 (DT)
[  383.935272] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  383.939874] pc : sta_set_sinfo+0xc74/0xcd0 [mac80211]
[  383.946643] lr : sta_set_sinfo+0xc70/0xcd0 [mac80211]
[  383.951850] sp : ffffffc00e0f3720
[  383.956881] x29: ffffffc00e0f3720 x28: 0000000000000000 x27: 0000000000000000
[  383.960187] x26: 0000000000000001 x25: ffffff80063d8ec8 x24: ffffff80063d8e98
[  383.967305] x23: 0000000000000002 x22: ffffff800eb34900 x21: ffffff801051aaa8
[  383.974424] x20: ffffff80063d8880 x19: ffffff801051a000 x18: 0000000000000361
[  383.981543] x17: 302076656476206e x16: 2035393a64633a34 x15: ffffffc008b5c2d0
[  383.988660] x14: 0000000000000a23 x13: 0000000000000361 x12: 00000000ffffffea
[  383.995777] x11: 00000000ffffefff x10: 00000000ffffefff x9 : 0001000000000000
[  384.002896] x8 : 08d0002400040110 x7 : 0000000000000024 x6 : 0000000000008e20
[  384.010014] x5 : ffffffc017469000 x4 : 0000000000000000 x3 : ffffff800eb98000
[  384.017132] x2 : 0000000000000000 x1 : ffffff800eb98000 x0 : 00000000ffffff94
[  384.024251] Call trace:
[  384.031360]  sta_set_sinfo+0xc74/0xcd0 [mac80211]
[  384.033621]  __sta_info_flush+0x104/0x160 [mac80211]
[  384.038482]  ieee80211_nan_func_terminated+0x4e0/0x640 [mac80211]
[  384.043519]  cfg80211_leave_mesh+0xcc/0x174 [cfg80211]
[  384.049507]  __cfg80211_stop_ap+0xc8/0xec [cfg80211]
[  384.054542]  cfg80211_stop_ap+0x4c/0x80 [cfg80211]
[  384.059662]  cfg80211_check_station_change+0x3138/0x4860 [cfg80211]
[  384.064266]  genl_family_rcv_msg_doit+0xb8/0x120
[  384.070426]  genl_rcv_msg+0x108/0x230
[  384.075285]  netlink_rcv_skb+0x5c/0x130
[  384.078844]  genl_rcv+0x38/0x50
[  384.082489]  netlink_unicast+0x258/0x2d0
[  384.085615]  netlink_sendmsg+0x1a4/0x3d0
[  384.089783]  ____sys_sendmsg+0x1c0/0x270
[  384.093689]  ___sys_sendmsg+0x7c/0xc0
[  384.097594]  __sys_sendmsg+0x48/0x90
[  384.101154]  __arm64_sys_sendmsg+0x24/0x30
[  384.104799]  invoke_syscall.constprop.0+0x5c/0x104
[  384.108708]  do_el0_svc+0x60/0x18c
[  384.113479]  el0_svc+0x18/0x54
[  384.116864]  el0t_64_sync_handler+0xbc/0x140
[  384.119904]  el0t_64_sync+0x174/0x178
[  384.124331] ---[ end trace 0000000000000000 ]---
[  389.194750] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 15
[  389.194866] ath11k c000000.wifi: failed to send WMI_PEER_DELETE cmd
[  389.201573] ath11k c000000.wifi: failed to delete peer vdev_id 0 addr be:9e:b1:99:32:e7 ret -108
[  389.207683] ath11k c000000.wifi: Failed to delete peer: be:9e:b1:99:32:e7 for VDEV: 0
[  389.216694] ath11k c000000.wifi: Found peer entry 94:83:c4:44:cd:95 n vdev 0 after it was supposedly removed
[  389.224807] ath11k c000000.wifi: failed to submit WMI_VDEV_DOWN cmd
[  389.234297] ath11k c000000.wifi: failed to down vdev_id 0: -108
[  389.240304] ath11k c000000.wifi: failed to submit WMI_VDEV_STOP cmd
[  389.246198] ath11k c000000.wifi: failed to stop WMI vdev 0: -108
[... cut ...]
[  400.553342] ath11k c000000.wifi: failed to set BA BUFFER SIZE 256 for vdev: 1
[  400.560040] ath11k c000000.wifi: failed to send WMI_VDEV_SET_PARAM_CMDID
[  405.008087] device wlan1 left promiscuous mode
[  405.008267] br-lan: port 6(wlan1) entered disabled state
[  405.115089] ath11k c000000.wifi: failed to clear rx_filter for monitor status ring: (-108)
[  405.719563] ath11k_warn: 115 callbacks suppressed
[  405.719584] ath11k c000000.wifi: failed to send WMI_PDEV_SET_PARAM cmd
[  405.723277] ath11k c000000.wifi: failed to enable PMF QOS: (-108

King-Of-Knights, 2024-04-16:
+1，遇到这个问题

uptopsea, 2024-06-02:
+1，遇到这个问题

coolsnowwolf, 2026-09-05:
已修复并推送 LEDE 独立提交 `ab930bff8bd98d4cf74196a77425acbda54561c8`。日志中的 ath11k 固件为旧版 `WLAN.HK.2.5.0.1-01208`，在运行一段时间后出现 WMI `-108` 和固件失联；已参照 OpenWrt 主线，为 IPQ6018 切换到 `WLAN.HK.2.5.0.1-03982`（Codelinaro 2024-10-14 固件），其他 ath11k 包保持不变。已完成下载、解包路径检查和 Makefile/diff 检查；本地目标配置未选择该固件包，未进行硬件运行验证。提交已推送且远端 master 与本地一致，请刷入新固件后验证，现关闭为已完成。

closed 2026-09-05 by coolsnowwolf