coolsnowwolf/lede #11992: 保存在`/etc/sysctl.conf`的参数无法被加载

devioa, 2024-03-14:
### 详细叙述

比如我在`sysctl.conf`里配置了`net.ipv4.tcp_congestion_control=bbr`和`net.netfilter.nf_conntrack_max=4194304`等参数
但是每次系统启动后这些参数都无法被加载，通过`sysctl -a|grep tcp_congestion_control`和`sysctl -a|grep nf_conntrack_max`查询
配置参数还是默认的`net.ipv4.tcp_congestion_control = cubic`和`net.netfilter.nf_conntrack_max = 65535`
但是手动执行`sysctl -w net.netfilter.nf_conntrack_max=4194304`后再看就是加载成功了的
使用 6f205863c1239b61e479db61b4e0362624aced15 源码编译
待会试下退回`6.1`内核再编译看看

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Linux OpenWrt 6.6.21 #0 SMP PREEMPT_DYNAMIC Wed Mar 13 11:52:52 2024 x86_64 GNU/Linux

### 详细日志

n/a

closed 2024-03-14 by devioa

mygithub1981, 2024-03-15:
> 找到原因了是我若智了，编译的时候忘了把 bbr 内核支持勾选上了，导致`sysctl: write error: No such file or directory`，然后后面的参数全都不会被加载

bbr内核编译的时候在哪选？从来没选过唉。这个对openwrt有影响吗？

devioa, 2024-03-15:
@mygithub1981 kernel modules → network support → tcp bbr，这个一般在 luci → applications → luci-turboacc 这个下面有默认勾选编译，但是我不喜欢用 turboacc 所以我在这里取消勾选后有时候忘了回 kernel modules 里去重新勾上