coolsnowwolf/lede #11994: x86 选择MT7922编译会报错

392377870, 2024-03-14:
### 详细叙述

x86 选择MT7922编译会报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

In file included from /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/linux/leds.h:3,
                 from ./include/linux/rfkill.h:36,
                 from /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/linux/rfkill.h:3,
                 from /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/include/net/cfg80211.h:29,
                 from /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/include/net/mac80211.h:22,
                 from /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/main.c:11:
./include/linux/leds.h:513:46: note: expected 'long unsigned int' but argument is of type 'long unsigned int *'
  513 |                                unsigned long delay_off,
      |                                ~~~~~~~~~~~~~~^~~~~~~~~
  CC [M]  /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/aead_api.o
  CC [M]  /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/wpa.o
/home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/sta_info.c: In function '__sta_info_alloc':
/home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/sta_info.c:556:50: error: implicit declaration of function 'prandom_u32_max'; did you mean 'prandom_u32_state'? [-Werror=implicit-function-declaration]
  556 |         sta->ampdu_mlme.dialog_token_allocator = prandom_u32_max(U8_MAX);
      |                                                  ^~~~~~~~~~~~~~~
      |                                                  prandom_u32_state
/home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/sta_info.c: In function 'sta_get_tidstats_msdu':
/home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/sta_info.c:2415:25: error: implicit declaration of function 'u64_stats_fetch_begin_irq'; did you mean 'u64_stats_fetch_begin'? [-Werror=implicit-function-declaration]
 2415 |                 start = u64_stats_fetch_begin_irq(&rxstats->syncp);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                         u64_stats_fetch_begin
  CC [M]  /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/scan.o
/home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/sta_info.c:2417:18: error: implicit declaration of function 'u64_stats_fetch_retry_irq'; did you mean 'u64_stats_fetch_retry'? [-Werror=implicit-function-declaration]
 2417 |         } while (u64_stats_fetch_retry_irq(&rxstats->syncp, start));
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                  u64_stats_fetch_retry
  CC [M]  /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/offchannel.o
  CC [M]  /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/ht.o
cc1: some warnings being treated as errors
  CC [M]  /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/agg-tx.o
make[9]: *** [scripts/Makefile.build:243: /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211/sta_info.o] Error 1
make[9]: *** Waiting for unfinished jobs....
make[8]: *** [scripts/Makefile.build:480: /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/mac80211] Error 2
make[7]: *** [Makefile:1913: /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24'
make[3]: *** [Makefile:401: /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/.built] Error 2
make[3]: Leaving directory '/home/openwrt-19810729/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#13.70#14.75#55.78
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/home/openwrt-19810729/lede'
make[1]: *** [package/Makefile:110: /home/openwrt-19810729/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt-19810729/lede'
make: *** [/home/openwrt-19810729/lede/include/toplevel.mk:231：world] 错误 2
openwrt-19810729@jbl123-virtual-machine:~/lede$

dph5199278, 2024-03-14:
同样是6.1.24版本的mac80211报错

WYC-2020, 2024-03-15:
试试最新的 l 我看 提交 好像修了

coolsnowwolf, 2024-03-15:
已修复

dph5199278, 2024-03-15:
> 已修复

还是会报错，错误信息：
```bash
make[4]: Entering directory '/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24'
make[5]: 'Kconfig.versions' is up to date.
make[7]: 'Kconfig.versions' is up to date.
make[8]: 'conf' is up to date.
boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
#
# configuration written to .config
#
Building backport-include/backport/autoconf.h ... done.
  CC [M]  /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/wireless/core.o
In file included from /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/wireless/core.c:25:
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h: In function '__bp_genl_info_userhdr_set':
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h:9:13: error: 'struct genl_info' has no member named 'userhdr'
    9 |         info->userhdr = userhdr;
      |             ^~
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h: In function '__bp_genl_info_userhdr':
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h:14:20: error: 'struct genl_info' has no member named 'userhdr'
   14 |         return info->userhdr;
      |                    ^~
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h: At top level:
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h:48:21: error: conflicting types for 'genl_info_userhdr'; have 'void *(struct genl_info *)'
   48 | static inline void *genl_info_userhdr(struct genl_info *info)
      |                     ^~~~~~~~~~~~~~~~~
In file included from /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h:3:
./include/net/genetlink.h:128:21: note: previous definition of 'genl_info_userhdr' with type 'void *(const struct genl_info *)'
  128 | static inline void *genl_info_userhdr(const struct genl_info *info)
      |                     ^~~~~~~~~~~~~~~~~
make[9]: *** [scripts/Makefile.build:243: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/wireless/core.o] Error 1
make[8]: *** [scripts/Makefile.build:480: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/wireless] Error 2
make[7]: *** [Makefile:1913: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24] Error 2
make[6]: *** [Makefile.build:13: modules] Error 2
make[5]: *** [Makefile.real:93: modules] Error 2
make[4]: *** [Makefile:121: modules] Error 2
make[4]: Leaving directory '/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24'
make[3]: *** [Makefile:401: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/.built] Error 2
make[3]: Leaving directory '/opt/lede/package/kernel/mac80211'
time: package/kernel/mac80211/regular/compile#1.00#0.15#19.78
    ERROR: package/kernel/mac80211 failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
make[2]: Leaving directory '/opt/lede'
make[1]: *** [package/Makefile:110: /opt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/lede'
make: *** [/opt/lede/include/toplevel.mk:231: world] Error 2
```

这次的信息不大一样了

dph5199278, 2024-03-15:
看着应该是可以了，这边是在打包时报错，我全部目录清理一遍，再试试

392377870, 2024-03-15:
> > 已修复
> 
> 还是会报错，错误信息：
> 
> ```shell
> make[4]: Entering directory '/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24'
> make[5]: 'Kconfig.versions' is up to date.
> make[7]: 'Kconfig.versions' is up to date.
> make[8]: 'conf' is up to date.
> boolean symbol CRYPTO_LIB_ARC4 tested for 'm'? test forced to 'n'
> #
> # configuration written to .config
> #
> Building backport-include/backport/autoconf.h ... done.
>   CC [M]  /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/wireless/core.o
> In file included from /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/wireless/core.c:25:
> /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h: In function '__bp_genl_info_userhdr_set':
> /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h:9:13: error: 'struct genl_info' has no member named 'userhdr'
>     9 |         info->userhdr = userhdr;
>       |             ^~
> /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h: In function '__bp_genl_info_userhdr':
> /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h:14:20: error: 'struct genl_info' has no member named 'userhdr'
>    14 |         return info->userhdr;
>       |                    ^~
> /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h: At top level:
> /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h:48:21: error: conflicting types for 'genl_info_userhdr'; have 'void *(struct genl_info *)'
>    48 | static inline void *genl_info_userhdr(struct genl_info *info)
>       |                     ^~~~~~~~~~~~~~~~~
> In file included from /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/backport-include/net/genetlink.h:3:
> ./include/net/genetlink.h:128:21: note: previous definition of 'genl_info_userhdr' with type 'void *(const struct genl_info *)'
>   128 | static inline void *genl_info_userhdr(const struct genl_info *info)
>       |                     ^~~~~~~~~~~~~~~~~
> make[9]: *** [scripts/Makefile.build:243: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/wireless/core.o] Error 1
> make[8]: *** [scripts/Makefile.build:480: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/net/wireless] Error 2
> make[7]: *** [Makefile:1913: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24] Error 2
> make[6]: *** [Makefile.build:13: modules] Error 2
> make[5]: *** [Makefile.real:93: modules] Error 2
> make[4]: *** [Makefile:121: modules] Error 2
> make[4]: Leaving directory '/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24'
> make[3]: *** [Makefile:401: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24/.built] Error 2
> make[3]: Leaving directory '/opt/lede/package/kernel/mac80211'
> time: package/kernel/mac80211/regular/compile#1.00#0.15#19.78
>     ERROR: package/kernel/mac80211 failed to build (build variant: regular).
> make[2]: *** [package/Makefile:116: package/kernel/mac80211/compile] Error 1
> make[2]: Leaving directory '/opt/lede'
> make[1]: *** [package/Makefile:110: /opt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
> make[1]: Leaving directory '/opt/lede'
> make: *** [/opt/lede/include/toplevel.mk:231: world] Error 2
> ```
> 
> 这次的信息不太一样了
编译通过了

dph5199278, 2024-03-16:
> 已修复
>
>
> 还是会报错，错误信息：
> ```shell
> make[4]: Entering directory '/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/backports-6.1.24'
> make[5]: 'Kconfig.versions' is up to date.
> make[7]: 'Kconfig.versions' [...]

是的，我清理目录后也编译通过了，可以关闭issue了

closed 2024-03-16 by 392377870