coolsnowwolf/lede #11997: 5.4内核的x86固件启动后日志有报错：Failed to load DMC firmware i915/glk_dmc_ver1_04.bin

itfish92, 2024-03-15:
### 详细叙述

只有5.4内核有这个报错，5.10、6.1没有。
求大佬解答一下，这个报错对系统有影响没有？没影响就忽略了。
issue里能搜到这个相关的，但都没有回复。Thanks！

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

J4105

### 详细日志

Wed Dec  6 18:11:54 2023 kern.notice kernel: [   63.602485] i915 0000:00:02.0: Failed to load DMC firmware i915/glk_dmc_ver1_04.bin. Disabling runtime power management.
Wed Dec  6 18:11:54 2023 kern.notice kernel: [   63.614051] i915 0000:00:02.0: DMC firmware homepage: [https://git.kernel.org/pub/scm/l ... mware.git/tree/i915](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915)

WYC-2020, 2024-03-15:
应该和去掉了gvt有关系

WYC-2020, 2024-03-15:
[image]
 这样修复 原来的修复有问题导致 915模块全部丢失

WYC-2020, 2024-03-15:
其他模块 还没看 有没相同的问题

itfish92, 2024-03-15:
> ![图像](https://private-user-images.githubusercontent.com/34025388/313151832-d088a9eb-fca9-4cff-bd67-d17581309d6a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA1MDQwNzIsIm5iZiI6MTcxMDUwMzc3MiwicGF0aCI6Ii8zNDAyNTM4OC8zMTMxNTE4MzItZDA4OGE5ZWItZmNhOS00Y2ZmLWJkNjctZDE3NTgxMzA5ZDZhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE1VDExNTYxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI0ZTViMTU5OTYzNGJiNThkYzRhMTg3YWEzYzExY2MxNmFkMzFlYzNmMDE1ZGRkOTVjMjNkYTU3ODZmYmIzMTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.fjWsITxvz8zlTPuiPkpAKg3KlYpU8JFdeWdSISjWf9M) 正确修复原来的修复有问题导致915模块全部丢失

谢谢大佬指点，我马上试试看

itfish92, 2024-03-15:
> 其他模块 还没看 有没相同的问题

> ![image](https://private-user-images.githubusercontent.com/34025388/313151832-d088a9eb-fca9-4cff-bd67-d17581309d6a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

> ![image](https://private-user-images.githubusercontent.com/34025388/313151832-d088a9eb-fca9-4cff-bd67-d17581309d6a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

大佬，折腾完了。问题依旧啊，这个报错影响系统正常运行不？只是报错，没什么影响我就不管了。感觉还是5.4内核稳点

WYC-2020, 2024-03-15:
> 大佬，折腾完了。问题依旧啊，这个报错影响系统正常运行不？只是报错，没什么影响我就不管了。感觉还是5.4内核稳点

你改了需要看下是否选择了，有可能没选择

closed 2024-03-15 by itfish92