coolsnowwolf/lede #12000: Cargo.lock反复编译失败 换了代理依旧报错，L大能帮忙看看吗

shedkiss, 2024-03-15:
### 详细叙述

从错误提示看是Cargo.lock反复校验失败，换了若干代理故障依旧，奇怪了
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

the listed checksum of /root/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/vendor/icu_locid/Cargo.lock has changed:
expected: 1aa9f21333d579ccbdf71478538b586518faa9d7f73f7ba8049649f5d4e33b43
actual: 568edd2ce05aa3cf0e0ee8487a5d328aca123897b945c431a9e58a87bf986f5e

directory sources are not intended to be edited, if modifications are required then it is recommended that [patch] is used with a forked copy of the source
Build completed unsuccessfully in 0:00:01
make[3]: *** [Makefile:111: /root/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/.built] Error 1
make[3]: Leaving directory '/root/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#0.92#0.35#1.61
ERROR: package/feeds/packages/rust [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/root/lede'
make[1]: *** [package/Makefile:110: /root/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/lede'
make: *** [/root/lede/include/toplevel.mk:231: world] Error 2

joinevans, 2024-11-29:
我也是这个错误 咋回事

coolsnowwolf, 2026-09-05:
该错误来自 2024 年 Rust 1.74.1 源码树（rustc-1.74.1-src/vendor/icu_locid/Cargo.lock）的旧构建目录，不能代表当前源码。当前 feeds/packages/lang/rust 已升级到 Rust 1.98.0（源码包哈希 271fa73d...），当前 Rust Makefile 没有修改 vendor/icu_locid/Cargo.lock 的补丁，OpenWrt 主线采用相同构建逻辑。遇到同类旧缓存错误时应删除对应 build_dir/target-*/host/rustc-*-src、重新下载/解压并重新编译；若当前版本仍复现，请附新的 Rust 版本、源码包哈希和 .cargo-checksum.json 差异。本 issue 关闭为过时构建缓存/版本问题。

closed 2026-09-05 by coolsnowwolf