coolsnowwolf/lede #12001: X86最新源码重复报错，配置和报错相关内容在附件请大佬看看~

ZGNCZMY, 2024-03-16:
### 详细叙述

J1编译一直重复附件内的报错，重新拉的最新源码只选了Luci APP，配置在附件内

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

[image]
[新文件.txt](https://github.com/coolsnowwolf/lede/files/14622226/default.txt)
[X86配置文件.config.log](https://github.com/coolsnowwolf/lede/files/14622240/X86.config.log)

ZGNCZMY, 2024-03-16:
@
/

ZGNCZMY, 2024-03-16:
> ### 详细叙述
> J1编译一直重复附件内的报错，重新拉的最新源码只选了Luci APP，配置在附件内
> 
> ### 重复 issue
> * [x]  没有类似的 issue
> 
> ### 具体型号
> X86
> 
> ### 详细日志
 [新文件.txt](https://github.com/coolsnowwolf/lede/files/14622226/default.txt) [X86配置文件.config.log](https://github.com/coolsnowwolf/lede/files/14622240/X86.config.log)

dph5199278, 2024-03-16:
将
```
CONFIG_NODEJS_16=y
# CONFIG_NODEJS_18 is not set
```
改为
```
# CONFIG_NODEJS_16 is not set
CONFIG_NODEJS_18=y
```
试下

ZGNCZMY, 2024-03-16:
> 将
> 
> ```
> CONFIG_NODEJS_16=y
> # CONFIG_NODEJS_18 is not set
> ```
> 
> 改为
> 
> ```
> # CONFIG_NODEJS_16 is not set
> CONFIG_NODEJS_18=y
> ```
> 
> 试下

好的，感谢支招。在配置那里改了，J1跑了一晚上了，等结果看看

ZGNCZMY, 2024-03-17:
> 将
> 
> ```
> CONFIG_NODEJS_16=y
> # CONFIG_NODEJS_18 is not set
> ```
> 
> 改为
> 
> ```
> # CONFIG_NODEJS_16 is not set
> CONFIG_NODEJS_18=y
> ```
> 
> 试下

非常感谢！！改了之后清理./staging_dir ./build_dir ./tmp重新J1跑了十多个小时过了

closed 2024-03-17 by ZGNCZMY