coolsnowwolf/lede #12011: 使用kernel6.6编译lede到strace报错，需要升级strace到6.6及以上版本

xpc0938, 2024-03-18:
### 详细叙述

如题，使用kernel6.6编译工程，同时配置了CONFIG_PACKAGE_strace=y，编译报错（kernel6.1配合strace无问题），
报错信息见详细日志。
参考解决办法：https://lore.kernel.org/buildroot/20231102184613.1d071280@windsurf/T/，
可以通过升级strace版本解决。
升级strace版本到6.6后，已验证kernel6.1和6.6均能编译成功。
看大佬考虑一下是否有必要升级strace版本。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

make[3]: Entering directory '/data/prj/lede/package/boot/grub2'
mkdir -p /data/prj/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /data/prj/lede/tmp/.root-copy.flock -c 'cp -fpR /data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-none/grub-2.12/.pkgdir/grub2-bios-setup/. /data/prj/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /data/prj/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.grub2-bios-setup_installed
echo "grub2-bios-setup" >> /data/prj/lede/staging_dir/target-x86_64_musl/pkginfo/grub2.none.install
make[3]: Leaving directory '/data/prj/lede/package/boot/grub2'
time: package/boot/grub2/none/compile#0.16#0.08#0.22
make[3]: Entering directory '/data/prj/lede/package/devel/strace'
rm -f /data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1/.built
touch /data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1=strace-6.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include   " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1=strace-6.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -I/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include   " LDFLAGS="-L/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro   " make -j1 -C /data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1/. CCOPT="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1=strace-6.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  " ;
make[4]: Entering directory '/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1'
make  all-recursive
make[5]: Entering directory '/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1'
Making all in bundled
make[6]: Entering directory '/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1/bundled'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1/bundled'
Making all in src
make[6]: Entering directory '/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1/src'
make  all-am
make[7]: Entering directory '/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1/src'
x86_64-openwrt-linux-musl-gcc -DHAVE_CONFIG_H   -I./linux/x86_64 -I./linux/x86_64 -I./linux/generic -I./linux/generic -I. -I.  -DIN_STRACE=1      -I/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/data/prj/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include    -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -Wdate-time -Wformat-security -Wimplicit-fallthrough=5 -Winit-self -Wlogical-op -Wmissing-prototypes -Wnested-externs -Wold-style-definition -Wtrampolines -Wundef -Wwrite-strings -Werror   -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1=strace-6.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro    -MT libstrace_a-btrfs.o -MD -MP -MF .deps/libstrace_a-btrfs.Tpo -c -o libstrace_a-btrfs.o `test -f 'btrfs.c' || echo './'`btrfs.c
In file included from btrfs.c:51:
xlat/btrfs_key_types.h:167:37: error: 'BTRFS_EXTENT_REF_V0_KEY' undeclared here (not in a function); did you mean 'BTRFS_EXTENT_DATA_KEY'?
  167 | static_assert((unsigned long long) (BTRFS_EXTENT_REF_V0_KEY)
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~
      |                                     BTRFS_EXTENT_DATA_KEY
xlat/btrfs_key_types.h:167:15: error: expression in static assertion is not an integer
  167 | static_assert((unsigned long long) (BTRFS_EXTENT_REF_V0_KEY)
      |               ^
xlat/btrfs_key_types.h:175:15: error: expression in static assertion is not an integer
  175 | static_assert((unsigned long long) (BTRFS_SHARED_BLOCK_REF_KEY)
      |               ^
make[7]: *** [Makefile:4306: libstrace_a-btrfs.o] Error 1
make[7]: Leaving directory '/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1/src'
make[6]: *** [Makefile:2345: all] Error 2
make[6]: Leaving directory '/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1/src'
make[5]: *** [Makefile:595: all-recursive] Error 1
make[5]: Leaving directory '/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1'
make[4]: *** [Makefile:500: all] Error 2
make[4]: Leaving directory '/data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1'
make[3]: *** [Makefile:88: /data/prj/lede/build_dir/target-x86_64_musl/linux-x86_64/strace-6.1/.built] Error 2
make[3]: Leaving directory '/data/prj/lede/package/devel/strace'
time: package/devel/strace/compile#0.21#0.07#0.30
    ERROR: package/devel/strace failed to build.
make[2]: *** [package/Makefile:116: package/devel/strace/compile] Error 1
make[2]: Leaving directory '/data/prj/lede'
make[1]: *** [package/Makefile:110: /data/prj/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/data/prj/lede'
make: *** [/data/prj/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2024-03-18:
试过5.4 5.10 5.15 了么

closed 2024-03-18 by aiamadeus

xpc0938, 2024-03-18:
> 试过5.4 5.10 5.15 了么

试过了，编译没问题。

commit a4880e2f (2024-03-18): strace: bump to version 6.6 (#12012)

Fixes: #12011