coolsnowwolf/lede #12013: There was an error on compilation, how can the problem be solved?

tuzen2008, 2024-03-18:
### 详细叙述

   Compiling cc v1.0.79
   Compiling core v0.0.0 (/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/library/core)
   Compiling libc v0.2.148
   Compiling memchr v2.5.0
   Compiling std v0.0.0 (/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/library/std)
   Compiling compiler_builtins v0.1.101
   Compiling unwind v0.0.0 (/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/library/unwind)
   Compiling rustc-std-workspace-core v1.99.0 (/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/library/rustc-std-workspac                                                                                            e-core)
   Compiling alloc v0.0.0 (/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/library/alloc)
   Compiling cfg-if v1.0.0
   Compiling adler v1.0.2
   Compiling rustc-demangle v0.1.23
error: failed to prepare thin LTO module: Invalid alignment value (Producer: 'LLVM17.0.4-rust-1.74.1-stable' Reader: 'LLVM 17.0.4-rust-1.74.1-                                                                                            stable')

error: could not compile `core` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...
error: failed to parse bitcode for LTO module: Invalid alignment value (Producer: 'LLVM17.0.4-rust-1.74.1-stable' Reader: 'LLVM 17.0.4-rust-1.                                                                                            74.1-stable')

error: could not compile `rustc-demangle` (lib) due to previous error
error: failed to parse bitcode for LTO module: Inserted value type doesn't match aggregate type (Producer: 'LLVM17.0.4-rust-1.74.1-stable' Rea                                                                                            der: 'LLVM 17.0.4-rust-1.74.1-stable')

error: could not compile `alloc` (lib) due to previous error
Build completed unsuccessfully in 0:08:37
Makefile:110: recipe for target '/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/.built' failed
make[3]: *** [/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/.built] Error 1
make[3]: Leaving directory '/home/tuzeen/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#1623.65#72.66#587.15
    ERROR: package/feeds/packages/rust [host] failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/rust/host/compile' failed
make[2]: *** [package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/home/tuzeen/lede'
package/Makefile:110: recipe for target '/home/tuzeen/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/tuzeen/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tuzeen/lede'
/home/tuzeen/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

OPENWRT

### 详细日志

   Compiling cc v1.0.79
   Compiling core v0.0.0 (/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/library/core)
   Compiling libc v0.2.148
   Compiling memchr v2.5.0
   Compiling std v0.0.0 (/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/library/std)
   Compiling compiler_builtins v0.1.101
   Compiling unwind v0.0.0 (/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/library/unwind)
   Compiling rustc-std-workspace-core v1.99.0 (/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/library/rustc-std-workspac                                                                                            e-core)
   Compiling alloc v0.0.0 (/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/library/alloc)
   Compiling cfg-if v1.0.0
   Compiling adler v1.0.2
   Compiling rustc-demangle v0.1.23
error: failed to prepare thin LTO module: Invalid alignment value (Producer: 'LLVM17.0.4-rust-1.74.1-stable' Reader: 'LLVM 17.0.4-rust-1.74.1-                                                                                            stable')

error: could not compile `core` (lib) due to previous error
warning: build failed, waiting for other jobs to finish...
error: failed to parse bitcode for LTO module: Invalid alignment value (Producer: 'LLVM17.0.4-rust-1.74.1-stable' Reader: 'LLVM 17.0.4-rust-1.                                                                                            74.1-stable')

error: could not compile `rustc-demangle` (lib) due to previous error
error: failed to parse bitcode for LTO module: Inserted value type doesn't match aggregate type (Producer: 'LLVM17.0.4-rust-1.74.1-stable' Rea                                                                                            der: 'LLVM 17.0.4-rust-1.74.1-stable')

error: could not compile `alloc` (lib) due to previous error
Build completed unsuccessfully in 0:08:37
Makefile:110: recipe for target '/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/.built' failed
make[3]: *** [/home/tuzeen/lede/build_dir/target-x86_64_musl/host/rustc-1.74.1-src/.built] Error 1
make[3]: Leaving directory '/home/tuzeen/lede/feeds/packages/lang/rust'
time: package/feeds/packages/rust/host-compile#1623.65#72.66#587.15
    ERROR: package/feeds/packages/rust [host] failed to build.
package/Makefile:114: recipe for target 'package/feeds/packages/rust/host/compile' failed
make[2]: *** [package/feeds/packages/rust/host/compile] Error 1
make[2]: Leaving directory '/home/tuzeen/lede'
package/Makefile:110: recipe for target '/home/tuzeen/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/tuzeen/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tuzeen/lede'
/home/tuzeen/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2024-03-18:
you may need to upgrade your host os g++ version to  11+

tuzen2008, 2024-03-20:
Thank you, I upgraded to 13 and the compilation passed.

closed 2024-03-20 by tuzen2008