coolsnowwolf/lede #12015: 编译过程报错 Error when bootstrapping CMake: Problem while running /home/lede/lede/staging_dir/host/bin/ninja

qx100, 2024-03-19:
### 详细叙述

大家好 ,我在编译过程中遇到一些问题

编译使用云服务器 系统为 Ubuntu 20.04.6 LTS

更换为Ubuntu 22 系统仍可以复现该问题

编译前下载了最新源码

插件添加了passwall2 

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ariaboard photonicat 光影猫

### 详细日志

[6/186] g++ -O2 -I/home/lede/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Bootstrap.cmk   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Source   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Source/LexerParser   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Utilities/std   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Utilities  -c /home/lede/lede/build_dir/host/cmake-3.25.2/Source/cmGeneratorTarget.cxx -o cmGeneratorTarget.o

FAILED: cmGeneratorTarget.o

g++ -O2 -I/home/lede/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Bootstrap.cmk   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Source   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Source/LexerParser   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Utilities/std   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Utilities  -c /home/lede/lede/build_dir/host/cmake-3.25.2/Source/cmGeneratorTarget.cxx -o cmGeneratorTarget.o

g++: fatal error: Killed signal terminated program cc1plus

compilation terminated.

[7/186] g++ -O2 -I/home/lede/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Bootstrap.cmk   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Source   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Source/LexerParser   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Utilities/std   -I/home/lede/lede/build_dir/host/cmake-3.25.2/Utilities  -c /home/lede/lede/build_dir/host/cmake-3.25.2/Source/cmGetSourceFilePropertyCommand.cxx -o cmGetSourceFilePropertyCommand.o
ninja: build stopped: subcommand failed.

---------------------------------------------

Error when bootstrapping CMake:

Problem while running /home/lede/lede/staging_dir/host/bin/ninja

---------------------------------------------

Log of errors: /home/lede/lede/build_dir/host/cmake-3.25.2/Bootstrap.cmk/cmake_bootstrap.log

---------------------------------------------

make[3]: *** [Makefile:60: /home/lede/lede/build_dir/host/cmake-3.25.2/.configured] Error 9

make[3]: Leaving directory '/home/lede/lede/tools/cmake'

time: tools/cmake/compile#48.30#18.39#3814.83

    ERROR: tools/cmake failed to build.

make[2]: *** [tools/Makefile:221: tools/cmake/compile] Error 1

make[2]: Leaving directory '/home/lede/lede'

make[1]: *** [tools/Makefile:217: /home/lede/lede/staging_dir/host/

stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2

make[1]: Leaving directory '/home/lede/lede'

make: *** [/home/lede/lede/include/toplevel.mk:231: world] Error 2

qx100, 2024-03-19:
参考了其他类似的issue
[issue 11371](https://github.com/coolsnowwolf/lede/issues/11371)
[issue 10914](https://github.com/coolsnowwolf/lede/issues/10914)
[issue 10581](https://github.com/coolsnowwolf/lede/issues/10581)
[issue 9303](https://github.com/coolsnowwolf/lede/issues/9303)
[issue 9350](https://github.com/coolsnowwolf/lede/issues/9350)
但是并没有对我的错误起到帮助,故新开一个issue

WYC-2020, 2024-03-19:
怕不是又是内存小了吧

qx100, 2024-03-19:
使用的是硅云香港机房的 配置为1核1024MiB内存

WYC-2020, 2024-03-19:
这配置 我怀疑单线程 都跑不了

qx100, 2024-03-19:
好的 谢谢 我更换虚拟机试一试

> 这配置 我怀疑单线程 都跑不了

closed 2024-03-22 by coolsnowwolf

jesee, 2024-12-21:
这个解决了吗，我今天更新了代码，也是出现这个问题：

[20/271] /debian-home/lede/staging_dir/host/bin/g++ -O2 -I/debian-home/lede/staging_dir/host/include     -D_FILE_OFFSET_BITS=64    -DCMAKE_BOOTSTRAP    -DCMake_HAVE_CXX_MAKE_UNIQUE=1 -DCMake_HAVE_CXX_FILESYSTEM=1   -I/debian-home/lede/build_dir/host/cmake-3.30.5/Bootstrap.cmk   -I/debian-home/lede/build_dir/host/cmake-3.30.5/Source   -I/debian-home/lede/build_dir/host/cmake-3.30.5/Source/LexerParser   -I/debian-home/lede/build_dir/host/cmake-3.30.5/Utilities/std   -I/debian-home/lede/build_dir/host/cmake-3.30.5/Utilities  -c /debian-home/lede/build_dir/host/cmake-3.30.5/Source/cmComputeLinkDepends.cxx -o cmComputeLinkDepends.o
ninja: build stopped: subcommand failed.
---------------------------------------------
Error when bootstrapping CMake:
Problem while running /debian-home/lede/staging_dir/host/bin/ninja
---------------------------------------------
Log of errors: /debian-home/lede/build_dir/host/cmake-3.30.5/Bootstrap.cmk/cmake_bootstrap.log
---------------------------------------------
make[3]: *** [Makefile:59: /debian-home/lede/build_dir/host/cmake-3.30.5/.configured] Error 9
make[3]: Leaving directory '/debian-home/lede/tools/cmake'
time: tools/cmake/compile#96.53#34.58#27.44
    ERROR: tools/cmake failed to build.
make[2]: *** [tools/Makefile:231: tools/cmake/compile] Error 1
make[2]: Leaving directory '/debian-home/lede'
make[1]: *** [tools/Makefile:227: /debian-home/lede/staging_dir/host/stamp/.tools_compile_nyyynyynnnyyynynyyyyyyynyyynynyynnynnyyyyyynyyyyyyyyyyynynnyyyyynyyy] Error 2
make[1]: Leaving directory '/debian-home/lede'
make: *** [/debian-home/lede/include/toplevel.mk:231: world] Error 2