coolsnowwolf/lede #12017: 无线网卡rtl8812au-ac ，不能download， 编译报错

LANEVO09, 2024-03-19:
### 详细叙述

内核6.6下，x86平台上，不能下载rtl8812au-ac，是啥问题？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86 平台

### 详细日志

+ curl -f --connect-timeout 20 --retry 5 --location https://sources.cdn.openwrt.org/rtl8812au-ac-2023-01-23-e72dacdc.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://sources.openwrt.org/rtl8812au-ac-2023-01-23-e72dacdc.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   146    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location https://mirror2.openwrt.org/sources/rtl8812au-ac-2023-01-23-e72dacdc.tar.xz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   153    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404
Download failed.
No more mirrors to try - giving up.

/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:464:17: error: too few arguments to function 'cfg80211_ch_switch_started_notify'
  464 |                 cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0, false);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/include/osdep_service_linux.h:97,
                 from /home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/include/osdep_service.h:58,
                 from /home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/include/drv_types.h:26,
                 from /home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:17:
/home/HPcomputer/workspace/ledeopenwrt/lede/staging_dir/target-x86_64_musl/usr/include/mac80211/net/cfg80211.h:8340:6: note: declared here
 8340 | void cfg80211_ch_switch_started_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:476:9: error: too few arguments to function 'cfg80211_ch_switch_notify'
  476 |         cfg80211_ch_switch_notify(adapter->pnetdev, &chdef);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/include/osdep_service_linux.h:97,
                 from /home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/include/osdep_service.h:58,
                 from /home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/include/drv_types.h:26,
                 from /home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:17:
/home/HPcomputer/workspace/ledeopenwrt/lede/staging_dir/target-x86_64_musl/usr/include/mac80211/net/cfg80211.h:8324:6: note: declared here
 8324 | void cfg80211_ch_switch_notify(struct net_device *dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c: In function 'rtw_cfg80211_indicate_connect':
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:1153:26: error: 'struct cfg80211_roam_info' has no member named 'bssid'
 1153 |                 roam_info.bssid = cur_network->network.MacAddress;
      |                          ^
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c: At top level:
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:10109:20: error: initialization of 'int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *, struct key_params *)' {aka 'int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *, struct key_params *)'} from incompatible pointer type 'int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *, struct key_params *)' {aka 'int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *, struct key_params *)'} [-Werror=incompatible-pointer-types]
10109 |         .add_key = cfg80211_rtw_add_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:10109:20: note: (near initialization for 'rtw_cfg80211_ops.add_key')
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:10110:20: error: initialization of 'int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *, void *, void (*)(void *, struct key_params *))' {aka 'int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *, void *, void (*)(void *, struct key_params *))'} from incompatible pointer type 'int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *, void *, void (*)(void *, struct key_params *))' {aka 'int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *, void *, void (*)(void *, struct key_params *))'} [-Werror=incompatible-pointer-types]
10110 |         .get_key = cfg80211_rtw_get_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:10110:20: note: (near initialization for 'rtw_cfg80211_ops.get_key')
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:10111:20: error: initialization of 'int (*)(struct wiphy *, struct net_device *, int,  u8,  bool,  const u8 *)' {aka 'int (*)(struct wiphy *, struct net_device *, int,  unsigned char,  _Bool,  const unsigned char *)'} from incompatible pointer type 'int (*)(struct wiphy *, struct net_device *, u8,  bool,  const u8 *)' {aka 'int (*)(struct wiphy *, struct net_device *, unsigned char,  _Bool,  const unsigned char *)'} [-Werror=incompatible-pointer-types]
10111 |         .del_key = cfg80211_rtw_del_key,
      |                    ^~~~~~~~~~~~~~~~~~~~
[... 12 lines cut ...]
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:10146:20: note: (near initialization for 'rtw_cfg80211_ops.stop_ap')
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:10232:24: error: initialization of 'int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  struct cfg80211_chan_def *)' from incompatible pointer type 'int (*)(struct wiphy *, struct wireless_dev *, struct cfg80211_chan_def *)' [-Werror=incompatible-pointer-types]
10232 |         .get_channel = cfg80211_rtw_get_channel,
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~
/home/HPcomputer/workspace/ledeopenwrt/lede/build_dir/target-x86_64_musl/linux-x86_64/rtl8812au-ac-2023-01-23-e72dacdc/os_dep/linux/ioctl_cfg80211.c:10232:24: note: (near initialization for 'rtw_cfg80211_ops.ge [... cut ...]
    ERROR: package/kernel/rtl8812au-ac failed to build.
make[2]: *** [package/Makefile:116: package/kernel/rtl8812au-ac/compile] Error 1
make[2]: Leaving directory '/home/HPcomputer/workspace/ledeopenwrt/lede'
make[1]: *** [package/Makefile:110: /home/HPcomputer/workspace/ledeopenwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/HPcomputer/workspace/ledeopenwrt/lede'
make: *** [/home/HPcomputer/workspace/ledeopenwrt/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2024-03-20:
没适配而已 6.6函数原型变化了

WYC-2020, 2024-03-20:
明明就是ac说成ct，害我去看ct结果ct修好了 一看靠 你说的是ac 结果两个都修了一下 替换 同名patch 
[050-backport-6.1.patch.txt](https://github.com/coolsnowwolf/lede/files/14664369/050-backport-6.1.patch.txt)

LANEVO09, 2024-03-21:
不好意思搞错了。
＠WYC-2020，编译通过了，我明天烧录下，感谢大佬。

LANEVO09, 2024-03-22:
我用windows测试 rtl8812  usb无线网卡没有问题。
但是编译完固件后，还是查找不到。
lsusb命令
root@OpenWrt:~# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux 6.1.82 xhci-hcd xHCI Host Controller
Bus 001 Device 002: ID 24ae:2010 RAPOO Rapoo 2.4G Wireless Device
Bus 001 Device 003: ID 0bda:b812 Realtek 802.11ac NIC
Bus 001 Device 004: ID 8087:0026
Bus 001 Device 001: ID 1d6b:0002 Linux 6.1.82 xhci-hcd xHCI Host Controller
lspci命令
root@OpenWrt:~# lspci|grep -i 'eth'
01:00.0 Ethernet controller: Intel Corporation Ethernet Controller I225-V (rev 03)
02:00.0 Ethernet controller: Intel Corporation Ethernet Controller I225-V (rev 03)
内核log只有这个
[   16.295101] usbcore: registered new interface driver rtl88XXau
软件包已安装
kmod-rtl8812au-ac | 6.1.82+2023-01-23-e72..c-2
不知道各位大佬有没有啥思路？

NickLiu635, 2024-09-02:
似乎不止8812au，还有其他的RTL网卡也会出类似问题
```
    ERROR: package/kernel/rtw88-usb failed to build.
    ERROR: package/kernel/rtl8812au-ct failed to build.
    ERROR: package/kernel/rtl8812au-ac failed to build.
```

WlphM, 2024-09-02:
> 似乎不止8812au，还有其他的RTL网卡也会出类似问题
> 
> ```
>     ERROR: package/kernel/rtw88-usb failed to build.
>     ERROR: package/kernel/rtl8812au-ct failed to build.
>     ERROR: package/kernel/rtl8812au-ac failed to build.
> ``
咋解决的，我是8812au-ac一直失败

coolsnowwolf, 2026-09-05:
当前 LEDE/packages checkout 已不再包含 `rtl8812au-ct`/`rtl8812au-ac` 这个旧的 2023-01-23 驱动包，因此不会再从已失效的 source mirror 下载该源码，也不会进入日志中所示的旧 cfg80211 API 编译路径。RTL8812AU 支持应使用当前 mac80211/rtw88 驱动与固件；该 issue 对应的旧包已淘汰，无需对现行源码制作补丁。

closed 2026-09-05 by coolsnowwolf