coolsnowwolf/lede #12019: error: Could not find libxml2 anywhere

rq1025330, 2024-03-20:
### 详细叙述

checking for x86_64-pc-linux-gnu-xml2-config... /home/rq/lede/staging_dir/hostpkg/bin/x86_64-pc-linux-gnu-xml2-config
configure: error: Could not find libxml2 anywhere, check ftp://xmlsoft.org/.
make[3]: *** [Makefile:167: /home/rq/lede/build_dir/hostpkg/libxslt-1.1.34/.configured] Error 1
make[3]: Leaving directory '/home/rq/lede/feeds/packages/libs/libxslt'
time: package/feeds/packages/libxslt/host-compile#4.46#1.96#7.48
    ERROR: package/feeds/packages/libxslt [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/libxslt/host/compile] Error 1
make[2]: Leaving directory '/home/rq/lede'
make[1]: *** [package/Makefile:110: /home/rq/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rq/lede'
make: *** [/home/rq/lede/include/toplevel.mk:231：world] 错误 2

系统ubuntu-22.04.4
sudo apt-get update
sudo apt-get install libxml2-dev
是安装了的，删除重新编译，重新装系统，结果还是报错，这是怎么回事呀？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

arm64

### 详细日志

checking for printf... yes
checking for sprintf... yes
checking for fprintf... yes
checking for snprintf... yes
checking for vfprintf... yes
checking for vsprintf... yes
checking for vsnprintf... yes
checking for sscanf... yes
checking for clock_gettime... yes
checking for perl... perl
Disabling crypto support
Disabling debug support
Disabling debugger
Disabling profiler
checking for x86_64-pc-linux-gnu-xml2-config... /home/rq/lede/staging_dir/hostpkg/bin/x86_64-pc-linux-gnu-xml2-config
configure: error: Could not find libxml2 anywhere, check ftp://xmlsoft.org/.
make[3]: *** [Makefile:167: /home/rq/lede/build_dir/hostpkg/libxslt-1.1.34/.configured] Error 1
make[3]: Leaving directory '/home/rq/lede/feeds/packages/libs/libxslt'
time: package/feeds/packages/libxslt/host-compile#4.46#1.96#7.48
    ERROR: package/feeds/packages/libxslt [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/packages/libxslt/host/compile] Error 1
make[2]: Leaving directory '/home/rq/lede'
make[1]: *** [package/Makefile:110: /home/rq/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rq/lede'
make: *** [/home/rq/lede/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2024-03-20:
你是都没明白 别人不是依赖于你宿主机上面装没装，好不 ，自己看是否有这个包 
[image]

rq1025330, 2024-03-20:
> 你都不清楚别人不是依赖于你上面的机子上面装没装，好不，自己看看是否有这个包 ![图像](https://private-user-images.githubusercontent.com/34025388/314453263-aabd7507-6c7e-4112-882b-a7edbd3f81e9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NDY3MTIsIm5iZiI6MTcxMDk0NjQxMiwicGF0aCI6Ii8zNDAyNTM4OC8zMTQ0NTMyNjMtYWFiZDc1MDctNmM3ZS00MTEyLTg4MmItYTdlZGJkM2Y4MWU5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDE0NTMzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYyYzBkOWI3NDBhY2JiZTlkZGU1YWYxOGZmMDI4NDkwNDZiZjJlYTcyYTRkN2YwODM1MjRlNGZkZTEwNDdmYzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IWBwvWyJ3hdU-elio7kZLYW5FRFg4O0qZUqRcJop9AY)

[image]
[image]
[image]

rq1025330, 2024-03-20:
> > 你都狮子别人不是靠你上面的机子上面装没装，好不，自己看看有没有这个包![图像](https://private-user-images.githubusercontent.com/34025388/314453263-aabd7507-6c7e-4112-882b-a7edbd3f81e9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NDY3MTIsIm5iZiI6MTcxMDk0NjQxMiwicGF0aCI6Ii8zNDAyNTM4OC8zMTQ0NTMyNjMtYWFiZDc1MDctNmM3ZS00MTEyLTg4MmItYTdlZGJkM2Y4MWU5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDE0NTMzMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYyYzBkOWI3NDBhY2JiZTlkZGU1YWYxOGZmMDI4NDkwNDZiZjJlYTcyYTRkN2YwODM1MjRlNGZkZTEwNDdmYzImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.IWBwvWyJ3hdU-elio7kZLYW5FRFg4O0qZUqRcJop9AY)
> 
> ![图像](https://private-user-images.githubusercontent.com/52870887/314550763-7eba189b-370f-4f37-a046-2b0a99254929.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NDcyODYsIm5iZiI6MTcxMDk0Njk4NiwicGF0aCI6Ii81Mjg3MDg4Ny8zMTQ1NTA3NjMtN2ViYTE4OWItMzcwZi00ZjM3LWEwNDYtMmIwYTk5MjU0OTI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDE1MDMwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRjOTJiNWIzYTY0NWI3NWM0ZDhmOWIxMzMxMGI1ODBjMTBhZWU2NWMxZWQ5ZGMxZmM5ZDhiNDc5ODdkOWJjZmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.l18F8W8foVCIMXG5-abytu0JXIpqh6f7mnDHfNweCA8) ![图像](https://private-user-images.githubusercontent.com/52870887/314550878-9d1abef2-da79-4620-9662-33192a52c353.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NDcyODYsIm5iZiI6MTcxMDk0Njk4NiwicGF0aCI6Ii81Mjg3MDg4Ny8zMTQ1NTA4NzgtOWQxYWJlZjItZGE3OS00NjIwLTk2NjItMzMxOTJhNTJjMzUzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDE1MDMwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI1NDRjOTRhNTFlNTZjMzk1NTdlMTAwZDkyOThkYTU4NzgxYWE3ODdhNjA4YWM1OGNmNGI1Y2EwNDI3MWY0NDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uYs-RtL3-tNzTP5iiARnt7FBWnWc6ahPVEYDFr4j8VA) ![图像](https://private-user-images.githubusercontent.com/52870887/314551012-3fa8f7ca-b3f3-44e6-90f4-4b19c9a4c446.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5NDcyODYsIm5iZiI6MTcxMDk0Njk4NiwicGF0aCI6Ii81Mjg3MDg4Ny8zMTQ1NTEwMTItM2ZhOGY3Y2EtYjNmMy00NGU2LTkwZjQtNGIxOWM5YTRjNDQ2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDE1MDMwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxMDg1OGZlYzkyNjc4M2ViNzVhMGMxYTBkMWM0YzJhMjQ1ZmU5MjA0YzI1MmQzNWQxYmZmZDQ5NDllNGRiMTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RlBEefu_jfQMOd1Qbyoblv7ugNDpeyEJ3ppFQaShY_E)

> 你都不清楚别人不是依赖于你上面的机子上面装没装，好不，自己看看是否有这个包 [...]

大佬 我也是学习学习，实在不好意思，这个文件看着都有呀。

WYC-2020, 2024-03-21:
> 大佬 我也是学习学习，实在不好意思，这个文件看着都有呀。

别人这个libexslt makefile 写了依赖于HOST_BUILD_DEPENDS:=libxml2/host 这个玩意的 我不知道你是如何编译的，自己这样编译一下 make package/feeds/packages/libxml2/host/compile V=s  实在不行 自己全新编译

rq1025330, 2024-03-22:
> 别人这个libexslt makefile 写了依赖于HOST_BUILD_DEPENDS:=libxml2/host 这个玩意的 我不知道你是如何编译的，自己这样编译一下 make package/feeds/packages/libxml2/host/compile V=s 实在不行 自己全新编译

[image]
我都重编几次了 找不到原因!

hzjnet, 2024-03-22:
> ![image](https://private-user-images.githubusercontent.com/52870887/315904226-1d0c06f5-a86d-4043-9560-34e91db277be.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

不要纠结，换个openwrt官方版本对应源码替换即可，这个源经常开倒车又不做测试就把源码发出来了！
我测试这个版本是正常的：
cat feeds/packages/libs/libxml2/Makefile|more
PKG_NAME:=libxml2
PKG_VERSION:=2.10.3
PKG_RELEASE:=2

PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.xz
PKG_SOURCE_URL:=@GNOME/libxml2/$(basename $(PKG_VERSION))
PKG_HASH:=5d2cc3d78bec3dbe212a9d7fa629ada25a7da928af432c93060ff5c17ee28a9c

find *bin|grep libxml2
bin/packages/arm_cortex-a15_neon-vfpv4/packages/libxml2_2.10.3-2_arm_cortex-a15_neon-vfpv4.ipk

rq1025330, 2024-03-22:
> 不要纠结，换个openwrt官方版本对应源码替换即可，这个源经常开倒车又不做测试就把源码发出来了！ 我测试这个版本是正常的： cat feeds/packages/libs/libxml2/Makefile|more PKG_NAME:=libxml2 PKG_VERSION:=2.10.3 PKG_RELEASE:=2 [...]

[image]
一样的呀？

WYC-2020, 2024-03-22:
只能你自己排查因为我这边一切正常

rq1025330, 2024-03-23:
> 这个问题几个月了，改Makefile，用imm的Makefile可以过。
直接用 imm libxml2/Makefile 替换就行？

leux0, 2024-03-24:
[image]

rq1025330, 2024-03-25:
> ![QQ截图20240324124334](https://private-user-images.githubusercontent.com/19219159/316283884-ffe42006-2380-485a-ac7f-4bb7c8ee5ab5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEzNDk5MDQsIm5iZiI6MTcxMTM0OTYwNCwicGF0aCI6Ii8xOTIxOTE1OS8zMTYyODM4ODQtZmZlNDIwMDYtMjM4MC00ODVhLWFjN2YtNGJiN2M4ZWU1YWI1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzI1VDA2NTMyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE5M2U2NjcxNTdjZmE5NDNjOGMzYTcwMGJjNDhmNmMwMzllZGExMGFhOWMyM2Q3NDc5ZDQxMTBlZjgxMmE2NDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5Cg4sTOTz1jh5zNlK22qd5TU-t7Vme_5ttBFsrSi3o4)
确实不报错了  package/libs/elfutils failed to build又开始了  ！

rq1025330, 2024-03-25:
> ![QQ截图20240324124334](https://private-user-images.githubusercontent.com/19219159/316283884-ffe42006-2380-485a-ac7f-4bb7c8ee5ab5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiw [...]

解决了elfutils又报下面错误！

make[3]: Entering directory '/home/rq/lede/feeds/packages/lang/python/python-yaml'
rm -f /home/rq/lede/build_dir/target-aarch64_generic_musl/pypi/PyYAML-6.0/.built
touch /home/rq/lede/build_dir/target-aarch64_generic_musl/pypi/PyYAML-6.0/.built_check
SHELL= flock /home/rq/lede/tmp/.pip.flock -c ' ARCH="x86_64" CC="gcc" CCSHARED="gcc -DPIC -fPIC" CXX="g++" LD="gcc" LDSHARED="gcc -shared" CFLAGS="-O2 -I/home/rq/lede/staging_dir/host/include -I/home/rq/lede/staging_dir/hostpkg/include -I/home/rq/lede/staging_dir/target-aarch64_generic_musl/host/include" CPPFLAGS="-I/home/rq/lede/staging_dir/host/include -I/home/rq/lede/staging_dir/hostpkg/include -I/home/rq/lede/staging_dir/target-aarch64_generic_musl/host/include -I/home/rq/lede/staging_dir/hostpkg/include/python3.10" LDFLAGS="-L/home/rq/lede/staging_dir/host/lib -L/home/rq/lede/staging_dir/hostpkg/lib -L/home/rq/lede/staging_dir/target-aarch64_generic_musl/host/lib -lpython3.10 -Wl,-rpath,/home/rq/lede/staging_dir/hostpkg/lib" PYTHONPATH="/home/rq/lede/staging_dir/hostpkg/lib/python3.10:/home/rq/lede/staging_dir/hostpkg/lib/python3.10/site-packages" PYTHONDONTWRITEBYTECODE=0 PYTHONOPTIMIZE="" /home/rq/lede/staging_dir/hostpkg/bin/pip3.10 --cache-dir "/home/rq/lede/dl/pip-cache" --disable-pip-version-check install --no-binary :all: --require-hashes  --requirement ../host-pip-requirements/Cython.txt  ' 
ERROR: Could not find a version that satisfies the requirement Cython==0.29.32 (from versions: none)
ERROR: No matching distribution found for Cython==0.29.32
make[3]: *** [Makefile:48: /home/rq/lede/build_dir/target-aarch64_generic_musl/pypi/PyYAML-6.0/.built] Error 1
make[3]: Leaving directory '/home/rq/lede/feeds/packages/lang/python/python-yaml'
time: package/feeds/packages/python-yaml/compile#0.37#0.33#30.77
    ERROR: package/feeds/packages/python-yaml failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/python-yaml/compile] Error 1
make[2]: Leaving directory '/home/rq/lede'
make[1]: *** [package/Makefile:110: /home/rq/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rq/lede'
make: *** [/home/rq/lede/include/toplevel.mk:231：world] 错误 2

rq1025330, 2024-03-26:
[image]
终于出来了 ，谢谢大佬们！

closed 2024-03-26 by rq1025330