coolsnowwolf/lede #12027: 每次刷系统后，docker都需要手动重启一次，否则容器无法运行

loydcheung, 2024-03-23:
### 详细叙述

如题，每次刷完系统包的第一次启动，docker都需要手动重启一次才行，否则容器显示绿色正常运行，实际是没有正常启动的，点击重启容器，会弹窗报错Containers: start 39ab4decd3bb...code:500 failed to create endpoint xxxxxx(容器名称) on network bridge: adding interface vethf2fa6cc to bridge docker0 failed: Device does not exist

大概是几个月前开始有这个问题，虽然没有很影响使用，但还是很难受，特别是有些需要随系统启动的应用。Google上搜了一圈有说是kernel的bug的。看看有没有大佬能解决。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

Containers: start 39ab4decd3bb...code:500 failed to create endpoint xxxxxx(容器名称) on network bridge: adding interface vethf2fa6cc to bridge docker0 failed: Device does not exist

supppig, 2024-04-02:
我的r2s也这样，不过问题不大，更新了点一点就好。
实在介意自己就在rc.local里面加个脚本呗，读取当前版本，如果当前版本与记录的版本不符合，就重启docker，然后记录当前版本。

coolsnowwolf, 2026-09-05:
当前 dockerd.init 已与 OpenWrt 主线一致（START=99、procd 管理、docker0 网络配置及 reload trigger 均相同）。`adding interface veth... to bridge docker0 failed: Device does not exist` 是刷机首次启动时 Docker 在网络设备/bridge 尚未完成创建的时序现象，不能仅凭日志归因于 Docker 包；手动重启后恢复也印证了这一点。该旧报告没有提供当前版本、完整 netifd 日志或可复现配置，暂无安全通用补丁，现关闭。

closed 2026-09-05 by coolsnowwolf