coolsnowwolf/lede #12030: x86 32位 N2600，编译出错，请指教

jonwhoo, 2024-03-25:
### 详细叙述

近期编译x86 32位 N2600，总是最后出错，请大神指教：
上网组件只添加了helloworld，

ptracesandbox.c: In function 'validate_fcntl':
ptracesandbox.c:857:15: error: 'F_SETLK64' undeclared (first use in this function); did you mean 'F_SETLK'?
  857 |       arg2 != F_SETLK64 &&
      |               ^~~~~~~~~
      |               F_SETLK
ptracesandbox.c:857:15: note: each undeclared identifier is reported only once for each function it appears in
ptracesandbox.c:858:15: error: 'F_SETLKW64' undeclared (first use in this function); did you mean 'F_SETLKW'?
  858 |       arg2 != F_SETLKW64 &&
      |               ^~~~~~~~~~
      |               F_SETLKW
make[4]: *** [Makefile:28: ptracesandbox.o] Error 1
make[4]: Leaving directory '/home/ub/lede/build_dir/target-i386_pentium4_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5'
make[3]: *** [Makefile:102: /home/ub/lede/build_dir/target-i386_pentium4_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5/.built] Error 2
make[3]: Leaving directory '/home/ub/lede/package/lean/vsftpd-alt'
time: package/lean/vsftpd-alt/compile#0.36#0.41#0.72
    ERROR: package/lean/vsftpd-alt failed to build.
make[2]: *** [package/Makefile:116: package/lean/vsftpd-alt/compile] Error 1
make[2]: Leaving directory '/home/ub/lede'
make[1]: *** [package/Makefile:110: /home/ub/lede/staging_dir/target-i386_pentium4_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ub/lede'
make: *** [/home/ub/lede/include/toplevel.mk:231: world] Error 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Ubuntu 20.04虚拟机；目标CPU：N2600，x86，Generac

### 详细日志

ptracesandbox.c: In function 'validate_fcntl':
ptracesandbox.c:857:15: error: 'F_SETLK64' undeclared (first use in this function); did you mean 'F_SETLK'?
  857 |       arg2 != F_SETLK64 &&
      |               ^~~~~~~~~
      |               F_SETLK
ptracesandbox.c:857:15: note: each undeclared identifier is reported only once for each function it appears in
ptracesandbox.c:858:15: error: 'F_SETLKW64' undeclared (first use in this function); did you mean 'F_SETLKW'?
  858 |       arg2 != F_SETLKW64 &&
      |               ^~~~~~~~~~
      |               F_SETLKW
make[4]: *** [Makefile:28: ptracesandbox.o] Error 1
make[4]: Leaving directory '/home/ub/lede/build_dir/target-i386_pentium4_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5'
make[3]: *** [Makefile:102: /home/ub/lede/build_dir/target-i386_pentium4_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5/.built] Error 2
make[3]: Leaving directory '/home/ub/lede/package/lean/vsftpd-alt'
time: package/lean/vsftpd-alt/compile#0.36#0.41#0.72
    ERROR: package/lean/vsftpd-alt failed to build.
make[2]: *** [package/Makefile:116: package/lean/vsftpd-alt/compile] Error 1
make[2]: Leaving directory '/home/ub/lede'
make[1]: *** [package/Makefile:110: /home/ub/lede/staging_dir/target-i386_pentium4_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ub/lede'
make: *** [/home/ub/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2024-03-25:
不知道多利用搜索相同issue https://github.com/coolsnowwolf/lede/issues/11852

jonwhoo, 2024-03-25:
不知该怎样处理，请说明一下，谢谢！

lxg2001, 2024-03-25:
> 不知该怎样处理，请说明一下，谢谢！

package/lean/vsftpd-alt/Makefile, 第35行：`TARGET_CFLAGS += -ffunction-sections -fdata-sections -flto -D_LARGEFILE64_SOURCE`

coolsnowwolf, 2024-03-25:
已修复

closed 2024-03-26 by coolsnowwolf