coolsnowwolf/lede #12033: x86 tailscale 编译错误

bob-zebedy, 2024-03-28:
### 详细叙述

x86 tailscale 编译错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

```
2024-03-28T05:27:47.8611202Z make[3]: Entering directory '/mnt/workdir/openwrt/feeds/packages/net/tailscale'
2024-03-28T05:27:47.9639830Z rm -f /mnt/workdir/openwrt/build_dir/target-x86_64_musl/tailscale-1.44.3/.built
2024-03-28T05:27:47.9655967Z touch /mnt/workdir/openwrt/build_dir/target-x86_64_musl/tailscale-1.44.3/.built_check
2024-03-28T05:27:47.9683089Z CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/mnt/workdir/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/mnt/workdir/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="tailscale.com/cmd/tailscale tailscale.com/cmd/tailscaled" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="tailscale.com/cmd/tailscale tailscale.com/cmd/tailscaled/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/mnt/workdir/openwrt/build_dir/target-x86_64_musl/tailscale-1.44.3" GO_BUILD_DIR="/mnt/workdir/openwrt/build_dir/target-x86_64_musl/tailscale-1.44.3/.go_work/build" GO_BUILD_BIN_DIR="/mnt/workdir/openwrt/build_dir/target-x86_64_musl/tailscale-1.44.3/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/mnt/workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="ccache_cc" CXX="ccache_cxx" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/mnt/workdir/openwrt/build_dir/target-x86_64_musl/tailscale-1.44.3=tailscale-1.44.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/mnt/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/mnt/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/mnt/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/mnt/workdir/openwrt/build_dir/target-x86_64_musl/tailscale-1.44.3=tailscale-1.44.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/mnt/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/mnt/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/mnt/workdir/openwrt/build_dir/target-x86_64_musl/tailscale-1.44.3/.go_work/build" GOCACHE="/mnt/workdir/openwrt/tmp/go-build" GOMODCACHE="/mnt/workdir/openwrt/dl/go-mod-cache" GOENV=off GOTOOLCHAIN=local /usr/bin/env bash ../../lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1711551492' -linkmode external -extldflags '-L/mnt/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/mnt/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-X 'tailscale.com/version.Long=1.44.3-1 (OpenWrt)' -X tailscale.com/version.Short=1.44.3 -buildid '1711551492' -linkmode external -extldflags '-L/mnt/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/mnt/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
2024-03-28T05:27:47.9704087Z Finding targets
2024-03-28T05:27:48.2134902Z 
2024-03-28T05:27:48.2135327Z Building targets
2024-03-28T05:27:48.3694909Z gvisor.dev/gvisor/pkg/sync
2024-03-28T05:27:48.3871706Z # gvisor.dev/gvisor/pkg/sync
2024-03-28T05:27:48.3873155Z ../../../../../dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20230504175454-7b0a1988a28f/pkg/sync/gate_unsafe.go:114:3: undefined: goready
2024-03-28T05:27:48.3874611Z ../../../../../dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20230504175454-7b0a1988a28f/pkg/sync/gate_unsafe.go:144:2: undefined: gopark
2024-03-28T05:27:48.3876141Z ../../../../../dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20230504175454-7b0a1988a28f/pkg/sync/rwmutex_unsafe.go:76:3: undefined: semacquire
2024-03-28T05:27:48.3878077Z ../../../../../dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20230504175454-7b0a1988a28f/pkg/sync/rwmutex_unsafe.go:102:4: undefined: semrelease
2024-03-28T05:27:48.3879357Z ../../../../../dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20230504175454-7b0a1988a28f/pkg/sync/rwmutex_unsafe.go:152:3: undefined: semacquire
2024-03-28T05:27:48.3880968Z ../../../../../dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20230504175454-7b0a1988a28f/pkg/sync/rwmutex_unsafe.go:179:3: undefined: semrelease
2024-03-28T05:27:48.3882517Z ../../../../../dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20230504175454-7b0a1988a28f/pkg/sync/rwmutex_unsafe.go:207:3: undefined: semrelease
2024-03-28T05:27:48.3883704Z ../../../../../dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20230504175454-7b0a1988a28f/pkg/sync/seqcount.go:87:6: undefined: canSpin
2024-03-28T05:27:48.3885516Z ../../../../../dl/go-mod-cache/gvisor.dev/gvisor@v0.0.0-20230504175454-7b0a1988a28f/pkg/sync/seqcount.go:89:4: undefined: doSpin
2024-03-28T05:27:49.6685137Z 
2024-03-28T05:27:49.7214026Z make[3]: *** [Makefile:105: /mnt/workdir/openwrt/build_dir/target-x86_64_musl/tailscale-1.44.3/.built] Error 1
2024-03-28T05:27:49.7215057Z make[3]: Leaving directory '/mnt/workdir/openwrt/feeds/packages/net/tailscale'
2024-03-28T05:27:49.7218084Z time: package/feeds/packages/tailscale/compile#1.81#0.61#1.86
2024-03-28T05:27:49.7223038Z     ERROR: package/feeds/packages/tailscale failed to build.
2024-03-28T05:27:49.7225333Z make[2]: *** [package/Makefile:116: package/feeds/packages/tailscale/compile] Error 1
2024-03-28T05:27:49.7230775Z make[2]: Leaving directory '/mnt/workdir/openwrt'
2024-03-28T05:27:49.7248516Z make[1]: *** [package/Makefile:110: /mnt/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-03-28T05:27:49.7252567Z make[1]: Leaving directory '/mnt/workdir/openwrt'
2024-03-28T05:27:49.7263566Z make: *** [/mnt/workdir/openwrt/include/toplevel.mk:231: world] Error 2
2024-03-28T05:27:49.7304045Z ##[error]Process completed with exit code 2.
```

WYC-2020, 2024-03-28:
升级一下 用op官方的替换掉

WYC-2020, 2024-03-28:
替换后防火墙的模式可能要改下 net/tailscale/files/tailscale.conf 这个配置文件里面的 改成iptables l这边还不是nftables

closed 2024-03-29 by bob-zebedy

KevinMX, 2024-04-25:
https://github.com/coolsnowwolf/packages/pull/797