coolsnowwolf/lede #12039: uboot-sunxi failed to build

Gabrielxzx, 2024-03-30:
### 详细叙述

编译sunxi处理器的板子，编译至uboot-sunxi时失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

orangepi_zero / orangepi_zero_plus

### 详细日志

[job-logs.txt](https://github.com/coolsnowwolf/lede/files/14812013/job-logs.txt)

WYC-2020, 2024-04-01:
目测是这个提交导致的，暂时的话撤销掉这个提交吧 [5c5b7af071ac3a429a40162dce9b777bc37a7fc9](https://github.com/coolsnowwolf/lede/commit/5c5b7af071ac3a429a40162dce9b777bc37a7fc9) 应该不会影响你目前编译的这个型号

WYC-2020, 2024-04-01:
如果不想撤回也可以这样
[image]
 但是不确定功能是否正常，不怕死的可以试试，最好就是等l修复因为他添加的所以应该是知道如何修的

Gabrielxzx, 2024-04-01:
> 如果不想撤回也可以这样 [...]

好的，感谢。
知道有人关注这个就好。

coolsnowwolf, 2024-04-05:
修复了

closed 2024-04-06 by Gabrielxzx