coolsnowwolf/lede #12041: 6.6编译错误

joyplayer, 2024-04-02:
### 详细叙述

SHELL= flock /openwrt/Lede_R22.12.2/tmp/.root-copy.flock -c 'cp -fpR /openwrt/Lede_R22.12.2/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /openwrt/Lede_R22.12.2/staging_dir/target-x86_64_musl/root-x86/'
touch /openwrt/Lede_R22.12.2/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
NOTICE: module '/openwrt/Lede_R22.12.2/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/drivers/gpu/drm/drm.ko' is built-in.
NOTICE: module '/openwrt/Lede_R22.12.2/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/drivers/gpu/drm/drm_panel_orientation_quirks.ko' is built-in.
ERROR: module '/openwrt/Lede_R22.12.2/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/drivers/gpu/drm/drm_exec.ko' is missing.
make[3]: *** [modules/video.mk:300: /openwrt/Lede_R22.12.2/bin/targets/x86/64/packages/kmod-drm_6.6.23-1_x86_64.ipk] Error 1

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

SHELL= flock /openwrt/Lede_R22.12.2/tmp/.root-copy.flock -c 'cp -fpR /openwrt/Lede_R22.12.2/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /openwrt/Lede_R22.12.2/staging_dir/target-x86_64_musl/root-x86/'
touch /openwrt/Lede_R22.12.2/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
NOTICE: module '/openwrt/Lede_R22.12.2/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/drivers/gpu/drm/drm.ko' is built-in.
NOTICE: module '/openwrt/Lede_R22.12.2/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/drivers/gpu/drm/drm_panel_orientation_quirks.ko' is built-in.
ERROR: module '/openwrt/Lede_R22.12.2/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.23/drivers/gpu/drm/drm_exec.ko' is missing.
make[3]: *** [modules/video.mk:300: /openwrt/Lede_R22.12.2/bin/targets/x86/64/packages/kmod-drm_6.6.23-1_x86_64.ipk] Error 1

WYC-2020, 2024-04-02:
make menuconfig 保存 然后编译一下内核 在编译模块 试试呢 我这边刚试了一切正常

joyplayer, 2024-04-06:
> make menuconfig 保存 然后编译一下内核 在编译模块 试试呢 我这边刚试了一切正常

编译两次正常了

closed 2024-04-06 by joyplayer