coolsnowwolf/lede #12042: qihoo_360v6问题

xianfanjie, 2024-04-02:
### 详细叙述

编译成功刷入固件后 连上WiFi进入后台led灯突然不亮了，WiFi也突然没有了，进不去web管理界面，拔电重启后又是一样的问题，再次重启后问题没有复现，进入web管理界面找到网络接口 修改协议为 pppoe拨号 无法成功，上不了网，led好像只有绿灯，重启的时候也是绿灯闪烁不是红灯。我之前使用的是4.4.60内核的 openwrt

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

qualcommax-ipq60xx-qihoo_360v6

### 详细日志

系统日志：
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x51af8014]
Mon Apr  1 21:32:24 2024 kern.notice kernel: [    0.000000] Linux version 6.1.83 (runner@fv-az1456-93) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6528-5d367a85b) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Apr  1 13:31:46 2024
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Machine model: Qihoo 360V6
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Zone ranges:
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000]   DMA32    empty
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000]   Normal   empty
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Movable zone start for each node
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Early memory node ranges
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000040000000-0x000000004a0fffff]
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004a100000-0x000000004a4fffff]
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004a500000-0x000000004a5fffff]
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004a600000-0x000000004e3fffff]
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000]   node   0: [mem 0x000000004e400000-0x000000005fffffff]
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] psci: probing for conduit method from DT.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] psci: PSCIv1.0 detected in firmware.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] psci: Using standard PSCI v0.2 function IDs
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] psci: MIGRATE_INFO_TYPE not supported.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] psci: SMC Calling Convention v1.0
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] percpu: Embedded 18 pages/cpu s34536 r8192 d31000 u73728
Mon Apr  1 21:32:24 2024 kern.debug kernel: [    0.000000] pcpu-alloc: s34536 r8192 d31000 u73728 alloc=18*4096
Mon Apr  1 21:32:24 2024 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] CPU features: detected: Spectre-v4
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] alternatives: applying boot alternatives
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
Mon Apr  1 21:32:24 2024 kern.notice kernel: [    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Memory: 433664K/524288K available (7808K kernel code, 870K rwdata, 2220K rodata, 1280K init, 276K bss, 90624K reserved, 0K cma-reserved)
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] Root IRQ handler: gic_handle_irq
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] arch_timer: cp15 and mmio timer(s) running at 24.00MHz (virt/virt).
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000070] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000084] pid_max: default: 32768 minimum: 301
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000541] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.000554] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.002403] cblist_init_generic: Setting adjustable number of callback queues.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.002411] cblist_init_generic: Setting shift to 2 and lim to 1.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.002611] rcu: Hierarchical SRCU implementation.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.002616] rcu: 	Max phase no-delay instances is 1000.
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.003281] smp: Bringing up secondary CPUs ...
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.003929] Detected VIPT I-cache on CPU1
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.004057] CPU1: Booted secondary processor 0x0000000001 [0x51af8014]
Mon Apr  1 21:32:24 2024 kern.info kernel: [    0.004711] Detected VIPT I-cache on CPU2
[... 401 lines cut ...]
Mon Apr  1 21:32:56 2024 daemon.info hostapd: wlan0: STA e4:a4:71:8b:a8:71 IEEE 802.11: authenticated
Mon Apr  1 21:32:56 2024 daemon.info hostapd: wlan0: STA e4:a4:71:8b:a8:71 IEEE 802.11: associated (aid 1)
Mon Apr  1 21:32:56 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED e4:a4:71:8b:a8:71 auth_alg=open
Mon Apr  1 21:32:56 2024 daemon.info hostapd: wlan0: STA e4:a4:71:8b:a8:71 RADIUS: starting accounting session AC6CCA9219129536
Mon Apr  1 21:33:12 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED e4:a4:71:8b:a8:71
Mon Apr  1 21:33:16 2024 daemon.info hostapd: wlan0: STA e4:a4:71:8b:a8:71 IEEE 802.11: authenticated
Mon Apr  1 21:33:16 2024 daemon.info hostapd: wlan0: STA e4:a4:71:8b:a8:71 IEEE 802.11: associated (aid 1)
Mon Apr  1 21:33:16 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED e4:a4:71:8b:a8:71 auth_alg=open
Mon Apr  1 21:33:16 2024 daemon.info hostapd: wlan0: STA e4:a4:71:8b:a8:71 RADIUS: starting accounting session CEEF7F447EFB5487
Mon Apr  1 21:33:22 2024 daemon.err uhttpd[2357]: sh: out of range
Mon Apr  1 21:33:22 2024 daemon.err uhttpd[2357]: sh: out of range
Mon Apr  1 21:33:27 2024 daemon.err uhttpd[2357]: sh: out of range
Mon Apr  1 21:33:32 2024 daemon.err uhttpd[2357]: sh: out of range
Mon Apr  1 21:33:37 2024 daemon.err uhttpd[2357]: sh: out of range
Mon Apr  1 21:34:03 2024 daemon.err uhttpd[2357]: sh: out of range
Mon Apr  1 21:34:03 2024 daemon.err uhttpd[2357]: sh: out of range
Mon Apr  1 21:34:20 2024 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED e4:a4:71:8b:a8:71
Mon Apr  1 21:34:42 2024 daemon.info hostapd: wlan0: STA 4e:b6:43:ec:f1:93 IEEE 802.11: authenticated
Mon Apr  1 21:34:42 2024 daemon.info hostapd: wlan0: STA 4e:b6:43:ec:f1:93 IEEE 802.11: associated (aid 1)
Mon Apr  1 21:34:42 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 4e:b6:43:ec:f1:93 auth_alg=open
Mon Apr  1 21:34:42 2024 daemon.info hostapd: wlan0: STA 4e:b6:43:ec:f1:93 RADIUS: starting accounting session 4F9D94B61C9AA3F2
Mon Apr  1 21:34:53 2024 daemon.info hostapd: wlan0: STA e4:a4:71:8b:a8:71 IEEE 802.11: authenticated
Mon Apr  1 21:34:53 2024 daemon.info hostapd: wlan0: STA e4:a4:71:8b:a8:71 IEEE 802.11: associated (aid 2)
Mon Apr  1 21:34:53 2024 daemon.notice hostapd: wlan0: AP-STA-CONNECTED e4:a4:71:8b:a8:71 auth_alg=open
Mon Apr  1 21:34:53 2024 daemon.info hostapd: wlan0: STA e4:a4:71:8b:a8:71 RADIUS: starting accounting session B595CB6A1E4CF29E

xianfanjie, 2024-04-02:
太长了 放一起发不出来
附上内核日志：

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x51af8014]
[    0.000000] Linux version 6.1.83 (runner@fv-az1456-93) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6528-5d367a85b) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Apr  1 13:31:46 2024
[    0.000000] Machine model: Qihoo 360V6
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000004a0fffff]
[    0.000000]   node   0: [mem 0x000000004a100000-0x000000004a4fffff]
[    0.000000]   node   0: [mem 0x000000004a500000-0x000000004a5fffff]
[    0.000000]   node   0: [mem 0x000000004a600000-0x000000004e3fffff]
[    0.000000]   node   0: [mem 0x000000004e400000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 18 pages/cpu s34536 r8192 d31000 u73728
[    0.000000] pcpu-alloc: s34536 r8192 d31000 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 ubi.mtd=rootfs root=mtd:ubi_rootfs rootfstype=squashfs rootwait root=/dev/ubiblock0_1
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 433664K/524288K available (7808K kernel code, 870K rwdata, 2220K rodata, 1280K init, 276K bss, 90624K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 24.00MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000070] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000084] pid_max: default: 32768 minimum: 301
[    0.000541] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.000554] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)
[    0.002403] cblist_init_generic: Setting adjustable number of callback queues.
[    0.002411] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.002611] rcu: Hierarchical SRCU implementation.
[    0.002616] rcu: 	Max phase no-delay instances is 1000.
[    0.003281] smp: Bringing up secondary CPUs ...
[    0.003929] Detected VIPT I-cache on CPU1
[    0.004057] CPU1: Booted secondary processor 0x0000000001 [0x51af8014]
[    0.004711] Detected VIPT I-cache on CPU2
[    0.004806] CPU2: Booted secondary processor 0x0000000002 [0x51af8014]
[    0.005424] Detected VIPT I-cache on CPU3
[    0.005516] CPU3: Booted secondary processor 0x0000000003 [0x51af8014]
[    0.005584] smp: Brought up 1 node, 4 CPUs
[    0.005592] SMP: Total of 4 processors activated.
[    0.005598] CPU features: detected: 32-bit EL0 Support
[    0.005602] CPU features: detected: CRC32 instructions
[    0.005665] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.005672] CPU: All CPU(s) started at EL1
[    0.005675] alternatives: applying system-wide alternatives
[    0.013031] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.013057] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[... 296 lines cut ...]
[   16.588983] br-lan: port 2(lan2) entered blocking state
[   16.589030] br-lan: port 2(lan2) entered disabled state
[   16.593415] device lan2 entered promiscuous mode
[   16.599953] nss-dp 3a001600.dp4 lan3: PHY Link up speed: 1000
[   16.603894] br-lan: port 3(lan3) entered blocking state
[   16.608766] br-lan: port 3(lan3) entered disabled state
[   16.614169] device lan3 entered promiscuous mode
[   17.609565] br-lan: port 3(lan3) entered blocking state
[   17.609617] br-lan: port 3(lan3) entered forwarding state
[   17.613913] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   17.868078] br-lan: port 4(wlan0) entered blocking state
[   17.868122] br-lan: port 4(wlan0) entered disabled state
[   17.872677] device wlan0 entered promiscuous mode
[... cut ...]
[   17.902558] device wlan1 entered promiscuous mode
[   17.907577] br-lan: port 5(wlan1) entered blocking state
[   17.912102] br-lan: port 5(wlan1) entered forwarding state
[   17.981022] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   18.649564] br-lan: port 4(wlan0) entered disabled state
[   23.358182] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   23.358388] br-lan: port 4(wlan0) entered blocking state
[   23.363668] br-lan: port 4(wlan0) entered forwarding state

wxzy322, 2024-04-03:
我的360 v6  无线别的啥都正常  就是会突然关机  和停电一样  也是6.1内核

closed 2024-12-03 by xianfanjie