coolsnowwolf/lede #12043: 6.6内核不支持 Intel Corporation Ethernet Controller X710网卡

cococle, 2024-04-02:
### 详细叙述

6.6内核不支持 Intel Corporation Ethernet Controller X710网卡

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

6.6内核不支持 Intel Corporation Ethernet Controller X710网卡

qingtian110, 2024-04-02:
假的，6.6 内核支持 ie40

[image]

WYC-2020, 2024-04-03:
有这玩意还不支持 ?
[image]

qingtian110, 2024-04-05:
> 有这玩意还不支持 ? [...]

它需要的是  kmod-ie40    不是vf

closed 2024-04-10 by coolsnowwolf