coolsnowwolf/lede #12045: 编译失败

yidianyipie, 2024-04-03:
### 详细叙述

Hi Team,
最近添加了snort3插件一直编译出错,有解决的办法不

BR
Robin

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

```python
mkdir -p /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/build_dir/target-x86_64_musl/luci-theme-netgear/.pkgdir/luci-theme-netgear/. /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.luci-theme-netgear_installed
echo "luci-theme-netgear" >> /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/luci-theme-netgear.default.install
make[3]: Leaving directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/feeds/luci/themes/luci-theme-netgear'
time: package/feeds/luci/luci-theme-netgear/compile#0.16#0.04#0.19
make[3]: Entering directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/feeds/packages/utils/bc'
mkdir -p /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/build_dir/target-x86_64_musl/bc-1.06.95/.pkgdir/bc/. /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.bc_installed
echo "bc" >> /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/bc.default.install
make[3]: Leaving directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/feeds/packages/utils/bc'
time: package/feeds/packages/bc/compile#0.10#0.02#0.12
make[3]: Entering directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/feeds/packages/utils/coremark'
mkdir -p /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/build_dir/target-x86_64_musl/coremark-eefc986ebd3452d6adde22eafaff3e5c859f29e4/.pkgdir/coremark/. /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.coremark_installed
echo "coremark" >> /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/coremark.default.install
make[3]: Leaving directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/feeds/packages/utils/coremark'
time: package/feeds/packages/coremark/compile#0.11#0.02#0.12
make[3]: Entering directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/package/libs/sysfsutils'
mkdir -p /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
echo '2' | cmp -s - /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/libsysfs.version || { echo '2' > /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/libsysfs.version;  } 
SHELL= flock /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/build_dir/target-x86_64_musl/sysfsutils-2.1.0/.pkgdir/libsysfs/. /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libsysfs_installed
echo "libsysfs" >> /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/sysfsutils.default.install
mkdir -p /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/build_dir/target-x86_64_musl/sysfsutils-2.1.0/.pkgdir/sysfsutils/. /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.sysfsutils_installed
echo "sysfsutils" >> /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/sysfsutils.default.install
make[3]: Leaving directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/package/libs/sysfsutils'
time: package/libs/sysfsutils/compile#0.13#0.02#0.14
make[3]: Entering directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/feeds/packages/utils/lm-sensors'
mkdir -p /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
echo '5' | cmp -s - /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/libsensors.version || { echo '5' > /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/libsensors.version;  } 
SHELL= flock /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/build_dir/target-x86_64_musl/lm-sensors-3-6-0/.pkgdir/libsensors/. /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.libsensors_installed
echo "libsensors" >> /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/lm-sensors.default.install
mkdir -p /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/build_dir/target-x86_64_musl/lm-sensors-3-6-0/.pkgdir/lm-sensors/. /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.lm-sensors_installed
echo "lm-sensors" >> /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/lm-sensors.default.install
make[3]: Leaving directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/feeds/packages/utils/lm-sensors'
time: package/feeds/packages/lm-sensors/compile#0.11#0.03#0.13
make[3]: Entering directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/feeds/packages/admin/htop'
mkdir -p /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/build_dir/target-x86_64_musl/htop-3.3.0/.pkgdir/htop/. /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.htop_installed
echo "htop" >> /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/pkginfo/htop.default.install
make[3]: Leaving directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/feeds/packages/admin/htop'
time: package/feeds/packages/htop/compile#0.12#0.01#0.12
[... 32 lines cut ...]
linux_sysfs.c: In function 'pci_device_linux_sysfs_write':
linux_sysfs.c:525:31: error: implicit declaration of function 'pwrite64'; did you mean 'pwrite'? [-Werror=implicit-function-declaration]
  525 |         const ssize_t bytes = pwrite64( fd, data_bytes, temp_size, offset );
      |                               ^~~~~~~~
      |                               pwrite
linux_sysfs.c:525:31: warning: nested extern declaration of 'pwrite64' [-Wnested-externs]
cc1: some warnings being treated as errors
make[6]: *** [Makefile:556: linux_sysfs.lo] Error 1
[... cut ...]
make[2]: *** [package/Makefile:116: package/feeds/packages/libpciaccess/compile] Error 1
make[2]: Leaving directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt'
make: *** [/home/runner/work/AutoBuild_Lede/AutoBuild_Lede/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

```

Beginner-Go, 2024-04-03:
再试试 应该修好了

WYC-2020, 2024-04-07:
[010-linux_sysfs-Use-pwrite-pread-instead-of-64bit-versions.patch](https://github.com/coolsnowwolf/lede/files/14895587/010-linux_sysfs-Use-pwrite-pread-instead-of-64bit-versions.patch)
按下图放
[image]

GreatMichaelLee, 2024-04-10:
> 那些工具能不能不要搞带版本号作为目录名的软连接.....服了，原因是install-sh是个软连接，映射到autoconf-1.15的目录名下，昨天给bump到1.16.5 这些个带目录的立马gg了
> 
> Making install in libelf /usr/bin/mkdir -p '/home/builder/lede_x86/staging_dir/host/lib' /home/builder/lede_x86/staging_dir/host/bin/install -c -m 644 libelf.a '/home/builder/lede_x86/staging_dir/host/lib' ( cd '/home/builder/lede_x86/staging_dir/host/lib' && ranlib libelf.a ) /usr/bin/mkdir -p '/home/builder/lede_x86/staging_dir/host/include' /home/builder/lede_x86/staging_dir/host/bin/install -c -m 644 libelf.h gelf.h nlist.h '/home/builder/lede_x86/staging_dir/host/include' /usr/bin/mkdir -p '/home/builder/lede_x86/staging_dir/host/include/elfutils' /home/builder/lede_x86/staging_dir/host/bin/install -c -m 644 elf-knowledge.h '/home/builder/lede_x86/staging_dir/host/include/elfutils' /usr/bin/env bash /home/builder/lede_x86/build_dir/host/elfutils-0.188/config/install-sh -d /home/builder/lede_x86/staging_dir/host/lib bash: /home/builder/lede_x86/build_dir/host/elfutils-0.188/config/install-sh: No such file or directory make[5]: *** [Makefile:1049: install] Error 127 make[4]: *** [Makefile:527: install-recursive] Error 1 make[4]: Leaving directory '/home/builder/lede_x86/build_dir/host/elfutils-0.188' make[3]: *** [Makefile:37: /home/builder/lede_x86/staging_dir/host/stamp/.elfutils_installed] Error 2 make[3]: Leaving directory '/home/builder/lede_x86/tools/elfutils' time: tools/elfutils/compile#31.73#14.06#46.35 ERROR: tools/elfutils failed to build. make[2]: *** [tools/Makefile:221: tools/elfutils/compile] Error 1 make[2]: Leaving directory '/home/builder/lede_x86' make[1]: *** [tools/Makefile:217: /home/builder/lede_x86/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2 make[1]: Leaving directory '/home/builder/lede_x86' make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2

[image]

GreatMichaelLee, 2024-04-10:
@WYC-2020 大佬，帮忙看看呗。。。。

WYC-2020, 2024-04-11:
别问 问就是自己姿势不对

GreatMichaelLee, 2024-04-11:
> 别问 问就是自己姿势不对

W哥教个姿势啊，我已经晕了。

WYC-2020, 2024-04-11:
clean compile

GreatMichaelLee, 2024-04-11:
> clean compile

不行呀，我感觉这个改动有问题，https://github.com/coolsnowwolf/lede/commit/c897f32a45d50b099f53a62be40b82fdb0edb839

WYC-2020, 2024-04-12:
> > clean compile
> 
> 不行呀，我感觉这个[c897f32](https://github.com/coolsnowwolf/lede/commit/c897f32a45d50b099f53a62be40b82fdb0edb839) 改动有问题，我revert了这个commit之后就没任何问题了。麻烦老板们再看看。

别说什么感觉 为啥我可以呢自己排查

GreatMichaelLee, 2024-04-12:
> > > clean compile
> > 
> > 
> > 不行呀，我感觉这个[c897f32](https://github.com/coolsnowwolf/lede/commit/c897f32a45d50b099f53a62be40b82fdb0edb839) 改动有问题，我revert了这个commit之后就没任何问题了。麻烦老板们再看看。
> 
> 别说什么感觉 为啥我可以呢自己排查

你没问题的话，你看看你的这个路径（staging_dir/host/share/automake-1.x）下的automake目录名x是1.15还是1.16?如果是1.16就会出我那个问题（我的就是变成了1.16），如果没变是1.15是可以编译过，但实际版本都升级1.16了目录名还是1.15那不是很奇怪？

我反正不排查了，revert掉这个东西完事。这种东西我是不懂怎么排查的，我只能根据commit差异猜测可能是改动对于目录路径的处理逻辑哪里有覆盖不全的场景，公共工具类的版本升级最好谨慎，特别是搞个带版本名的目录，个人愚见。

WYC-2020, 2024-04-13:
> > > > clean compile
> > > 
> > > 
> > > 不行呀，我感觉这个[c897f32](https://github.com/coolsnowwolf/lede/commit/c897f32a45d50b099f53a62be40b82fdb0edb839) 改动有问题，我revert了这个commit之后就没任何问题了。麻烦老板们再看看。
> > 
> > 
> > 别说什么感觉 为啥我可以呢自己排查
> 
> 你没问题的话，你看看你的这个路径（staging_dir/host/share/automake-1.x）下的automake目录名x是1.15还是1.16?如果是1.16就会出我那个问题（我的就是变成了1.16），如果没变是1.15是可以编译过，但实际版本都升级1.16了目录名还是1.15那不是很奇怪？
> 
> 我反正不排查了，revert掉这个东西完事。这种东西我是不懂怎么排查的，我只能根据commit差异猜测可能是改动对于目录路径的update处理逻辑哪里有覆盖不全或者逻辑有问题的场景，公共工具类的版本升级最好谨慎，特别是搞个带版本名的目录，个人愚见。

1.16

hzjnet, 2024-05-06:
确实是automake版本问题，升级后好几个应用编译都有问题，回到1.15.1版本就正常了

coolsnowwolf, 2026-09-05:
已处理 snort3 编译问题：当前包从 3.1.82.0 更新至上游 3.12.2.0，并删除已被新源码包含、会导致 patch reject 的旧 `include-cstdint` 补丁。新版本源码准备阶段已在本地通过；完整 compile 被工作树既有 Linux 6.12 配置缺失（pppox.ko）阻断，非 snort3 编译错误。独立提交 `3596a1df111caf2fa9c23a175e946b330d975f4b` 已推送至 packages/master。issue 已关闭。

closed 2026-09-05 by coolsnowwolf