coolsnowwolf/lede #12051: RK3328的U-BOOT编译失败

tmggg, 2024-04-04:
### 详细叙述

更新到最新代码，按二次编译操作后无法进行编译

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

RK3328-R2S

### 详细日志

make[3]: Entering directory '/home/tmgg/lede/package/boot/uboot-rockchip'
rm -f /home/tmgg/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.07/.built
touch /home/tmgg/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.07/.built_check
make  -C /home/tmgg/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.07 CROSS_COMPILE=aarch64-openwrt-linux-musl- DTC="/home/tmgg/lede/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.84/scripts/dtc/dtc" HOSTCC="gcc" HOSTCFLAGS="-O2 -I/home/tmgg/lede/staging_dir/host/include -I/home/tmgg/lede/staging_dir/hostpkg/include -I/home/tmgg/lede/staging_dir/target-aarch64_generic_musl/host/include -I/home/tmgg/lede/staging_dir/host/include -I/home/tmgg/lede/staging_dir/hostpkg/include -I/home/tmgg/lede/staging_dir/target-aarch64_generic_musl/host/include -std=gnu11" HOSTLDFLAGS="-L/home/tmgg/lede/staging_dir/host/lib -L/home/tmgg/lede/staging_dir/hostpkg/lib -L/home/tmgg/lede/staging_dir/target-aarch64_generic_musl/host/lib" LOCALVERSION="-OpenWrt-r6532-b171813f4" STAGING_PREFIX="/home/tmgg/lede/staging_dir/host" PKG_CONFIG_PATH="/home/tmgg/lede/staging_dir/host/lib/pkgconfig" PKG_CONFIG_LIBDIR="/home/tmgg/lede/staging_dir/host/lib/pkgconfig" PKG_CONFIG_EXTRAARGS="--static" V='' PATH=/home/tmgg/lede/staging_dir/host/bin:/home/tmgg/lede/staging_dir/target-aarch64_generic_musl/host/bin:/home/tmgg/lede/staging_dir/hostpkg/bin:/home/tmgg/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin:/home/tmgg/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin:/home/tmgg/lede/staging_dir/host/bin:/home/tmgg/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/bin:/home/tmgg/lede/staging_dir/host/bin:/home/tmgg/lede/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin BL31=/home/tmgg/lede/staging_dir/target-aarch64_generic_musl/image/rk322xh_bl31_v1.49.elf
make[4]: Entering directory '/home/tmgg/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.07'
  ENVC    include/generated/env.txt
  ENVP    include/generated/env.in
  ENVT    include/generated/environment.h
make -f ./scripts/Makefile.build obj=scripts/dtc
  CC      env/common.o
  AR      env/built-in.o
  CC      lib/efi_loader/dtbdump.o
  LD      lib/efi_loader/dtbdump_efi.so
  OBJCOPY lib/efi_loader/dtbdump.efi
  CC      lib/efi_loader/initrddump.o
  LD      lib/efi_loader/initrddump_efi.so
  OBJCOPY lib/efi_loader/initrddump.efi
  LD      u-boot
  OBJCOPY u-boot.srec
  OBJCOPY u-boot-nodtb.bin
  RELOC   u-boot-nodtb.bin
  CAT     u-boot-dtb.bin
  COPY    u-boot.bin
  SYM     u-boot.sym
  MKIMAGE u-boot.img
  DTOC    tpl/dts/dt-plat.c
  DTOC    include/generated/dt-structs-gen.h
  DTOC    include/generated/dt-decl.h
  PLAT    tpl/dts/dt-plat.o
  CC      tpl/drivers/ram/rockchip/sdram_rk3328.o
  AR      tpl/drivers/ram/rockchip/built-in.o
  AR      tpl/drivers/ram/built-in.o
  AR      tpl/drivers/built-in.o
  LD      tpl/u-boot-tpl
  OBJCOPY tpl/u-boot-tpl-nodtb.bin
  COPY    tpl/u-boot-tpl.bin
  SYM     tpl/u-boot-tpl.sym
  MKIMAGE u-boot-dtb.img
  BINMAN  .binman_stamp
Traceback (most recent call last):
  File "/home/tmgg/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.07/tools/binman/control.py", line 11, in <module>
    import importlib.resources
ModuleNotFoundError: No module named 'importlib.resources'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./tools/binman/binman", line 46, in <module>
    from binman import control
  File "/home/tmgg/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.07/tools/binman/control.py", line 14, in <module>
    import importlib_resources
ModuleNotFoundError: No module named 'importlib_resources'
Makefile:1111: recipe for target '.binman_stamp' failed
make[4]: *** [.binman_stamp] Error 1
make[4]: Leaving directory '/home/tmgg/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.07'
Makefile:377: recipe for target '/home/tmgg/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.07/.built' failed
make[3]: *** [/home/tmgg/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.07/.built] Error 2
make[3]: Leaving directory '/home/tmgg/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r2s-rk3328/compile#7.78#1.83#12.41
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r2s-rk3328).
package/Makefile:114: recipe for target 'package/boot/uboot-rockchip/compile' failed
make[2]: *** [package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/tmgg/lede'
package/Makefile:110: recipe for target '/home/tmgg/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: *** [/home/tmgg/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tmgg/lede'
/home/tmgg/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

lxg2001, 2024-04-04:
`pip install importlib-resources`

lxg2001, 2024-04-04:
`pip3 install importlib-resources`

lxg2001, 2024-04-04:
报错信息显示，当尝试运行u-boot的binman工具时，Python无法导入importlib.resources模块。这通常意味着你的Python环境中缺少这个模块。

importlib.resources是Python 3.7及以后版本中引入的一个标准库模块，如果你的Python版本低于3.7，则会出现这个错误。另外，如果你的Python版本符合要求，可能是你的Python环境不完整或损坏。

为了解决这个问题，你可以尝试以下方法：

检查Python版本：
确保你的Python版本至少是3.7。你可以在终端中运行python3 --version（或python --version，取决于你的系统配置）来检查你的Python版本。

安装或更新Python：
如果你的Python版本低于3.7，你需要升级到更高的版本。你可以从Python官方网站下载并安装新版本的Python。

安装importlib_resources：
如果你使用的是Python 3.6或更低版本，并且无法升级Python，你可以通过pip安装importlib_resources作为一个独立的包：

bash
复制
pip3 install importlib_resources

或者，如果你的系统使用的是Python 2.x，你可能需要安装一个兼容的importlib_resources版本或使用其他方式来解决这个问题，因为importlib_resources是为Python 3设计的。

检查环境：
如果你已经在使用Python 3.7或更高版本，但仍然遇到这个问题，可能是你的Python环境配置有问题。确保你的环境变量（如PYTHONPATH）设置正确，并且Python安装没有损坏。

使用虚拟环境：
为了避免系统Python环境中的冲突，你可以创建一个Python虚拟环境，并在其中安装所需的依赖项。这可以通过venv或virtualenv模块来完成。

重新编译U-Boot：
在解决了Python环境问题后，你可能需要重新编译U-Boot以确保所有依赖项都已正确安装。

如果上述方法都不能解决问题，你可能需要查看U-Boot的文档或寻求社区的帮助，以获取针对你特定环境的更具体的指导。

lxg2001, 2024-04-04:
报错信息显示，当尝试运行u-boot的binman工具时，Python无法导入importlib.resources模块。这通常意味着你的Python环境中缺少这个模块。

importlib.resources是Python 3.7及以后版本中引入的一个标准库模块，如果你的Python版本低于3.7，则会出现这个错误。另外，如果你的Python版本符合要求，可能是你的Python环境不完整或损坏。

为了解决这个问题，你可以尝试以下方法：

检查Python版本：
确保你的Python版本至少是3.7。你可以在终端中运行python3 --version（或python --version，取决于你的系统配置）来检查你的Python版本。

安装或更新Python：
如果你的Python版本低于3.7，你需要升级到更高的版本。你可以从Python官方网站下载并安装新版本的Python。

安装importlib_resources：
如果你使用的是Python 3.6或更低版本，并且无法升级Python，你可以通过pip安装importlib_resources作为一个独立的包：

```bash
复制
pip3 install importlib_resources
```

或者，如果你的系统使用的是Python 2.x，你可能需要安装一个兼容的importlib_resources版本或使用其他方式来解决这个问题，因为importlib_resources是为Python 3设计的。

检查环境：
如果你已经在使用Python 3.7或更高版本，但仍然遇到这个问题，可能是你的Python环境配置有问题。确保你的环境变量（如PYTHONPATH）设置正确，并且Python安装没有损坏。

使用虚拟环境：
为了避免系统Python环境中的冲突，你可以创建一个Python虚拟环境，并在其中安装所需的依赖项。这可以通过venv或virtualenv模块来完成。

重新编译U-Boot：
在解决了Python环境问题后，你可能需要重新编译U-Boot以确保所有依赖项都已正确安装。

如果上述方法都不能解决问题，你可能需要查看U-Boot的文档或寻求社区的帮助，以获取针对你特定环境的更具体的指导。

lxg2001, 2024-04-04:
报错信息显示，当尝试运行u-boot的binman工具时，Python无法导入importlib.resources模块。这通常意味着你的Python环境中缺少这个模块。

importlib.resources是Python 3.7及以后版本中引入的一个标准库模块，如果你的Python版本低于3.7，则会出现这个错误。另外，如果你的Python版本符合要求，可能是你的Python环境不完整或损坏。

为了解决这个问题，你可以尝试以下方法：

检查Python版本：
确保你的Python版本至少是3.7。你可以在终端中运行python3 --version（或python --version，取决于你的系统配置）来检查你的Python版本。

安装或更新Python：
如果你的Python版本低于3.7，你需要升级到更高的版本。你可以从Python官方网站下载并安装新版本的Python。

安装importlib_resources：
如果你使用的是Python 3.6或更低版本，并且无法升级Python，你可以通过pip安装importlib_resources作为一个独立的包：

```bash
pip3 install importlib_resources
```

或者，如果你的系统使用的是Python 2.x，你可能需要安装一个兼容的importlib_resources版本或使用其他方式来解决这个问题，因为importlib_resources是为Python 3设计的。

检查环境：
如果你已经在使用Python 3.7或更高版本，但仍然遇到这个问题，可能是你的Python环境配置有问题。确保你的环境变量（如PYTHONPATH）设置正确，并且Python安装没有损坏。

使用虚拟环境：
为了避免系统Python环境中的冲突，你可以创建一个Python虚拟环境，并在其中安装所需的依赖项。这可以通过venv或virtualenv模块来完成。

重新编译U-Boot：
在解决了Python环境问题后，你可能需要重新编译U-Boot以确保所有依赖项都已正确安装。

如果上述方法都不能解决问题，你可能需要查看U-Boot的文档或寻求社区的帮助，以获取针对你特定环境的更具体的指导。

WYC-2020, 2024-04-07:
放此处
[image]
[318-revert-Use-importlib-to-find-the-help.patch](https://github.com/coolsnowwolf/lede/files/14896014/318-revert-Use-importlib-to-find-the-help.patch)

tmggg, 2024-04-07:
> 放此处 [...]

感谢

tmggg, 2024-04-08:
编译成功

closed 2024-04-08 by tmggg

2253845067, 2024-04-10:
> 编译成功

[image]
是加到这个位置吗，我加了还是报错
[image]