coolsnowwolf/lede #12056: h6xk rk3568 编译失败，**Filename 'rockchip-tpl' not found**

ghmole, 2024-04-07:
### 详细叙述

when build ./lede/build_dir/target-aarch64_generic_musl/u-boot-opc-h68k-rk3568/u-boot-2023.07, **'rockchip-tpl' not found**

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

h68k

### 详细日志

  aarch64-openwrt-linux-musl-gcc -E -Wp,-MD,spl/.u-boot-spl.lds.d -D__KERNEL__ -D__UBOOT__ -DCONFIG_SPL_BUILD -D__ARM__          -mstrict-align -ffunction-sections -fdata-sections -fno-common -ffixed-x18  -mgeneral-regs-only    -mbranch-protection=none -pipe -march=armv8-a+crc -D__LINUX_ARM_ARCH__=8  -I./arch/arm/mach-rockchip/include -Ispl/include -Iinclude   -I./arch/arm/include -include ./include/linux/kconfig.h -nostdinc -isystem /root/works/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/include -include ./include/u-boot/u-boot.lds.h -include ./include/config.h -DCPUDIR=arch/arm/cpu/armv8  -DIMAGE_MAX_SIZE=0x40000 -DIMAGE_TEXT_BASE=0x0 -ansi -D__ASSEMBLY__ -x assembler-with-cpp -std=c99 -P -o spl/u-boot-spl.lds arch/arm/cpu/armv8/u-boot-spl.lds
  ( cd spl && aarch64-openwrt-linux-musl-ld.bfd -z noexecstack --no-warn-rwx-segments -T u-boot-spl.lds --gc-sections -Bstatic --gc-sections  --no-dynamic-linker --build-id=none -Ttext 0x0 arch/arm/cpu/armv8/start.o --whole-archive arch/arm/mach-rockchip/built-in.o arch/arm/cpu/armv8/built-in.o arch/arm/cpu/built-in.o arch/arm/lib/built-in.o board/rockchip/evb_rk3568/built-in.o common/spl/built-in.o common/init/built-in.o boot/built-in.o common/built-in.o cmd/built-in.o env/built-in.o lib/built-in.o disk/built-in.o drivers/built-in.o dts/built-in.o fs/built-in.o  --no-whole-archive -L /root/works/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0 -lgcc -Map u-boot-spl.map -o u-boot-spl )
  aarch64-openwrt-linux-musl-objcopy -j .text -j .secure_text -j .secure_data -j .rodata -j .data -j __u_boot_list -j .rela.dyn -j .got -j .got.plt -j .binman_sym_table -j .text_rest -j .dtb.init.rodata -j .efi_runtime -j .efi_runtime_rel  -O binary  spl/u-boot-spl spl/u-boot-spl-nodtb.bin
make -f ./scripts/Makefile.build obj=spl/dts spl_dtbs
mkdir -p spl/dts/
  ./tools/fdtgrep -b bootph-all -b bootph-pre-ram -b u-boot,dm-pre-reloc -b u-boot,dm-spl -RT dts/dt.dtb -n /chosen -n /config -O dtb | ./tools/fdtgrep -r -O dtb - -o spl/dts/dt-spl.dtb -P bootph-all -P bootph-pre-ram -P bootph-pre-sram -P bootph-verify -P u-boot,dm-pre-reloc -P u-boot,dm-spl -P u-boot,dm-tpl -P u-boot,dm-vpl -P clock-names -P interrupt-parent -P assigned-clocks -P assigned-clock-rates -P assigned-clock-parents
  cp spl/dts/dt-spl.dtb spl/u-boot-spl.dtb
  cat spl/u-boot-spl-nodtb.bin spl/u-boot-spl.dtb > spl/u-boot-spl-dtb.bin
  cp spl/u-boot-spl-dtb.bin spl/u-boot-spl.bin
  aarch64-openwrt-linux-musl-objdump -t spl/u-boot-spl > spl/u-boot-spl.sym
  ./tools/mkimage -f auto -A arm -T firmware -C none -O u-boot -a 0x00a00000 -e 0x00a00000 -p 0x0 -n "U-Boot 2023.07-rc6-OpenWrt-r6542+27-3de103535 for evb_rk3568 board" -E  -b arch/arm/dts/rk3568-opc-h68k.dtb  -d u-boot-nodtb.bin u-boot.img >/dev/null  && cat /dev/null
  cp dts/dt.dtb u-boot.dtb
  ./tools/mkimage -f auto -A arm -T firmware -C none -O u-boot -a 0x00a00000 -e 0x00a00000 -p 0x0 -n "U-Boot 2023.07-rc6-OpenWrt-r6542+27-3de103535 for evb_rk3568 board" -E  -b arch/arm/dts/rk3568-opc-h68k.dtb  -d u-boot-nodtb.bin u-boot-dtb.img >/dev/null  && cat /dev/null
  ./tools/binman/binman   --toolpath ./tools  build -u -d u-boot.dtb -O . -m  -I . -I . -I ./board/rockchip/evb_rk3568 -I arch/arm/dts -a of-list="rk3568-opc-h68k"  -a atf-bl31-path=/root/works/lede/staging_dir/target-aarch64_generic_musl/image/rk3568_bl31_v1.43.elf -a tee-os-path= -a opensbi-path= -a default-dt="rk3568-opc-h68k" -a scp-path= -a rockchip-tpl-path= -a spl-bss-pad= -a tpl-bss-pad=1 -a spl-dtb=y -a tpl-dtb= -a pre-load-key-path=
binman: **Filename 'rockchip-tpl' not found** in input path (.,.,./board/rockchip/evb_rk3568,arch/arm/dts) (cwd='/root/works/lede/build_dir/target-aarch64_generic_musl/u-boot-opc-h68k-rk3568/u-boot-2023.07')
make[4]: *** [Makefile:1111: .binman_stamp] Error 1
make[4]: Leaving directory '/root/works/lede/build_dir/target-aarch64_generic_musl/u-boot-opc-h68k-rk3568/u-boot-2023.07'
make[3]: *** [Makefile:392: /root/works/lede/build_dir/target-aarch64_generic_musl/u-boot-opc-h68k-rk3568/u-boot-2023.07/.built] Error 2
make[3]: Leaving directory '/root/works/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/opc-h68k-rk3568/compile#89.75#13.76#99.12
    ERROR: package/boot/uboot-rockchip failed to build (build variant: opc-h68k-rk3568).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/root/works/lede'
make[1]: *** [package/Makefile:110: /root/works/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/works/lede'
make: *** [/root/works/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2024-04-07:
我这边一切正常呢
[image]
你看看是不是 和这个issue一样 这里又问题https://github.com/coolsnowwolf/lede/issues/12051  是的话吧那个patches 放好 在编译试试

ghmole, 2024-04-09:
是这个问题，当这个patch放进去后可以编译了

closed 2024-04-09 by ghmole

2253845067, 2024-04-10:
> 是这个问题，当这个patch放进去后可以编译了

[image]
[image]
大佬看看是哪里不对吗，我编译的69k

ghmole, 2024-04-10:
make clean 后再重新编

ghmole, 2024-04-10:
这不是一个问题，#12063 是个新问题，我觉得是有个文件名写错了，应该是1.21.bin