coolsnowwolf/lede #12061: br-lan: received packet on eth2 with own address as source address

eearphon, 2024-04-09:
### 详细叙述

J4125 i210 ESxi 8.0，eth0虚拟网卡(ESXi里查看mac 00:0c:29:f1:e6:85)，eth1(WAN)、eth2(桥接到br-lan)直通，log偶尔出现如下提示，毫无规律，但感觉没影响使用

root@OpenWrt:~# date
Tue Apr  9 11:15:27 CST 2024
root@OpenWrt:~# 
root@OpenWrt:~# logread | grep 'with own address as source address'
Mon Apr  8 17:47:59 2024 kern.warn kernel: [26398.559529] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0)
Tue Apr  9 07:54:01 2024 kern.warn kernel: [77155.930527] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0)
Tue Apr  9 07:54:01 2024 kern.warn kernel: [77156.074518] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0)
Tue Apr  9 07:54:01 2024 kern.warn kernel: [77156.204865] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0)
Tue Apr  9 07:54:01 2024 kern.warn kernel: [77156.335611] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0)
Tue Apr  9 07:54:01 2024 kern.warn kernel: [77156.465949] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0)
Tue Apr  9 07:54:01 2024 kern.warn kernel: [77156.596556] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0)
Tue Apr  9 07:54:02 2024 kern.warn kernel: [77156.727378] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0)
root@OpenWrt:~# 
root@OpenWrt:~# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP group default qlen 1000
    link/ether 00:0c:29:f1:e6:85 brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 5d:62:e3:57:2e:51 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::c656:4fff:fe27:2751/64 scope link 
       valid_lft forever preferred_lft forever
4: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP group default qlen 1000
    link/ether 5d:62:e3:57:2e:52 brd ff:ff:ff:ff:ff:ff
5: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 80:3a:16:db:2c:89 brd ff:ff:ff:ff:ff:ff
6: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 00:0c:29:f1:e6:85 brd ff:ff:ff:ff:ff:ff
    inet 192.168.8.1/24 brd 192.168.8.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 2408:744:5803:2e8f::1/60 scope global dynamic noprefixroute 
       valid_lft 45169sec preferred_lft 45169sec
    inet6 f8a1:65e3:c0af::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::20c:29ff:fe51:9684/64 scope link 
       valid_lft forever preferred_lft forever
7: pppoe-wan: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc fq_codel state UNKNOWN group default qlen 3
    link/ppp 
    inet 11.22.206.123 peer 11.22.192.1/32 scope global pppoe-wan
       valid_lft forever preferred_lft forever
    inet6 2408:742:5840:228f:c386:4f50:c27:2751/64 scope global dynamic noprefixroute 
       valid_lft 2591571sec preferred_lft 604371sec
    inet6 fe80::c656:4f50:c27:2751 peer fe80::d6c1:c8ff:fe93:8d30/128 scope link 
       valid_lft forever preferred_lft forever
8: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
    link/ether 02:42:c6:d1:82:dc brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
root@OpenWrt:~# 
root@OpenWrt:~# cat /etc/config/network 

config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'f8a1:65e3:c0af::/48'
        option packet_steering '1'

config interface 'lan'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.8.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option ifname 'eth0 eth2'

config interface 'wan'
        option ifname 'eth1'
        option proto 'pppoe'
        option username 'myusername'
        option password 'mypassword'
        option ipv6 'auto'
        option keepalive '0'

config interface 'wan6'
        option ifname 'eth1'
        option proto 'dhcpv6'

config interface 'docker'
        option ifname 'docker0'
        option proto 'none'
        option auto '0'

config device
        option type 'bridge'
        option name 'docker0'
root@OpenWrt:~# 
root@OpenWrt:~# cat /etc/os-release 
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r6542-3de103535"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r6542-3de103535"
root@OpenWrt:~# 
root@OpenWrt:~# uname -a
Linux OpenWrt 5.15.153 #0 SMP Sat Apr 6 11:37:57 2024 x86_64 GNU/Linux

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

J4125 i210 ESxi 8.0

### 详细日志

编译无措

coolsnowwolf, 2024-04-10:
局域网里面有环路

GreatMichaelLee, 2024-04-10:
> ### 详细叙述
> J4125 i210 ESXi 8.0，eth0虚拟网卡(ESXi里查看mac 00:0c:29:f1:e6:85)，eth1(WAN)、eth2(桥接到br-lan)直通，log偶尔出现如下提示，毫无规律，但感觉没影响使用
> 
> root@OpenWrt:~# root@OpenWrt:~# date Tue Apr 9 11:15:27 CST 2024 root@OpenWrt:~# root@OpenWrt:~# logread | grep 'with own address as source address' Mon Apr 8 17:47:59 2024 kern.warn kernel: [26398.559529] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0) Tue Apr 9 07:54:01 2024 kern.warn kernel: [77155.930527] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0) Tue Apr 9 07:54:01 2024 kern.warn kernel: [77156.074518] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0) Tue Apr 9 07:54:01 2024 kern.warn kernel: [77156.204865] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0) Tue Apr 9 07:54:01 2024 kern.warn kernel: [77156.335611] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0) Tue Apr 9 07:54:01 2024 kern.warn kernel: [77156.465949] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0) Tue Apr 9 07:54:01 2024 kern.warn kernel: [77156.596556] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0) Tue Apr 9 07:54:02 2024 kern.warn kernel: [77156.727378] br-lan: received packet on eth2 with own address as source address (addr:00:0c:29:f1:e6:85, vlan:0) root@OpenWrt:~# root@OpenWrt:~# ip addr 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000 link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00 inet 127.0.0.1/8 scope host lo valid_lft forever preferred_lft forever inet6 ::1/128 scope host valid_lft forever preferred_lft forever 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP group default qlen 1000 link/ether 00:0c:29:f1:e6:85 brd ff:ff:ff:ff:ff:ff 3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000 link/ether 5d:62:e3:57:2e:51 brd ff:ff:ff:ff:ff:ff inet6 fe80::c656:4fff:fe27:2751/64 scope link valid_lft forever preferred_lft forever 4: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP group default qlen 1000 link/ether 5d:62:e3:57:2e:52 brd ff:ff:ff:ff:ff:ff 5: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000 link/ether 80:3a:16:db:2c:89 brd ff:ff:ff:ff:ff:ff 6: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000 link/ether 00:0c:29:f1:e6:85 brd ff:ff:ff:ff:ff:ff inet 192.168.8.1/24 brd 192.168.8.255 scope global br-lan valid_lft forever preferred_lft forever inet6 2408:744:5803:2e8f::1/60 scope global dynamic noprefixroute valid_lft 45169sec preferred_lft 45169sec inet6 f8a1:65e3:c0af::1/60 scope global noprefixroute valid_lft forever preferred_lft forever inet6 fe80::20c:29ff:fe51:9684/64 scope link valid_lft forever preferred_lft forever 7: pppoe-wan: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc fq_codel state UNKNOWN group default qlen 3 link/ppp inet 11.22.206.123 peer 11.22.192.1/32 scope global pppoe-wan valid_lft forever preferred_lft forever inet6 2408:742:5840:228f:c386:4f50:c27:2751/64 scope global dynamic noprefixroute valid_lft 2591571sec preferred_lft 604371sec inet6 fe80::c656:4f50:c27:2751 peer fe80::d6c1:c8ff:fe93:8d30/128 scope link valid_lft forever preferred_lft forever 8: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default link/ether 02:42:c6:d1:82:dc brd ff:ff:ff:ff:ff:ff inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0 valid_lft forever preferred_lft forever root@OpenWrt:~# root@OpenWrt:~# cat /etc/config/network
> 
> config interface 'loopback' option ifname 'lo' option proto 'static' option ipaddr '127.0.0.1' option netmask '255.0.0.0'
> 
> config globals 'globals' option ula_prefix 'f8a1:65e3:c0af::/48' option packet_steering '1'
> 
> config interface 'lan' option type 'bridge' option proto 'static' option ipaddr '192.168.8.1' option netmask '255.255.255.0' option ip6assign '60' option ifname 'eth0 eth2'
> 
> config interface 'wan' option ifname 'eth1' option proto 'pppoe' option username 'myusername' option password 'mypassword' option ipv6 'auto' option keepalive '0'
> 
> config interface 'wan6' option ifname 'eth1' option proto 'dhcpv6'
> 
> config interface 'docker' option ifname 'docker0' option proto 'none' option auto '0'
> 
> config device option type 'bridge' option name 'docker0' root@OpenWrt:~# root@OpenWrt:~# cat /etc/os-release NAME="OpenWrt" VERSION="SNAPSHOT" ID="openwrt" ID_LIKE="lede openwrt" PRETTY_NAME="OpenWrt SNAPSHOT" VERSION_ID="snapshot" HOME_URL="https://openwrt.org/" BUG_URL="https://bugs.openwrt.org/" SUPPORT_URL="https://forum.openwrt.org/" BUILD_ID="r6542-3de103535" OPENWRT_BOARD="x86/64" OPENWRT_ARCH="x86_64" OPENWRT_TAINTS="no-all" OPENWRT_DEVICE_MANUFACTURER="OpenWrt" OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/" OPENWRT_DEVICE_PRODUCT="Generic" OPENWRT_DEVICE_REVISION="v0" OPENWRT_RELEASE="OpenWrt SNAPSHOT r6542-3de103535" root@OpenWrt:~# root@OpenWrt:~# uname -a Linux OpenWrt 5.15.153 #0 SMP Sat Apr 6 11:37:57 2024 x86_64 GNU/Linux
> 
> ### 重复 issue
> * [x]  没有类似的 issue
> 
> ### 具体型号
> J4125 i210 ESXi 8.0
> 
> ### 详细日志
> 编译无错

估计你是用mesh路由把，环路。

closed 2024-04-10 by coolsnowwolf

eearphon, 2024-04-10:
奇怪，就一个AP接eth2，这也想过的，就是没想出环在哪了。。。

eearphon, 2024-04-13:
> 估计你是用mesh路由把，环路。

AX3600是mesh路由，请问怎么会导致log出现环路提示？