coolsnowwolf/lede #12062: e2guardian编译报错

binge8, 2024-04-10:
### 详细叙述

最新代码编译报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x64.armv8

### 详细日志

'''bash
/home/runner/work/op/op/openwrt/package/network/services/e2guardian'
2024-04-09T19:01:34.4856841Z rm -f /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/e2guardian-3.2.0/.configured_*
2024-04-09T19:01:34.4885293Z rm -f /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/stamp/.e2guardian_installed
2024-04-09T19:01:34.4942871Z (cd /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/e2guardian-3.2.0; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf ./autom4te.cache; [ -e ./config.rpath ] || ln -s /home/runner/work/op/op/openwrt/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/runner/work/op/op/openwrt/staging_dir/host/bin/autom4te AUTOCONF=/home/runner/work/op/op/openwrt/staging_dir/host/bin/autoconf AUTOMAKE=/home/runner/work/op/op/openwrt/staging_dir/host/bin/automake ACLOCAL=/home/runner/work/op/op/openwrt/staging_dir/host/bin/aclocal AUTOHEADER=/home/runner/work/op/op/openwrt/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/runner/work/op/op/openwrt/staging_dir/host/bin/libtoolize LIBTOOL=/home/runner/work/op/op/openwrt/staging_dir/host/bin/libtool M4=/home/runner/work/op/op/openwrt/staging_dir/host/bin/m4 AUTOPOINT=true LIBTOOLIZE='/home/runner/work/op/op/openwrt/staging_dir/host/bin/libtoolize --install' /home/runner/work/op/op/openwrt/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/runner/work/op/op/openwrt/staging_dir/host/share/aclocal -I /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/host/share/aclocal -I /home/runner/work/op/op/openwrt/staging_dir/hostpkg/share/aclocal -I /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . . || true; fi; );
2024-04-09T19:01:34.6651479Z autoreconf: Entering directory `.'
2024-04-09T19:01:34.6659366Z autoreconf: configure.ac: not using Gettext
2024-04-09T19:01:35.6666328Z autoreconf: running: /home/runner/work/op/op/openwrt/staging_dir/host/bin/aclocal -I /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/host/share/aclocal -I /home/runner/work/op/op/openwrt/staging_dir/hostpkg/share/aclocal -I /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . --force 
2024-04-09T19:01:36.0688642Z configure.ac:12: error: AM_INIT_AUTOMAKE expanded multiple times
2024-04-09T19:01:36.0690712Z /home/runner/work/op/op/openwrt/staging_dir/host/share/aclocal-1.16/init.m4:29: AM_INIT_AUTOMAKE is expanded from...
2024-04-09T19:01:36.0692380Z configure.ac:9: the top level
2024-04-09T19:01:36.0693385Z /home/runner/work/op/op/openwrt/staging_dir/host/share/aclocal-1.16/init.m4:29: AM_INIT_AUTOMAKE is expanded from...
2024-04-09T19:01:36.0694559Z configure.ac:12: the top level
2024-04-09T19:01:36.0695267Z autom4te: /home/runner/work/op/op/openwrt/staging_dir/host/bin/m4 failed with exit status: 1
2024-04-09T19:01:36.0715923Z aclocal.real: error: /home/runner/work/op/op/openwrt/staging_dir/host/bin/autom4te failed with exit status: 1
2024-04-09T19:01:36.0824439Z autoreconf: /home/runner/work/op/op/openwrt/staging_dir/host/bin/aclocal failed with exit status: 1
2024-04-09T19:01:36.0847095Z ( cd /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/e2guardian-3.2.0; ./autogen.sh )
2024-04-09T19:01:36.0877987Z + cp README.md README
2024-04-09T19:01:36.0890880Z + aclocal -I m4
2024-04-09T19:01:36.4960630Z configure.ac:12: error: AM_INIT_AUTOMAKE expanded multiple times
2024-04-09T19:01:36.4962860Z /home/runner/work/op/op/openwrt/staging_dir/host/share/aclocal-1.16/init.m4:29: AM_INIT_AUTOMAKE is expanded from...
2024-04-09T19:01:36.4964419Z configure.ac:9: the top level
2024-04-09T19:01:36.4965482Z /home/runner/work/op/op/openwrt/staging_dir/host/share/aclocal-1.16/init.m4:29: AM_INIT_AUTOMAKE is expanded from...
2024-04-09T19:01:36.4966768Z configure.ac:12: the top level
2024-04-09T19:01:36.4967754Z autom4te: /home/runner/work/op/op/openwrt/staging_dir/host/bin/m4 failed with exit status: 1
2024-04-09T19:01:36.4983389Z aclocal.real: error: autom4te failed with exit status: 1
2024-04-09T19:01:36.5093010Z make[3]: *** [Makefile:86: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/e2guardian-3.2.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
2024-04-09T19:01:36.5094379Z make[3]: Leaving directory '/home/runner/work/op/op/openwrt/package/network/services/e2guardian'
2024-04-09T19:01:36.5097274Z time: package/network/services/e2guardian/compile#1.04#0.16#2.17
2024-04-09T19:01:36.5104163Z     ERROR: package/network/services/e2guardian failed to build.
2024-04-09T19:01:36.5108435Z make[2]: *** [package/Makefile:116: package/network/services/e2guardian/compile] Error 1
2024-04-09T19:01:36.5113897Z make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
2024-04-09T19:01:36.5120540Z make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2024-04-09T19:01:36.5124639Z make[1]: Leaving directory '/home/runner/work/op/op/openwrt'
2024-04-09T19:01:36.5133154Z make: *** [/home/runner/work/op/op/openwrt/include/toplevel.mk:231: world] Error 2
'''

WYC-2020, 2024-04-11:
[0002-AM_INIT_AUTOMAKE-expanded-multiple-times.patch](https://github.com/coolsnowwolf/lede/files/14940233/0002-AM_INIT_AUTOMAKE-expanded-multiple-times.patch)
老规矩

closed 2024-04-11 by binge8

hzjnet, 2024-05-07:
故障依旧

hzjnet, 2024-06-30:
> [0002-AM_INIT_AUTOMAKE-expanded-multiple-times.patch](https://github.com/coolsnowwolf/lede/files/14940233/0002-AM_INIT_AUTOMAKE-expanded-multiple-times.patch) 老规矩

这个补丁放在：package/network/services/e2guardian/patches/0002-AM_INIT_AUTOMAKE-expanded-multiple-times.patch
编译提示如下：
. /openwrt/openwrt-cools-M-git-fsmallprogram-20240629.feeds/include/shell.sh; /openwrt/openwrt-cools-M-git-fsmallprogram-20240629.feeds/staging_dir/host/bin/libdeflate-gzip -dc /openwrt/openwrt-cools-M-git-fsmallprogram-20240629.feeds/dl/v3.2.0.tar.gz | tar -C /openwrt/openwrt-cools-M-git-fsmallprogram-20240629.feeds/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/e2guardian-3.2.0/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /openwrt/openwrt-cools-M-git-fsmallprogram-20240629.feeds/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/e2guardian-3.2.0

Applying ./patches/0002-AM_INIT_AUTOMAKE-expanded-multiple-times.patch using plaintext: 
patching file configure.ac
Hunk #1 FAILED at 6.
1 out of 1 hunk FAILED -- saving rejects to file configure.ac.rej
Patch failed!  Please fix ./patches/0002-AM_INIT_AUTOMAKE-expanded-multiple-times.patch!
make[3]: *** [Makefile:89: /openwrt/openwrt-cools-M-git-fsmallprogram-20240629.feeds/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/e2guardian-3.2.0/.prepared_2b18f17289827b1bf065b7f6d401845e_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/openwrt/openwrt-cools-M-git-fsmallprogram-20240629.feeds/package/network/services/e2guardian'
time: package/network/services/e2guardian/compile#0.09#0.05#0.13
    ERROR: package/network/services/e2guardian failed to build.
make[2]: *** [package/Makefile:116: package/network/services/e2guardian/compile] Error 1

不知道是放错位置还是需要新补丁啊？

hzjnet, 2024-06-30:
> 这个补丁放在：package/network/services/e2guardian/patches/0002-AM_INIT_AUTOMAKE-expanded-multiple-times.patch 编译提示如下： . /openwrt/openwrt-cools-M-git-fsmallprogram-20240629.feeds/include/shell.sh; [...]

喔  看了补丁才知道更换主版本为5.3.5

find bin|grep e2guardian              
bin/packages/arm_cortex-a15_neon-vfpv4/base/e2guardian_5.3.5-1_arm_cortex-a15_neon-vfpv4.ipk
bin/packages/arm_cortex-a15_neon-vfpv4/luci/luci-app-e2guardian_git-24.174.35735-b8fa320-1_all.ipk

Deerczl, 2024-07-08:
> 喔 看了补丁才知道更换主版本为5.3.5
>
> find bin|grep e2guardian bin/packages/arm_cortex-a15_neon-vfpv4/base/e2guardian_5.3.5-1_arm_cortex-a15_neon-vfpv4.ipk [...]

我也遇到了同样的问题，请问如何指定编译的版本呢？