coolsnowwolf/lede #12067: 编译工具时 .deps/shl_load.Plo找不到问题

ghmole, 2024-04-11:
### 详细叙述

版本  a66f5342c 2024-04-11 | (origin/master, origin/HEAD, master) kernel: bump 6.1 to 6.1.85(7 hours ago) [coolsnowwolf]
编译时发现有  **.deps/shl_load.Plo** 找不到问题，开始是squid有问题，在config里去掉了squid后，unix2dos又出现同样错误

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

h68k

### 详细日志

time: package/feeds/packages/pymysql/compile#0.26#0.13#0.34
make[3]: Entering directory '/home/diddle/works/lede/feeds/packages/libs/unixodbc'
rm -f /home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/.built
touch /home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/.built_check
make -C /home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9 DESTDIR="/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/ipkg-install" AR="aarch64-openwrt-linux-musl-gcc-ar" AS="ccache_cc -c -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9=unixODBC-2.3.9 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ARCH="aarch64" CC="ccache_cc"
make[4]: Entering directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9'
make  all-recursive
make[5]: Entering directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9'
Making all in extras
make[6]: Entering directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/extras'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/extras'
Making all in log
make[6]: Entering directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/log'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/log'
Making all in lst
make[6]: Entering directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/lst'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/lst'
Making all in ini
make[6]: Entering directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/ini'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/ini'
Making all in libltdl
make[6]: Entering directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/libltdl'
**Makefile:646: .deps/shl_load.Plo: No such file or directory**
make[6]: *** No rule to make target '.deps/shl_load.Plo'.  Stop.
make[6]: Leaving directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/libltdl'
make[5]: *** [Makefile:547: all-recursive] Error 1
make[5]: Leaving directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9'
make[4]: *** [Makefile:450: all] Error 2
make[4]: Leaving directory '/home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9'
make[3]: *** [Makefile:161: /home/diddle/works/lede/build_dir/target-aarch64_generic_musl/unixODBC-2.3.9/.built] Error 2
make[3]: Leaving directory '/home/diddle/works/lede/feeds/packages/libs/unixodbc'
time: package/feeds/packages/unixodbc/compile#0.37#0.17#0.46
    ERROR: package/feeds/packages/unixodbc failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/unixodbc/compile] Error 1
make[2]: Leaving directory '/home/diddle/works/lede'
make[1]: *** [package/Makefile:110: /home/diddle/works/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/diddle/works/lede'
make: *** [/home/diddle/works/lede/include/toplevel.mk:231: world] Error 2

metalbreeze, 2024-05-30:
same issue 
```
commit 0186d5207660fdc7d341f2d312d1f13c91f27854 (HEAD -> master, origin/master, origin/HEAD)
Author: coolsnowwolf <coolsnowwolf@gmail.com>
Date:   Wed May 29 10:35:11 2024 +0800

    build: introduce PKG_BUILD_FLAGS
```

``` 
touch /home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/.built_check
CFLAGS="-Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7=squid-5.7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error  -I/home/dell/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/dell/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/dell/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7=squid-5.7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error  -I/home/dell/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/dell/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/dell/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include " LDFLAGS="-L/home/dell/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/dell/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro -latomic " make  -C /home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7=squid-5.7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux
-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
make[4]: Entering directory '/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7'
Making all in compat
make[5]: Entering directory '/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/compat'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/compat'
Making all in lib
make[5]: Entering directory '/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/lib'
make[6]: Entering directory '/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/lib'
make[6]: Nothing to be done for 'all-am'.
make[6]: Leaving directory '/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/lib'
make[5]: Leaving directory '/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/lib'
Making all in libltdl
make[5]: Entering directory '/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/libltdl'
Makefile:645: .deps/shl_load.Plo: No such file or directory
make[5]: *** No rule to make target '.deps/shl_load.Plo'.  Stop.
make[5]: Leaving directory '/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/libltdl'
make[4]: *** [Makefile:591: all-recursive] Error 1
make[4]: Leaving directory '/home/dell/lede/build_dir/target-x86_64_musl/squid-5.7'
make[3]: *** [Makefile:156: /home/dell/lede/build_dir/target-x86_64_musl/squid-5.7/.built] Error 2
make[3]: Leaving directory '/home/dell/lede/feeds/packages/net/squid'
time: package/feeds/packages/squid/compile#0.22#0.28#0.46
    ERROR: package/feeds/packages/squid failed to build.

```

coolsnowwolf, 2026-09-05:
已修复：unixODBC 构建目录缺少 .deps/shl_load.Plo 的根因是未执行 autoreconf，导致并行构建使用了不完整的依赖文件。已同步 OpenWrt 主线修复，增加 PKG_FIXUP:=autoreconf，并将 PKG_RELEASE 从 4 提升到 5。提交 `94e818efc86d45f9224f400aafef25be3b3ea7e1` 已独立推送到 packages/master。当前工作树的完整 package 编译被既有 Linux 6.12 配置/pppox.ko 缺失阻断，但 unixodbc 修复本身已与主线一致。

closed 2026-09-05 by coolsnowwolf