coolsnowwolf/lede #12070: 使用SD卡的设备烧录后无法启动

Gabrielxzx, 2024-04-13:
### 详细叙述

RT
使用SD卡的设备烧录后无法启动

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

orangepi_zero / orangepi_zero_plus / nanopi_r2c

### 详细日志

暂无，编译正常，但烧录进SD后无法启动。

coolsnowwolf, 2024-04-14:
看起来像是 TF 的高速卡兼容问题，和 uboot 的

Gabrielxzx, 2024-04-14:
> 看起来像是 TF 的高速卡兼容问题，和 uboot 的

可以看看nanopi_neo2的串口日志做参考，和orangepi_zero还不一致

> U-Boot SPL 2024.01-OpenWrt-r0-373bf02 (Apr 11 2024 - 06:09:38 +0000)
DRAM: 512 MiB
Trying to boot from MMC1
NOTICE:  BL31: v2.10.0	(release):OpenWrt v2.10-1 (sunxi-a64)
NOTICE:  BL31: Built : 06:09:38, Apr 11 2024
NOTICE:  BL31: Detected Allwinner H5 SoC (1718)
NOTICE:  BL31: Found U-Boot DTB at 0x20a40e8, model: FriendlyARM NanoPi NEO 2
NOTICE:  PMIC: Assuming H5 reference regulator design

coolsnowwolf, 2024-04-15:
> 也有类似的问题 4月4号源码编译的r2s固件可以启动 4月13号编译的不行

你尝试 revert ATF 版本试试

MilesPoupart, 2024-04-15:
> > 也有类似的问题 4月4号源码编译的r2s固件可以启动 4月13号编译的不行
> 
> 你尝试 revert ATF 版本试试

好 俺把这几个commit revert了试试看
[image]

Gabrielxzx, 2024-04-15:
> 好 俺把这几个commit revert了试试看 [...]

有效果记得来反馈下啊(づ ●─● )づ

MilesPoupart, 2024-04-15:
> 有效果记得来反馈下啊(づ ●─● )づ

[image]
revert这四个commit之后开出来了（r2s）

sijyun, 2024-04-15:
我的 R2S 恢復,這一項就可以了.  

arm-trusted-firmware-rockchip-vendor: add RK3528/RK3588 support

lysgwl, 2024-04-19:
编译了好几个版本的Rockchip芯片固件，r2s设备确实是无法启动。是因为升级了内核原因造成的吗？不太确认。

lysgwl, 2024-04-19:
> 我的 R2S 恢復,這一項就可以了.
> 
> arm-trusted-firmware-rockchip-vendor: add RK3528/RK3588 support

增加了选项，也是不可以的。你是怎么操作的？

sijyun, 2024-04-19:
> > 我的 R2S 恢復,這一項就可以了.
> > arm-trusted-firmware-rockchip-vendor: add RK3528/RK3588 support
> 
> 增加了选项，也是不可以的。你是怎么操作的？

前置準備好後, (我是用自己電腦編譯)

**動作**
git revert 3b681972500466d5ea56c1e5cea428954687c6f2
make menuconfig
make V=s ( -j=1 可選用)

**確認**
git status

Changes to be committed:
  (use "git restore --staged <file>..." to unstage)
        modified:   package/boot/arm-trusted-firmware-rockchip-vendor/Makefile
        modified:   package/boot/arm-trusted-firmware-rockchip-vendor/atf-version.mk
        deleted:    package/boot/arm-trusted-firmware-rockchip-vendor/pack-firmware.sh
        modified:   package/boot/uboot-rockchip/Makefile

[image]

lysgwl, 2024-04-20:
> > > 我的 R2S 恢復,這一項就可以了.
> > > arm-trusted-firmware-rockchip-vendor: add RK3528/RK3588 support
> > 
> > 
> > 增加了选项，也是不可以的。你是怎么操作的？
> 
> 前置準備好後, (我是用自己電腦編譯)
> 
> **動作** git revert [3b68197](https://github.com/coolsnowwolf/lede/commit/3b681972500466d5ea56c1e5cea428954687c6f2) make menuconfig make V=s ( -j=1 可選用)
> 
> **確認** git status
> 
> Changes to be committed: (use "git restore --staged ..." to unstage) modified: package/boot/arm-trusted-firmware-rockchip-vendor/Makefile modified: package/boot/arm-trusted-firmware-rockchip-vendor/atf-version.mk deleted: package/boot/arm-trusted-firmware-rockchip-vendor/pack-firmware.sh modified: package/boot/uboot-rockchip/Makefile
> 
> ![r2s-k6 1 87-0420](https://private-user-images.githubusercontent.com/58562726/324045036-aa582a72-bce5-4fee-9ea7-17e76f0b1ac9.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM1Nzg5NjksIm5iZiI6MTcxMzU3ODY2OSwicGF0aCI6Ii81ODU2MjcyNi8zMjQwNDUwMzYtYWE1ODJhNzItYmNlNS00ZmVlLTllYTctMTdlNzZmMGIxYWM5LlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDIwVDAyMDQyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY4ZDVmZDAyMTgwNDliZTkxZjZhODllNDdkYjU2MWMzNjdiZjUwZTliODU3MjM2Mjc1ZmM3OThhMGIzZDkxMTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.UKmhDXmJ7rAhV16o45ND9zCO1BZcRSkq1QJvgnYGueI)

额，嗖嘎，你是回退了，是吧。我还以为增加一个rockchip选项。昨天我增加了一个选项，也是不行。

Gabrielxzx, 2024-04-20:
SUNXI：
git revert e4172db --no-edit
RK：
git revert f788f35 --no-edit
git revert 3b68197 --no-edit

回退后可以启动

Gabrielxzx, 2024-04-21:
> > 也有类似的问题 4月4号源码编译的r2s固件可以启动 4月13号编译的不行
> 
> 你尝试 revert ATF 版本试试

SUNXI：
git revert e4172db --no-edit
RK：
git revert https://github.com/coolsnowwolf/lede/commit/f788f35b36162cf0a780aa5d98bafeb70341ba63 --no-edit
git revert https://github.com/coolsnowwolf/lede/commit/3b681972500466d5ea56c1e5cea428954687c6f2 --no-edit

revert后可以启动

facerleo, 2024-04-25:
现在还是不行吗

Gabrielxzx, 2024-04-25:
> > 也有类似的问题 4月4号源码编译的r2s固件可以启动 4月13号编译的不行
> 
> 你尝试 revert ATF 版本试试

大佬帮忙看看吧，RK的能跑了，但SUNXI还是无法bringup

closed 2024-04-27 by coolsnowwolf

Gabrielxzx, 2024-04-28:
> > > 也有类似的问题 4月4号源码编译的r2s固件可以启动 4月13号编译的不行
> > 
> > 
> > 你尝试 revert ATF 版本试试
> 
> 大佬帮忙看看吧，今天更新后，RK的能跑了，但SUNXI还是无法bringup

大佬帮忙瞅瞅吧，SUNXI的板子确实无法启动，已经测试确认删除261-sunxi-add-popstick-defconfig.patch就可以启动。
rm -rf package/boot/uboot-sunxi/patches/261-sunxi-add-popstick-defconfig.patch