coolsnowwolf/lede #12080: 最新代码5.15内核编译错误

ysuolmai, 2024-04-18:
### 详细叙述

最新版默认6.1，编译出来没看到nss,菜单里也没看到有WiFi，就换成5.15来编译，死活不能成功.看到有类似issue但是没有人解释原因直接被complete了。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

兆能M2

### 详细日志

Packaged contents of /home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/ipkg-aarch64_cortex-a53/kmod-crypto-user into /home/parallels/lede/bin/targets/qualcommax/ipq60xx/packages/kmod-crypto-user_5.15.155-1_aarch64_cortex-a53.ipk
NOTICE: module '/home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-5.15.155/lib/crypto/libarc4.ko' is built-in.
rstrip.sh: /home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/ipkg-aarch64_cortex-a53/kmod-crypto-arc4/lib/modules/5.15.155/arc4.ko: relocatable
Packaged contents of /home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/ipkg-aarch64_cortex-a53/kmod-crypto-arc4 into /home/parallels/lede/bin/targets/qualcommax/ipq60xx/packages/kmod-crypto-arc4_5.15.155-1_aarch64_cortex-a53.ipk
rstrip.sh: /home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/ipkg-aarch64_cortex-a53/kmod-crypto-authenc/lib/modules/5.15.155/authenc.ko: relocatable
rstrip.sh: /home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/ipkg-aarch64_cortex-a53/kmod-crypto-authenc/lib/modules/5.15.155/authencesn.ko: relocatable
Packaged contents of /home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/ipkg-aarch64_cortex-a53/kmod-crypto-authenc into /home/parallels/lede/bin/targets/qualcommax/ipq60xx/packages/kmod-crypto-authenc_5.15.155-1_aarch64_cortex-a53.ipk
rstrip.sh: /home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/ipkg-aarch64_cortex-a53/kmod-crypto-hmac/lib/modules/5.15.155/hmac.ko: relocatable
Packaged contents of /home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/ipkg-aarch64_cortex-a53/kmod-crypto-hmac into /home/parallels/lede/bin/targets/qualcommax/ipq60xx/packages/kmod-crypto-hmac_5.15.155-1_aarch64_cortex-a53.ipk
ERROR: module '/home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-5.15.155/arch/arm64/crypto/sha512-arm64.ko' is missing.
make[3]: *** [modules/crypto.mk:1090: /home/parallels/lede/bin/targets/qualcommax/ipq60xx/packages/kmod-crypto-sha512_5.15.155-1_aarch64_cortex-a53.ipk] Error 1
make[3]: Leaving directory '/home/parallels/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.43#0.16#2.55
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/parallels/lede'
make[1]: *** [package/Makefile:110: /home/parallels/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/parallels/lede'
make: *** [/home/parallels/lede/include/toplevel.mk:231: world] Error 2

ysuolmai, 2024-04-19:
换成5.4也是一样的问题, sha512-arm64.ko 缺失
[log.txt](https://github.com/coolsnowwolf/lede/files/15032977/log.txt)

eearphon, 2024-04-19:
kmod-crypto-sha512 拿掉一试

ysuolmai, 2024-04-19:
> kmod-crypto-sha512 拿掉一试

又提示sha256-arm64.ko缺失...

make[3]: Leaving directory '/home/parallels/lede/package/network/config/netifd'
time: package/network/config/netifd/compile#0.08#0.02#0.10
make[3]: Entering directory '/home/parallels/lede/package/firmware/linux-firmware'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/home/parallels/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#0.09#0.01#0.10
make[3]: Entering directory '/home/parallels/lede/package/kernel/linux'
mkdir -p /home/parallels/lede/staging_dir/target-aarch64_cortex-a53_musl/root-qualcommax/stamp

SHELL= flock /home/parallels/lede/tmp/.root-copy.flock -c 'cp -fpR /home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/packages/.pkgdir/kernel/. /home/parallels/lede/staging_dir/target-aarch64_cortex-a53_musl/root-qualcommax/'
touch /home/parallels/lede/staging_dir/target-aarch64_cortex-a53_musl/root-qualcommax/stamp/.kernel_installed
NOTICE: module '/home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-5.4.273/crypto/sha256_generic.ko' is built-in.
NOTICE: module '/home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-5.4.273/lib/crypto/libsha256.ko' is built-in.
ERROR: module '/home/parallels/lede/build_dir/target-aarch64_cortex-a53_musl/linux-qualcommax_ipq60xx/linux-5.4.273/arch/arm64/crypto/sha256-arm64.ko' is missing.
make[3]: *** [modules/crypto.mk:1039: /home/parallels/lede/bin/targets/qualcommax/ipq60xx/packages/kmod-crypto-sha256_5.4.273-1_aarch64_cortex-a53.ipk] Error 1
make[3]: Leaving directory '/home/parallels/lede/package/kernel/linux'
time: package/kernel/linux/compile#2.09#0.13#2.22
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/parallels/lede'
make[1]: *** [package/Makefile:110: /home/parallels/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/parallels/lede'
make: *** [/home/parallels/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2024-04-23:
5.15都被砍掉了很多配置和patch都删除了， 如果你只是简单改成5.15编译 那肯定失败，非要5.15自己回退回去看看

ysuolmai, 2024-04-23:
> 5.15都被砍掉了很多配置和patch都删除了， 如果你只是简单改成5.15编译 那肯定失败，非要5.15自己回退回去看看

嗯我放弃了, 编6.1可以成功,但是没有wifi. 大雕说要改dts, 但是改了也是没有效果. 暂时先用别人的代码编译了用了.

wxzy322, 2024-07-06:
> > 5.15都被砍掉了很多配置和patch都删除了， 如果你只是简单改成5.15编译 那肯定失败，非要5.15自己回退回去看看
> 
> 嗯我放弃了, 编6.1可以成功,但是没有wifi. 大雕说要改dts, 但是改了也是没有效果. 暂时先用别人的代码编译了用了.

你给啥机器编译啊    我编译的6.1  有wifi  和nss    360v6   但是 点击 重启就关机了

iYongHsu, 2025-02-20:
> > > 5.15都被砍掉了很多配置和patch都删除了， 如果你只是简单改成5.15编译 那肯定失败，非要5.15自己回退回去看看
> > 
> > 
> > 嗯我放弃了, 编6.1可以成功,但是没有wifi. 大雕说要改dts, 但是改了也是没有效果. 暂时先用别人的代码编译了用了.
> 
> 你给啥机器编译啊 我编译的6.1 有wifi 和nss 360v6 但是 点击 重启就关机了

兄弟，这个点击重启就关机的问题，现在最新源码编译还有问题，你有解决了吗？

coolsnowwolf, 2026-09-05:
原日志来自 5.15.155 的旧构建环境。当前 LEDE 的 `kmod-crypto-sha512` 已按内核版本和架构分别声明 `CONFIG_CRYPTO_SHA512_ARM64@lt6.18` 及 `arch/arm64/crypto/sha512-arm64.ko`，不再使用旧的无条件模块列表；当前 6.x 构建也已切换到新的 `sha512.ko/libsha512.ko` 路径。原 issue 中的缺失模块问题对应旧源码，当前代码未复现。

closed 2026-09-05 by coolsnowwolf